chipsalliance/Surelog #2199: Problems with parameter substitution - continuation

RRozak, 2021-11-16:
Continuation of https://github.com/chipsalliance/Surelog/issues/2167

The problem occurs only in some of the examples. They are harder to understand, so let’s first look at the simplified one:
```systemverilog
module prim_submodule;
   parameter int PadType = 0;
   int           o = PadType;
endmodule // prim_submodule

module prim_pad_attr;
   import my_pkg::*;
   parameter struct_t TargetCfg = 3;

   parameter int PadType = 0;

   prim_submodule #(
      .PadType(PadType)
   ) u_submodule();
endmodule // prim_pad_attr

module almost_top;
   import my_pkg::*;
   parameter struct_t TargetCfg = 1;

   prim_pad_attr #(
      .PadType(TargetCfg.dio_pad_type)
   ) u_prim_pad_attr();
endmodule

module top;
   import my_pkg::*;
   parameter struct_t TargetCfg = 2;

   almost_top #(
      .TargetCfg(TargetCfg)
   ) u_top();
endmodule
```
When PadType in prim_submodule (line 3) is substituted with TargetCfg.dio_pad_type, we don’t know if it is the TargetCfg from prim_pad_attr, almost_top or top module. There is no information about it in this vpiParamAssign node.

To know the example on which the problem occurs, please consider the following files:
```systemverilog
package pinmux_pkg;
   typedef enum int {
      A = 1
   } enum1_e;

   typedef struct packed {
      enum1_e dio_pad_type;
   } struct1_t;

   typedef enum int {
      B = 2
   } enum2_e;

   typedef struct packed {
      enum2_e dio_pad_type;
   } struct2_t;
endpackage : pinmux_pkg

module prim_generic_pad_attr(output int o);
   parameter int PadTypeInGeneric = 0;
   assign o = PadTypeInGeneric;                  // assigning value of TargetCfg.dio_pad_type
endmodule : prim_generic_pad_attr

module prim_submodule(output int o);
   parameter int PadType = 0;

   prim_generic_pad_attr #(
       .PadTypeInGeneric(PadType)
   ) u_impl_generic(
      .o(o)
   );
endmodule

module prim_pad_attr(output int o);
   import pinmux_pkg::*;

   parameter int PadType = 0;

   prim_submodule #(
      .PadType(PadType)
   ) u_submodule(
      .o(o)
   );
endmodule

module almost_top(output int o);
   import pinmux_pkg::*;
   parameter struct1_t TargetCfg = 1;

   prim_pad_attr #(
      .PadType(TargetCfg.dio_pad_type)
   ) u_prim_pad_attr(
      .o(o)
   );
endmodule

module top(output int o);
   import pinmux_pkg::*;
   parameter struct2_t TargetCfg = 2;                // assigning value of TargetCfg

   almost_top #(
      .TargetCfg(TargetCfg)
   ) u_top(
      .o(o)
   );
endmodule
```
And
```systemverilog
package pinmux_pkg;
   typedef enum int {
      A = 1
   } enum1_e;

   typedef struct packed {
      enum1_e dio_pad_type;
   } struct1_t;

   typedef enum int {
      B = 2
   } enum2_e;

   typedef struct packed {
      enum2_e dio_pad_type;
   } struct2_t;
endpackage : pinmux_pkg

module prim_generic_pad_attr(output int o);
   parameter int PadTypeInGeneric = 0;
   assign o = PadTypeInGeneric;                  // assigning value of TargetCfg.dio_pad_type
endmodule : prim_generic_pad_attr

module prim_submodule(output int o);
   parameter int PadType = 0;

   prim_generic_pad_attr #(
       .PadTypeInGeneric(PadType)
   ) u_impl_generic(
      .o(o)
   );
endmodule

module prim_pad_attr(output int o);
   import pinmux_pkg::*;

   parameter int PadType = 0;

   prim_submodule #(
      .PadType(PadType)
   ) u_submodule(
      .o(o)
   );
endmodule

module almost_top(output int o);
   import pinmux_pkg::*;
   parameter struct1_t TargetCfg = 1;                        // assigning value of TargetCfg

   prim_pad_attr #(
      .PadType(TargetCfg.dio_pad_type)
   ) u_prim_pad_attr(
      .o(o)
   );
endmodule

module top(output int o);
   import pinmux_pkg::*;
   parameter struct2_t TargetCfg = 2;

   almost_top

     u_top(
      .o(o)
   );
endmodule
```
In the first file port o of top module should have value 2 and in the second this port should have value 1. 
There is no information in uhdm about which TargetCfg was used, so it is impossible to know which value should be assigned. Here is the full diff of the trees:
```
641a642
>       |vpiOverriden:1
645,647c646,650
<         |vpiDecompile:1
<         |vpiSize:64
<         |UINT:1
---
>         |vpiDecompile:2
>         |vpiSize:32
>         |UINT:2
>         |vpiTypespec:
>         \_struct_typespec: (pinmux_pkg::struct1_t), line:6:12, endln:6:18, parent:work@almost_top.TargetCfg
```
The nodes that differ are corresponding to assigning a value of parameter of almost_top module. The problem is with the nodes that are deeper in hierarchy and they are exactly the same in uhdm. vpiParent relationships are also the same.

The algorithm to know the value by going up in the hierarchy until the definition is found also fails.
Please consider the following file: (I added only TargetCfg parameter in prim_pad_attr and the type definition to the first file)
```systemverilog
package pinmux_pkg;
   typedef enum int {
      A = 1
   } enum1_e;

   typedef struct packed {
      enum1_e dio_pad_type;
   } struct1_t;

   typedef enum int {
      B = 2
   } enum2_e;

   typedef struct packed {
      enum2_e dio_pad_type;
   } struct2_t;

   typedef enum int {
      C = 3
   } enum3_e;

   typedef struct packed {
      enum3_e dio_pad_type;
   } struct3_t;
endpackage : pinmux_pkg

module prim_generic_pad_attr(output int o);
   parameter int PadTypeInGeneric = 0;
   assign o = PadTypeInGeneric;                  // assigning value of TargetCfg.dio_pad_type
endmodule : prim_generic_pad_attr

module prim_submodule(output int o);
   parameter int PadType = 0;

   prim_generic_pad_attr #(
       .PadTypeInGeneric(PadType)
   ) u_impl_generic(
      .o(o)
   );
endmodule

module prim_pad_attr(output int o);
   import pinmux_pkg::*;
   parameter struct3_t TargetCfg = 3;           // unused parameter
   parameter int PadType = 0;

   prim_submodule #(
      .PadType(PadType)
   ) u_submodule(
      .o(o)
   );
endmodule

module almost_top(output int o);
   import pinmux_pkg::*;
   parameter struct1_t TargetCfg = 1;

   prim_pad_attr #(
      .PadType(TargetCfg.dio_pad_type)
   ) u_prim_pad_attr(
      .o(o)
   );
endmodule

module top(output int o);
   import pinmux_pkg::*;
   parameter struct2_t TargetCfg = 2;               // assigning value of TargetCfg

   almost_top #(
      .TargetCfg(TargetCfg)
   ) u_top(
      .o(o)
   );
endmodule
```

The definition found in this way will be the one from prim_pad_attr module.

alaindargelas, 2021-11-18:
I think I found a solution, the hier_path object just have to carry another payload which is its value assigned at the location (hierarchy) where it is created. The client has to use the payload value which is assigned to the root of the hier_path and decode the value of the specific field.

alaindargelas, 2021-11-18:
#2212 should fix this problem, please confirm

RRozak, 2021-11-18:
I think it will work. Currently I am trying to find the best way to handle it, but since we know the value and the type it should be possible.
I found one problem. Why there is `77` in that line?
https://github.com/chipsalliance/Surelog/blob/2a2d58d420121b46cb76db1e734d875be521097b/tests/ComplexParamOverload/ComplexParamOverload.log#L831
It is the value of unused parameter which has only the same name as the one that is in hier_path.

In case of this file (3th code from the top of the issue):
```systemverilog
package pinmux_pkg;
   typedef enum int {
      A = 1
   } enum1_e;

   typedef struct packed {
      enum1_e dio_pad_type;
   } struct1_t;

   typedef enum int {
      B = 2
   } enum2_e;

   typedef struct packed {
      enum2_e dio_pad_type;
   } struct2_t;
endpackage : pinmux_pkg

module prim_generic_pad_attr(output int o);
   parameter int PadTypeInGeneric = 0;
   assign o = PadTypeInGeneric;                  // assigning value of TargetCfg.dio_pad_type
endmodule : prim_generic_pad_attr

module prim_submodule(output int o);
   parameter int PadType = 0;

   prim_generic_pad_attr #(
       .PadTypeInGeneric(PadType)
   ) u_impl_generic(
      .o(o)
   );
endmodule

module prim_pad_attr(output int o);
   import pinmux_pkg::*;

   parameter int PadType = 0;

   prim_submodule #(
      .PadType(PadType)
   ) u_submodule(
      .o(o)
   );
endmodule

module almost_top(output int o);
   import pinmux_pkg::*;
   parameter struct1_t TargetCfg = 1;                        // assigning value of TargetCfg

   prim_pad_attr #(
      .PadType(TargetCfg.dio_pad_type)
   ) u_prim_pad_attr(
      .o(o)
   );
endmodule

module top(output int o);
   import pinmux_pkg::*;
   parameter struct2_t TargetCfg = 2;

   almost_top

     u_top(
      .o(o)
   );
endmodule
```

your solution gave exactly what we wanted. The parameter was substituted by constant:
```
          |vpiParamAssign:
          \_param_assign: , line:20:18, endln:20:38, parent:work@top.u_top.u_prim_pad_attr.u_submodule.u_impl_generic
            |vpiOverriden:1
            |vpiRhs:
            \_hier_path: (TargetCfg.dio_pad_type), line:28:26, endln:28:33
              |vpiName:TargetCfg.dio_pad_type
              |vpiActual:
              \_ref_obj: (TargetCfg), parent:TargetCfg.dio_pad_type
                |vpiName:TargetCfg
              |vpiActual:
              \_ref_obj: (dio_pad_type), parent:TargetCfg.dio_pad_type
                |vpiName:dio_pad_type
              |vpiExpr:
              \_constant: , line:48:36, endln:48:37
                |vpiConstType:9
                |vpiDecompile:1
                |vpiSize:64
                |UINT:1
              |vpiTypespec:
              \_int_typespec: , line:37:14, endln:37:17, parent:TargetCfg.dio_pad_type
            |vpiLhs:
            \_parameter: (work@top.u_top.u_prim_pad_attr.u_submodule.u_impl_generic.PadTypeInGeneric), line:20:18, endln:20:38, parent:work@top.u_top.u_prim_pad_attr.u_submodule.u_impl_generic
```
If it is not possible in general, I will try handle getting the value of member selection in verilator.

alaindargelas, 2021-11-18:
I found one problem. Why there is 77 in that line?
Surelog/tests/ComplexParamOverload/ComplexParamOverload.log

Line 831 in 2a2d58d

 |vpiDecompile:77 

It's the value of the local parameter when doing binding. Simply ignore, it's like the default value of a parameter

alaindargelas, 2021-11-20:
Let me know when we can close this issue

RRozak, 2021-11-22:
I didn't noticed it before, but there is a problem with the file that I pasted in previous comment: https://github.com/chipsalliance/Surelog/issues/2199#issuecomment-972911093
Since the vpiExpr value is related to the first component of vpiHierPath, the typespec is wrong.
```
          |vpiParamAssign:
          \_param_assign: , line:20:18, endln:20:38, parent:work@top.u_top.u_prim_pad_attr.u_submodule.u_impl_generic
            |vpiOverriden:1
            |vpiRhs:
            \_hier_path: (TargetCfg.dio_pad_type), line:28:26, endln:28:33
              |vpiName:TargetCfg.dio_pad_type
              |vpiActual:
              \_ref_obj: (TargetCfg), parent:TargetCfg.dio_pad_type
                |vpiName:TargetCfg
              |vpiActual:
              \_ref_obj: (dio_pad_type), parent:TargetCfg.dio_pad_type
                |vpiName:dio_pad_type
              |vpiExpr:
              \_constant: , line:48:36, endln:48:37
                |vpiConstType:9
                |vpiDecompile:1
                |vpiSize:64
                |UINT:1
              |vpiTypespec:
              \_int_typespec: , line:37:14, endln:37:17, parent:TargetCfg.dio_pad_type
            |vpiLhs:
            \_parameter: (work@top.u_top.u_prim_pad_attr.u_submodule.u_impl_generic.PadTypeInGeneric), line:20:18, endln:20:38, parent:work@top.u_top.u_prim_pad_attr.u_submodule.u_impl_generic
```
The typespec is int, but should be struct. And since there is no information about the type of the struct and its members, we are unable to compute the value of the field.

RRozak, 2021-11-22:
This solution works on the other files.

alaindargelas, 2021-11-23:
Working on it

alaindargelas, 2021-11-24:
#2229 should fix this, please confirm.
BTW, the struct is struct1_t, I confirmed with VCS, the struct in the lower level masks the struct in the upper level of the hierarchy, so the hier_path addresses a member of the struct1_t.

RRozak, 2021-11-24:
I meant that it should have the `struct1_t` typespec. But it still has `int`.
Full source code:
```systemverilog
package pinmux_pkg;
   typedef enum int {
      A = 1
   } enum1_e;

   typedef struct packed {
      enum1_e dio_pad_type;
   } struct1_t;

   typedef enum int {
      B = 2
   } enum2_e;

   typedef struct packed {
      enum2_e dio_pad_type;
   } struct2_t;
endpackage : pinmux_pkg

module prim_generic_pad_attr(output int o);
   parameter int PadTypeInGeneric = 0;
   assign o = PadTypeInGeneric;
endmodule : prim_generic_pad_attr

module prim_submodule(output int o);
   parameter int PadType = 0;

   prim_generic_pad_attr #(
       .PadTypeInGeneric(PadType)
   ) u_impl_generic(
      .o(o)
   );
endmodule

module prim_pad_attr(output int o);
   parameter int PadType = 0;

   prim_submodule #(
      .PadType(PadType)
   ) u_submodule(
      .o(o)
   );
endmodule

module almost_top(output int o);
   import pinmux_pkg::*;
   parameter struct1_t TargetCfg = 1;

   prim_pad_attr #(
      .PadType(TargetCfg.dio_pad_type)
   ) u_prim_pad_attr(
      .o(o)
   );
endmodule

module top(output int o);
   import pinmux_pkg::*;
   parameter struct2_t TargetCfg = 2;

   almost_top

     u_top(
      .o(o)
   );
endmodule
```

full uhdm ( I mark with <----------------------- the problematic fields):
[... 673 lines cut ...]
          |vpiInstance:
          \_package: pinmux_pkg (pinmux_pkg::) top.sv:1:1: , endln:17:24, parent:work@top
      |vpiInstance:
      \_package: pinmux_pkg (pinmux_pkg::) top.sv:1:1: , endln:17:24, parent:work@top
  |vpiParamAssign:
  \_param_assign: , line:57:24, endln:57:37, parent:work@top
    |vpiRhs:
    \_constant: , line:57:36, endln:57:37
      |vpiConstType:9
      |vpiDecompile:2
      |vpiSize:64
      |UINT:2
    |vpiLhs:
    \_parameter: (work@top.TargetCfg), line:57:24, endln:57:37, parent:work@top
  |vpiTypedef:
  \_enum_typespec: (pinmux_pkg::enum1_e), line:4:6, endln:4:13
  |vpiTypedef:
  \_enum_typespec: (pinmux_pkg::enum2_e), line:12:6, endln:12:13
  |vpiTypedef:
  \_struct_typespec: (pinmux_pkg::struct1_t), line:6:12, endln:6:18, parent:work@almost_top.TargetCfg
  |vpiTypedef:
  \_struct_typespec: (pinmux_pkg::struct2_t), line:14:12, endln:14:18, parent:work@top.TargetCfg
  |vpiTypedef:
  \_import: (pinmux_pkg), line:56:11, endln:56:24
```

RRozak, 2021-11-24:
The problem isn't solved yet, but we've already bumped Surelog submodule. I remove the `blocker` label.

alaindargelas, 2021-11-25:
I was working off this test:
https://github.com/chipsalliance/Surelog/blob/master/tests/ComplexParamOverload/ComplexParamOverload.log
Which seems all fixed.

I'll check what is the difference with your test.

alaindargelas, 2021-11-25:
#2231 should take care of this test, please confirm.

RRozak, 2021-11-26:
Yes, it works. All tests from this issue passed.

closed 2021-11-26 by alaindargelas