chipsalliance/Surelog #2203: Division by zero on Msys2

hs-apotell, 2021-11-17:
Noticing some platform specific difference in regression tests. Here's one such scenario -

Test Name: DecValue
Build: https://github.com/chipsalliance/Surelog/actions/runs/1458231572
Platform: Msys2
Logs: Both Linux and Msys2 attached

Msys2 build is reporting the following error but none of the other platforms are reporting the same.
```
[ERR:EL0532] dut.sv:4: Division by zero in instance "work@top".
````

Also, the UHDM output also very different on Msys2 (which very likely could be an artifact of the above error, but worth an effort to confirm after fixing the above error).

alaindargelas, 2021-11-18:
The argument passing must not work correctly on Msys2:

cat DecValue.sl
-parse -d uhdm -d coveruhdm -elabuhdm -d ast dut.sv -nobuiltin -PA="\'d7_200" -PB="\'d1_800"

hs-apotell, 2021-11-19:
The backquote in the argument was the problem. Resolved in regression script.

closed 2021-11-19 by hs-apotell