chipsalliance/Surelog #2204: Is the order of input files on command line significant?

hs-apotell, 2021-11-17:
I am diagnosing a problem with differences in outputs across platforms and Msys2 output seems different in a number of test cases. Here's an example -

Build: https://github.com/chipsalliance/Surelog/actions/runs/1458231572
Test case: DiffSimpleIncludeAndMacros
Log files from both Linux and Msys2 attached

As far as the input goes, the only significant difference I could notice is the order of the input files. Is that significant? If not, these differences are unexpected and needs attention.

[msys2-gcc.zip](https://github.com/chipsalliance/Surelog/files/7552290/msys2-gcc.zip)
[linux-gcc.zip](https://github.com/chipsalliance/Surelog/files/7552291/linux-gcc.zip)

alaindargelas, 2021-11-17:
Yes, input file order is significant in Verilog, especially for macro definition.
This testcase is about macros, so if there is an input file order difference, that is what needs fixing (The test harness), not Surelog.

hs-apotell, 2021-11-17:
The input thru' the sl file is wildcard `top*.v`. The expansion of this is unpredictable. How are you expecting to enforce an order??

hs-apotell, 2021-11-17:
As per this [discussion](https://serverfault.com/questions/122737/in-bash-are-wildcard-expansions-guaranteed-to-be-in-order), bash (and other POSIX compliant shells) expands wildcard in alphabetically sorted order. I can enforce that for other platforms in the regression script to be in compliance. But is that always the _expected_ order?

alaindargelas, 2021-11-18:
This case is fixed by: #2630
Please pick an explicit order for the other tests and replace the * with that order.

closed 2021-11-18 by alaindargelas

hs-apotell, 2021-11-18:
I don't see a pull request nor an issue with that number. What's #2630 ?? Did you get the wrong reference?

alaindargelas, 2021-11-18:
Looks like it: #2208