chipsalliance/Surelog #2278: Casting to width given by sum with parameter doesn't work

RRozak, 2021-12-08:
Here is the code:
```systemverilog
module submodule #(
   parameter int P = 1
) (
   output logic [P + 1:0] a
);
   assign a = (P + 2)'(31);
endmodule // submodule

module top(output logic [2:0] o, output logic [4:0] p);
   submodule #(.P(1)) u_sub1 (.a(o));
   submodule #(.P(3)) u_sub3 (.a(p));
endmodule // top
```
Surelog substitutes `(P + 2)` by constant `3`, even in uhdmAllModules section. The constant `3` is proper only in case of first instance.
Here is uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@submodule (work@submodule) tests/CastToSumWithParameterIn2InstancesOfSubmodule/top.sv:1:1: , endln:7:10, parent:work@top
  |vpiDefName:work@submodule
  |vpiFullName:work@submodule
  |vpiPort:
  \_port: (a), line:4:27, parent:work@submodule
    |vpiName:a
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@submodule.a), line:4:27, parent:work@submodule
        |vpiName:a
        |vpiFullName:work@submodule.a
        |vpiNetType:36
  |vpiContAssign:
  \_cont_assign: , line:6:11, endln:6:27, parent:work@submodule
    |vpiRhs:
    \_operation: , line:6:15, endln:6:27
      |vpiOpType:67
      |vpiOperand:
      \_constant: , line:6:24, endln:6:26
        |vpiConstType:9
        |vpiDecompile:31
        |vpiSize:64
        |UINT:31
      |vpiTypespec:
      \_integer_typespec: , line:6:16, endln:6:21
        |INT:3                                                <-------------- Casting to width 3
    |vpiLhs:
    \_ref_obj: (work@submodule.a), line:6:11, endln:6:12
      |vpiName:a
      |vpiFullName:work@submodule.a
      |vpiActual:
      \_logic_net: (work@top.u_sub1.a), line:4:27, endln:4:28, parent:work@top.u_sub1
        |vpiName:a
        |vpiFullName:work@top.u_sub1.a
        |vpiNetType:36
        |vpiRange:
        \_range: , line:4:18, endln:4:25
          |vpiLeftRange:
          \_constant: , line:4:18, endln:4:19
            |vpiConstType:7
            |vpiDecompile:2
            |vpiSize:64
            |INT:2
          |vpiRightRange:
          \_constant: , line:4:24, endln:4:25
            |vpiConstType:9
[... 220 lines cut ...]
    |vpiParameter:
    \_parameter: (work@top.u_sub3.P), line:2:18, endln:2:23, parent:work@top.u_sub3
      |vpiName:P
      |vpiFullName:work@top.u_sub3.P
      |UINT:1
      |vpiTypespec:
      \_int_typespec: , line:2:14, endln:2:17, parent:work@top.u_sub3.P
    |vpiParamAssign:
    \_param_assign: , line:2:18, endln:2:23, parent:work@top.u_sub3
      |vpiOverriden:1
      |vpiRhs:
      \_constant: , line:2:22, endln:2:23
        |vpiConstType:9
        |vpiDecompile:3
        |vpiSize:64
        |UINT:3
        |vpiTypespec:
        \_int_typespec: , line:2:14, endln:2:17, parent:work@submodule.P
      |vpiLhs:
      \_parameter: (work@top.u_sub3.P), line:2:18, endln:2:23, parent:work@top.u_sub3
  |vpiNet:
  \_logic_net: (work@top.o), line:9:31, endln:9:32, parent:work@top
  |vpiNet:
  \_logic_net: (work@top.p), line:9:53, endln:9:54, parent:work@top
```

alaindargelas, 2021-12-10:
Fixed by #2289

RRozak, 2021-12-10:
The problem still occurs. The diff of uhdms before the fix and after is empty.

RRozak, 2021-12-10:
It works after the fix. I made a mistake when I was checking previously.
Thank you.

closed 2021-12-10 by RRozak