chipsalliance/Surelog #2287: Bit selects of multidimensional array are parsed wrong

RRozak, 2021-12-09:
Let's take this code as example:
```systemverilog
module top;
   typedef struct packed {
      int x;
   } struct_t;

   struct_t [1:0][2:0] a;
   assign a[0][0].x = 1;
endmodule
```
Currently, the uhdm of expression `a[0][0].x` is:
```
            \_hier_path: (a[0][0].x), line:7:11, endln:7:12
              |vpiName:a[0][0].x
              |vpiActual:
              \_bit_select: (a), parent:a[0][0].x
                |vpiName:a
                |vpiIndex:
                \_constant: , line:7:13, endln:7:14
                  |vpiConstType:9
                  |vpiDecompile:0
                  |vpiSize:64
                  |UINT:0
              |vpiActual:
              \_bit_select: (work@top.a[0][0].x), parent:a[0][0].x
                |vpiFullName:work@top.a[0][0].x
                |vpiIndex:
                \_constant: , line:7:16, endln:7:17
                  |vpiConstType:9
                  |vpiDecompile:0
                  |vpiSize:64
                  |UINT:0
              |vpiActual:
              \_ref_obj: (x), parent:a[0][0].x
                |vpiName:x
```
According to 37.17.13, the indices should be accessible by vpi_iterate.
Current uhdm is much harder to handle, for example we need to parse the vpiFullName.

alaindargelas, 2021-12-10:
This is where we don't want to follow VPI, VPI would require to do a full bit blasting.

RRozak, 2021-12-10:
Ok. Important thing for us is to know to what the index is related to.

Suppose that in the code above `a[0][0].x` had been changed to `a[0][0].x[0]`. First `[0]` should point to `a` (and currently it does), second `[0]` should point to `a` or `a[0]`, the third one should point to `x`.
Currently `a[0][0].x[0]` gives exactly the same uhdm as `a[0][0].x`. It is another error.

But if we remove the `x` from `a[0][0].x`, it is parsed as it is said in uhdm:
```
    \_var_select: (work@top.a), line:7:11, endln:7:12
      |vpiName:a
      |vpiFullName:work@top.a
      |vpiIndex:
      \_constant: , line:7:13, endln:7:14, parent:work@top.a
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
      |vpiIndex:
      \_constant: , line:7:16, endln:7:17, parent:work@top.a
        |vpiConstType:9
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
```
By the way, what is the difference between `bit_select` and `var_select`?

alaindargelas, 2021-12-11:
Fixed by #2292

alaindargelas, 2021-12-11:
var_select will typically have multiple selectors a[2:0][3] types (part select and bit select).
bit_select is of the simple kind.

closed 2021-12-11 by alaindargelas