chipsalliance/Surelog #2290: Value of parameter of parametrized type in submodule is wrong

RRozak, 2021-12-10:
Code:
```systemverilog
module submodule(output int a);
   parameter int X = 20;
   typedef logic [X-1:0] my_type_t;
   parameter my_type_t Y = '1;
   assign a = int'(Y);
endmodule

module top(output int o, output int p, output int r);
   submodule u_sub_default(.a(o));
   submodule #(.X(5)) u_sub_5(.a(p));
endmodule
```
The value of parameter Y (and output port) is proper only in `u_sub_default` instance. 
In other instance the value of `Y` is 1.

Below is uhdm. I marked the problem with `<------------------------- wrong value`
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@submodule (work@submodule) port.sv:1:1: , endln:6:10, parent:work@top
  |vpiDefName:work@submodule
  |vpiFullName:work@submodule
  |vpiPort:
  \_port: (a), line:1:29, parent:work@submodule
    |vpiName:a
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@submodule.a), line:1:29, parent:work@submodule
        |vpiName:a
        |vpiFullName:work@submodule.a
  |vpiContAssign:
  \_cont_assign: , line:5:11, endln:5:22, parent:work@submodule
    |vpiRhs:
    \_operation: , line:5:15, endln:5:22
      |vpiOpType:67
      |vpiOperand:
      \_ref_obj: (work@submodule.Y), line:5:20, endln:5:21
        |vpiName:Y
        |vpiFullName:work@submodule.Y
      |vpiTypespec:
      \_int_typespec: , line:5:15, endln:5:18
    |vpiLhs:
    \_ref_obj: (work@submodule.a), line:5:11, endln:5:12
      |vpiName:a
      |vpiFullName:work@submodule.a
      |vpiActual:
      \_int_var: (work@top.u_sub_default.a), line:1:29, endln:1:30, parent:work@top.u_sub_default
        |vpiName:a
        |vpiFullName:work@top.u_sub_default.a
        |vpiAutomatic:1
        |vpiVisibility:1
  |vpiNet:
  \_logic_net: (work@submodule.a), line:1:29, parent:work@submodule
  |vpiParameter:
  \_parameter: (work@submodule.X), line:2:18, endln:2:24, parent:work@submodule
    |vpiName:X
    |vpiFullName:work@submodule.X
    |UINT:20
    |vpiTypespec:
    \_int_typespec: , line:2:14, endln:2:17, parent:work@submodule.X
  |vpiParameter:
  \_parameter: (work@submodule.Y), line:4:24, endln:4:30, parent:work@submodule
    |vpiName:Y
    |vpiFullName:work@submodule.Y
[... 502 lines cut ...]
        \_int_typespec: , line:2:14, endln:2:17, parent:work@submodule.X
      |vpiLhs:
      \_parameter: (work@top.u_sub_5.X), line:2:18, endln:2:24, parent:work@top.u_sub_5
    |vpiParamAssign:
    \_param_assign: , line:4:24, endln:4:30, parent:work@top.u_sub_5
      |vpiOverriden:1
      |vpiRhs:
      \_constant: , line:4:28, endln:4:30
        |vpiConstType:9
        |vpiDecompile:1
        |vpiSize:5
        |UINT:1                                             <------------------------- wrong value
        |vpiTypespec:
        \_logic_typespec: (my_type_t), line:3:12, endln:3:17, parent:work@submodule.Y
      |vpiLhs:
      \_parameter: (work@top.u_sub_5.Y), line:4:24, endln:4:30, parent:work@top.u_sub_5
    |vpiTypedef:
    \_logic_typespec: (my_type_t), line:3:12, endln:3:17
  |vpiVariables:
  \_int_var: (work@top.o), line:8:23, endln:8:24, parent:work@top
  |vpiVariables:
  \_int_var: (work@top.p), line:8:37, endln:8:38, parent:work@top
  |vpiVariables:
  \_int_var: (work@top.r), line:8:51, endln:8:52, parent:work@top
```

alaindargelas, 2021-12-12:
Sorry, I don't see the problem, please be more explicit about what value you expect.
For the default
```
       \_ref_obj: (work@top.u_sub_default.YYY), id:98, line:5:20, endln:5:23, parID:97
          |vpiName:YYY
          |vpiFullName:work@top.u_sub_default.YYY
          |vpiActual:
          \_constant: , id:46, line:4:30, endln:4:32
            |vpiConstType:9
            |vpiDecompile:1
            |vpiSize:20   <<<<< Size 20
            |UINT:1     <<<< Value 1
```

For the overridden:
```
\_ref_obj: (work@top.u_sub_5.YYY), id:102, line:5:20, endln:5:23, parID:101
          |vpiName:YYY
          |vpiFullName:work@top.u_sub_5.YYY
          |vpiActual:
          \_constant: , id:58, line:4:30, endln:4:32
            |vpiConstType:9
            |vpiDecompile:1
            |vpiSize:5    <<<<< size 5
            |UINT:1       <<<<< value 1
            |vpiTypespec:
            \_logic_typespec: (my_type_t), id:16, line:3:12, endln:3:17, parent:work@submodule.YYY, parID:22
```

What am I missing?

RRozak, 2021-12-13:
The value should be `'1`, as it is in uhdmAllModules:
```
  |vpiParamAssign:
  \_param_assign: , line:4:24, endln:4:30, parent:work@submodule
    |vpiRhs:
    \_constant: , line:4:28, endln:4:30
      |vpiConstType:3
      |vpiDecompile:'1
      |vpiSize:20
      |BIN:1
      |vpiTypespec:
      \_logic_typespec: (my_type_t), line:3:12, endln:3:17, parent:work@submodule.Y
    |vpiLhs:
    \_parameter: (work@submodule.Y), line:4:24, endln:4:30, parent:work@submodule
```

alaindargelas, 2021-12-16:
Fixed by #2313.
Surelog now computes the right value my creating a value that is a mask 2^n -1 of the size of the lhs size in the case of unsized constant '1

closed 2021-12-16 by alaindargelas