chipsalliance/Surelog #2301: Casting in function is parsed wrong

RRozak, 2021-12-13:
Source:
```systemverilog
module top #(
   parameter int unsigned Depth = 3,
   parameter int unsigned PTR_WIDTH = 15
)(
   output logic [PTR_WIDTH-1:0] o
);
   assign o = get_casted_depth();

   function automatic [PTR_WIDTH-1:0] get_casted_depth();
      return (PTR_WIDTH)'(Depth);
   endfunction
endmodule
```
In the following example the cast operation casts to `int` instead of `PTR_WIDTH`.
Uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top) prim_fifo_async.sv:1:1: , endln:12:10, parent:work@top
  |vpiDefName:work@top
  |vpiFullName:work@top
  |vpiPort:
  \_port: (o), line:5:33, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: 
      |vpiActual:
      \_logic_net: (work@top.o), line:5:33, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiNetType:36
  |vpiContAssign:
  \_cont_assign: , line:7:11, endln:7:33, parent:work@top
    |vpiRhs:
    \_func_call: (get_casted_depth), line:7:15, endln:7:33
      |vpiName:get_casted_depth
      |vpiFunction:
      \_function: (work@top.get_casted_depth), line:9:4, endln:11:15, parent:work@top
    |vpiLhs:
    \_ref_obj: (work@top.o), line:7:11, endln:7:12
      |vpiName:o
      |vpiFullName:work@top.o
      |vpiActual:
      \_logic_net: (work@top.o), line:5:33, endln:5:34, parent:work@top
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiNetType:36
        |vpiRange:
        \_range: , line:5:18, endln:5:31
          |vpiLeftRange:
          \_constant: , line:5:18, endln:5:27
            |vpiConstType:7
            |vpiDecompile:14
            |vpiSize:64
            |INT:14
          |vpiRightRange:
          \_constant: , line:5:30, endln:5:31
            |vpiConstType:9
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
  |vpiTaskFunc:
  \_function: (work@top.get_casted_depth), line:9:4, endln:11:15, parent:work@top
    |vpiVisibility:1
    |vpiAutomatic:1
[... 233 lines cut ...]
  |vpiParamAssign:
  \_param_assign: , line:2:27, endln:2:36, parent:work@top
    |vpiRhs:
    \_constant: , line:2:35, endln:2:36
      |vpiConstType:9
      |vpiDecompile:3
      |vpiSize:32
      |UINT:3
      |vpiTypespec:
      \_int_typespec: , line:2:14, endln:2:17, parent:work@top.Depth
    |vpiLhs:
    \_parameter: (work@top.Depth), line:2:27, endln:2:36, parent:work@top
  |vpiParamAssign:
  \_param_assign: , line:3:27, endln:3:41, parent:work@top
    |vpiRhs:
    \_constant: , line:3:39, endln:3:41
      |vpiConstType:9
      |vpiDecompile:15
      |vpiSize:32
      |UINT:15
      |vpiTypespec:
      \_int_typespec: , line:3:14, endln:3:17, parent:work@top.PTR_WIDTH
    |vpiLhs:
    \_parameter: (work@top.PTR_WIDTH), line:3:27, endln:3:41, parent:work@top
```

alaindargelas, 2021-12-14:
Fixed by #2305

RRozak, 2021-12-14:
Here is the part of the current uhdm tree:
```
      \_operation: , line:10:14, endln:10:33
        |vpiOpType:67
        |vpiOperand:
        \_ref_obj: (work@top.get_casted_depth.Depth), line:10:27, endln:10:32
          |vpiName:Depth
          |vpiFullName:work@top.get_casted_depth.Depth
        |vpiTypespec:
        \_int_typespec: , line:3:14, endln:3:17, parent:work@top.PTR_WIDTH
          |vpiExpr:
          \_parameter: (work@top.PTR_WIDTH), line:3:27, endln:3:41, parent:work@top
            |vpiName:PTR_WIDTH
            |vpiFullName:work@top.PTR_WIDTH
            |UINT:15
            |vpiTypespec:
            \_int_typespec: , line:3:14, endln:3:17, parent:work@top.PTR_WIDTH
```
Could you change the parameter node under `vpiExpr` to `ref_obj`? The `vpiParameter` node is rather used to parameter declaration, not the reference.

alaindargelas, 2021-12-14:
Fixed by #2307

closed 2021-12-14 by alaindargelas