chipsalliance/Surelog #2321: surelog cannot find builtin.sv

pieter3d, 2021-12-17:
I recently updated surelog to the latest version from the repo (just today), previous pull was about 2 months ago.
The parsing calls I make in my code now result in errors like this:

`[FTL:CM0008] Verilog File "/home/pieter/tmp/../lib/surelog/sv/builtin.sv" does not exist.`

Seems like something has changed that is causing Surelog to not find the installed builtin.sv - clearly it's using a bad path.

I call the parser like this:
```C++
  SURELOG::SymbolTable symbol_table_;
  SURELOG::ErrorContainer errors(&symbol_table_);
  SURELOG::CommandLineParser clp(&errors, &symbol_table_);
  clp.noPython();
  clp.setParse(true);
  clp.setElaborate(true);
  clp.setElabUhdm(true);
  clp.setCompile(true);
  clp.setwritePpOutput(true);
  clp.setWriteUhdm(true);
  bool success = clp.parseCommandLine(argc, argv); // returns false!
```

alaindargelas, 2021-12-18:
Remind me, you are doing a make install to some place not standard, then you are creating your own main and linking with libsurelog.so?

I think the problem is more around the setup you have.

Can you list where sv/builtin.sv gets actually installed?

alaindargelas, 2021-12-18:
BTW, this file has caused enough grief, that I'm thinking of embedding it in the C++ code.

pieter3d, 2021-12-18:
+1 to embedding it.

I installed with a simple `sudo make install`, no custom prefix or anything. vanilla build and install.
I found the file here : `/usr/local/lib/surelog/sv/builtin.sv`
Note also that `hellouhdm` in the build directory runs just fine, no complaints there.

alaindargelas, 2021-12-18:
Please set a BP here and see what happens in your exec:
https://github.com/chipsalliance/Surelog/blob/ffbbebdb8b3f48539a02f4c80019fe39e2c6339e/src/CommandLine/CommandLineParser.cpp#L509

pieter3d, 2021-12-18:
It searched:
`<pwd>/sv/builtin.sv`
`<pwd>/../lib/surelog/sv/builtin.sv`
where `<pwd>` is the directory I launched from.

alaindargelas, 2021-12-22:
I'm back on this

alaindargelas, 2021-12-22:
@pieter3d can you provide the smallest main.cpp that reproduce your issue?
I'm wondering if you are tempering with the argc, argv.

pieter3d, 2021-12-22:
Sure, seems super easy to repro this.
```c++
#include <surelog/surelog.h>
#include <cstdio>

int main(int argc, const char *argv[]) {
  SURELOG::SymbolTable symbol_table;
  SURELOG::ErrorContainer errors(&symbol_table);
  SURELOG::CommandLineParser clp(&errors, &symbol_table);
  clp.noPython();
  clp.setParse(true);
  clp.setElaborate(true);
  clp.setElabUhdm(true);
  clp.setCompile(true);
  clp.setwritePpOutput(true);
  clp.setWriteUhdm(true);
  bool success = clp.parseCommandLine(argc, argv);
  if (!success || clp.help()) {
    if (!clp.help()) {
      std::cout << "Problems parsing arguments." << std::endl;
    }
    return -1;
  }

  std::cout << "Success";
  return 0;
}
```

Build with:
`g++ -std=c++17 -I/usr/local/include -I/usr/local/include/surelog ../src/test.cc /usr/local/lib/surelog/libsurelog.a -o test`

Output (with or without valid surelog argv contents):
```
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[FTL:CM0008] Verilog File "/home/pieter/work/pieter3d/simview/build/../lib/surelog/sv/builtin.sv" does not exist.

Problems parsing arguments.
```

alaindargelas, 2021-12-27:
#2352 should fix this.
Please try it out.

pieter3d, 2021-12-27:
That works now - thanks!

pieter3d, 2021-12-27:
Out of curiosity, what caused the issue in the first place?

closed 2021-12-27 by pieter3d

alaindargelas, 2021-12-28:
No idea, there were several changes with path separators to support Windows, maybe one was botched. I did not try to fix the path issue itself, instead I embedded the file in the c++ code.