chipsalliance/Surelog #2336: Missing `default_nettype compiler directive

kamilrakoczy, 2021-12-21:
TEST:
```
module ok;
reg a;
assign b=a;
endmodule

`default_nettype none

module bad;
reg a;
assign b=a;
endmodule
```
According to ``22.8 `default_nettype``:
```
The directive `default_nettype controls the net type created for implicit net declarations (see 6.10). It
can be used only outside design elements. Multiple `default_nettype directives are allowed. The latest
occurrence of this directive in the source controls the type of nets that will be implicitly declared.
Syntax 22-7 contains the syntax of the directive.
```

There isn't any information about ``default_nettype`` in generated UHDM.

alaindargelas, 2022-01-07:
It cannot be in UHDM as an explicit "member", as it is not in the Verilog Object model.
On the other hand, it should influence the net type of implicit nets during UHDM creation.

alaindargelas, 2022-01-12:
Fixed with #2424

closed 2022-01-12 by alaindargelas