chipsalliance/Surelog #2353: Please allow to use external dependencies instead of bundling them

yurivict, 2021-12-27:
These are: antlr4, UHDM, flatbuffers, googletest.

alaindargelas, 2021-12-30:
Can you motivate your ask? What problem are you facing?
The only optional is antlr4, you can switch to the official antlr4 (See CMakeLists.txt). 

But it looks like you should use the pre-packaged Surelog here that does work as a front-end to Yosys: 
https://antmicro.com/blog/2021/10/openlane-asic-build-flow-with-systemverliog-support/

yurivict, 2021-12-30:
Stable dependencies like ```flatbuffers``` should be used through external packages for these reasons:
1. In case security vulnerabilities would be later found in them - they should be fixed in one place, not in many bundled copies.
2. Maintenance costs are lower when there's no bundling - compilation times are shorter, no need to update hashes/versions of bundled packages with each version update.
3. In case the package requires/would require patches - they would be done in one place, vs. in all bundled copies.
4. Bundling wastes CPU resources - compilation time grows exponentially with bundling because same software is recompiled again and again.

alaindargelas, 2021-12-31:
Can you submit a pull request?

hzeller, 2023-03-15:
Fixed with #3498

closed 2023-03-15 by hzeller