chipsalliance/Surelog #2364: Investigate use of precompiled headers

hs-apotell, 2021-12-31:
The current build times for UHDM on windows & msys2 platform are over 6x compared to the same on unix & osx. Use of precompiled headers got them closer (a lot closer) but it negatively impacted Surelog performance.

Needs further investigation as to why compilation step would impact performance, especially on unix.

alaindargelas, 2021-12-31:
Actually, I don't see a negative performance anymore on Linux.

hs-apotell, 2021-12-31:
The precompiled header changes was merged and reverted again. It's not part of the current master.

Change that introduced it - https://github.com/chipsalliance/UHDM/pull/550
Change that reverted it again - https://github.com/chipsalliance/UHDM/pull/552

If you can, I would recommend pulling that PR into your local environment and evaluating.