chipsalliance/Surelog #2373: Breaks with Python option enabled: error: no matching function for call to 'evalScript'

yurivict, 2022-01-01:
```
/usr/bin/c++ -DPLI_DLLESPEC="" -DPLI_DLLISPEC="" -DSURELOG_WITH_PYTHON -I/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src -I/disk-samsung/freebsd-ports/cad/surelog/work/.build/generated/src -I/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/third_party/antlr4_fast/runtime/Cpp/runtime/src -I/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/third_party/flatbuffers/include -I/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/third_party/googletest/googletest/include -I/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/third_party/googletest/googlemock/include -I/usr/local/include/python3.8 -I/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/third_party/UHDM/third_party/capnproto/c++/src/capnp/.. -I/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/third_party/UHDM/third_party/capnproto/c++/src/kj/.. -isystem /disk-samsung/freebsd-ports/cad/surelog/work/.build/third_party/UHDM/generated -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -fno-omit-frame-pointer  -isystem /usr/local/include -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free  -O2 -pipe -fno-omit-frame-pointer -fstack-protector-strong -isystem /usr/local/include -fno-strict-aliasing -fno-omit-frame-pointer  -isystem /usr/local/include -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -Wall -O3  -DNDEBUG  -std=gnu++17 -MD -MT CMakeFiles/surelog.dir/src/ErrorReporting/ErrorContainer.cpp.o -MF CMakeFiles/surelog.dir/src/ErrorReporting/ErrorContainer.cpp.o.d -o CMakeFiles/surelog.dir/src/ErrorReporting/ErrorContainer.cpp.o -c /disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/ErrorReporting/ErrorContainer.cpp
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/ErrorReporting/ErrorContainer.cpp:254:15: error: no matching function for call to 'evalScript'
        tmp = PythonAPI::evalScript("__main__", "SLformatMsg", args,
              ^~~~~~~~~~~~~~~~~~~~~
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/API/PythonAPI.h:57:22: note: candidate function not viable: cannot convert argument of incomplete type 'void *const' to 'PyThreadState *' (aka '_ts *') for 4th argument
  static std::string evalScript(std::string module, std::string function,
                     ^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/API/PythonAPI.h:60:15: note: candidate function not viable: requires 3 arguments, but 4 were provided
  static void evalScript(std::string function, SV3_1aPythonListener* listener,
              ^
/disk-samsung/freebsd-ports/cad/surelog/work/Surelog-0.0-3950-g88ae53316/src/API/PythonAPI.h:70:15: note: candidate function not viable: requires 2 arguments, but 4 were provided
  static bool evalScript(std::string script, Design* design);
              ^
```

clang-12
python-3.8

alaindargelas, 2022-01-02:
Pretty much everybody has ran away from the Python API of Surelog, it is not maintained.
Why do you want to use it? 
UHDM is a far superior API.

yurivict, 2022-01-02:
> Why do you want to use it?

I didn't know that ```Pretty much everybody has ran away from the Python API of Surelog, it is not maintained.```

alaindargelas, 2022-01-02:
Please use the UHDM API.

closed 2022-01-02 by alaindargelas

alaindargelas, 2022-02-10:
FYI, I fixed a compilation issue in Python since there was another inquiry.
Maybe that fix your issue too.