chipsalliance/Surelog #2388: No fields of vpiTypedef

RRozak, 2022-01-04:
There is no body of the `vpiTypedef` nodes in the uhdm of the following example:
```systemverilog
package flash_ctrl_pkg;
   typedef struct packed {
      logic phase;
   } data_region_attr_t;

   parameter data_region_attr_t HwDataAttr[1] = '{
      '{
         phase: 1'b1
      }
   };
endpackage : flash_ctrl_pkg

module flash_mp_data_region_sel import flash_ctrl_pkg::*; (
   input data_region_attr_t region_attrs_i [1],
   output logic x
);
   assign x = region_attrs_i[0].phase;
endmodule

module top(output logic o);
   import flash_ctrl_pkg::*; 
   flash_mp_data_region_sel u_hw_sel (
      .region_attrs_i(HwDataAttr),
      .x(o)
   );
endmodule
```
Uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallPackages:
\_package: flash_ctrl_pkg (flash_ctrl_pkg::) top.sv:1:1: , endln:11:28, parent:work@top
  |vpiName:flash_ctrl_pkg
  |vpiFullName:flash_ctrl_pkg::
  |vpiParameter:
  \_parameter: (flash_ctrl_pkg::HwDataAttr), line:6:33, endln:10:5, parent:flash_ctrl_pkg::
    |vpiSize:1
    |vpiTypespec:
    \_array_typespec: 
      |vpiRange:
      \_range: , line:6:44, endln:6:45, parent:flash_ctrl_pkg::HwDataAttr
        |vpiLeftRange:
        \_constant: , line:6:44, endln:6:45
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
        |vpiRightRange:
        \_operation: 
          |vpiOpType:11
          |vpiOperand:
          \_constant: , line:6:44, endln:6:45
            |vpiDecompile:1
            |vpiSize:64
            |UINT:1
            |vpiConstType:9
          |vpiOperand:
          \_constant: 
            |vpiSize:64
            |INT:1
            |vpiConstType:7
      |vpiElemTypespec:
      \_struct_typespec: (flash_ctrl_pkg::data_region_attr_t), line:2:12, endln:2:18, parent:flash_ctrl_pkg::HwDataAttr
        |vpiName:flash_ctrl_pkg::data_region_attr_t
        |vpiInstance:
        \_package: flash_ctrl_pkg (flash_ctrl_pkg::) top.sv:1:1: , endln:11:28, parent:work@top
[... 859 lines cut ...]
      |vpiLowConn:
      \_ref_obj: (work@top.u_hw_sel.region_attrs_i)
        |vpiName:region_attrs_i
        |vpiFullName:work@top.u_hw_sel.region_attrs_i
        |vpiActual:
        \_struct_net: (work@top.u_hw_sel.region_attrs_i), line:14:29, endln:14:43, parent:work@top.u_hw_sel.region_attrs_i
      |vpiTypedef:
      \_struct_typespec: (flash_ctrl_pkg::data_region_attr_t), line:2:12, endln:2:18, parent:flash_ctrl_pkg::HwDataAttr
    |vpiPort:
    \_port: (x), line:15:17, endln:15:18, parent:work@top.u_hw_sel
      |vpiName:x
      |vpiDirection:2
      |vpiHighConn:
      \_ref_obj: (work@top.o), line:24:10, endln:24:11
        |vpiName:o
        |vpiFullName:work@top.o
        |vpiActual:
        \_logic_net: (work@top.o), line:20:25, endln:20:26, parent:work@top
      |vpiLowConn:
      \_ref_obj: (work@top.u_hw_sel.x)
        |vpiName:x
        |vpiFullName:work@top.u_hw_sel.x
        |vpiActual:
        \_logic_net: (work@top.u_hw_sel.x), line:15:17, endln:15:18, parent:work@top.u_hw_sel
```

alaindargelas, 2022-01-05:
Can you be more specific as of where you can't see the typespec?
I can see it here:

```
    |vpiPort:
    \_port: (region_attrs_i), id:153, line:14:29, endln:14:43, parent:work@top.u_hw_sel, parID:237
      |vpiName:region_attrs_i
      |vpiDirection:1
      |vpiHighConn:
      \_operation: , id:200, line:23:23, endln:23:33
        |vpiTypespec:
        \_struct_typespec: (flash_ctrl_pkg::data_region_attr_t), id:201, line:2:12, endln:2:18, parID:200
          |vpiName:flash_ctrl_pkg::data_region_attr_t
          |vpiInstance:
          \_package: flash_ctrl_pkg (flash_ctrl_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:11:28, parent:work@top, parID:209
          |vpiPacked:1
          |vpiTypespecMember:
          \_typespec_member: (phase), id:202, line:3:13, endln:3:18, parent:flash_ctrl_pkg::data_region_attr_t, parID:201
            |vpiName:phase
            |vpiTypespec:
            \_logic_typespec: , id:203, line:3:7, endln:3:12, parent:phase, parID:202
              |vpiInstance:
              \_package: flash_ctrl_pkg (flash_ctrl_pkg::), id:0 tests/ScratchPad.sv:1:1: , endln:11:28, parent:work@top, parID:209
        |vpiOpType:75
        |vpiOperand:
        \_operation: , id:204, line:7:7, endln:9:8, parID:200
          |vpiOpType:75
          |vpiOperand:
          \_tagged_pattern: , id:205, parID:204
            |vpiPattern:
            \_constant: , id:207, line:8:17, endln:8:21, parID:205
              |vpiDecompile:1'b1
              |vpiSize:1
              |BIN:1
              |vpiConstType:3
            |vpiTypespec:
            \_string_typespec: (phase), id:206, line:8:10, endln:8:15, parID:205
              |vpiName:phase
```

RRozak, 2022-01-05:
This issue is about `vpiTypedef,` not `vpiTypespec`.
In all vpiTypedef nodes there is no field `vpiName` or any type information.
This is one of examples:
```
      |vpiTypedef:
      \_struct_typespec: (flash_ctrl_pkg::data_region_attr_t), line:2:12, endln:2:18, parent:flash_ctrl_pkg::HwDataAttr
```

alaindargelas, 2022-01-06:
That is just because the printing order has changed.
```
       \_operation: , id:200, line:28:20, endln:28:30
        |vpiTypespec:
        \_struct_typespec: (flash_ctrl_pkg::data_region_attr_t), id:201, line:3:9, endln:3:15, parID:200
          |vpiName:flash_ctrl_pkg::data_region_attr_t
          |vpiInstance:
          \_package: flash_ctrl_pkg (flash_ctrl_pkg::), id:0 tests/ScratchPad.sv:2:1: , endln:15:28, parent:work@top, parID:209
          |vpiPacked:1
          |vpiTypespecMember:
          \_typespec_member: (phase), id:202, line:4:10, endln:4:15, parent:flash_ctrl_pkg::data_region_attr_t, parID:201
            |vpiName:phase
            |vpiTypespec:
            \_logic_typespec: , id:203, line:4:4, endln:4:9, parent:phase, parID:202
              |vpiInstance:
              \_package: flash_ctrl_pkg (flash_ctrl_pkg::), id:0 tests/ScratchPad.sv:2:1: , endln:15:28, parent:work@top, parID:209
```

alaindargelas, 2022-01-06:
The information is present, it does not show in the same place since we added typespec infor for parameter RHS. What are you basing your fact that there is no information? From the text output of the vpi_visitor? Or from accessing the data programatically?

alaindargelas, 2022-01-06:
@RRozak, please comment.

alaindargelas, 2022-01-12:
@rkapuscik, please reopen if this is still an issue

closed 2022-01-12 by alaindargelas

RRozak, 2022-01-12:
You're right. The bug was in Verilator. It handled that typedef like it was empty. When I saw that it is also empty on output of uhdm-dump, I thought that it is a problem on Surelog side and created this issue.