chipsalliance/Surelog #2406: Inaccurate column information and missing end location information in error messages

hs-apotell, 2022-01-07:
In most of the cases in preprocess, parse and elaboration, the reported error messages are missing the column information. They seem to be defaulted to zero in most cases. Here are a few such cases -

* https://github.com/chipsalliance/Surelog/blob/136d59f1baa2857accb0d39171bf950474780687/src/Design/Design.cpp#L128
* https://github.com/chipsalliance/Surelog/blob/136d59f1baa2857accb0d39171bf950474780687/src/Design/Design.cpp#L366
* https://github.com/chipsalliance/Surelog/blob/136d59f1baa2857accb0d39171bf950474780687/src/Design/FileContent.cpp#L98
* https://github.com/chipsalliance/Surelog/blob/136d59f1baa2857accb0d39171bf950474780687/src/DesignCompile/CompileClass.cpp#L88

Also, `Location` should include end location (both line and column).

Accurate location information is critical for language servers and linters to work correctly.

alaindargelas, 2022-01-07:
Please contribute PRs, I don't have time to work on this.

alaindargelas, 2022-05-25:
That seems to be largely fixed by #2963

closed 2022-05-25 by alaindargelas