chipsalliance/Surelog #2417: "Current Working Directory" conundrum

hs-apotell, 2022-01-10:
Surelog is highly sensitive to the current working directory. When compiled as part of a larger application alongside other libraries this restriction poses a lot of challenges since libraries expect the current working directory to be the location of the executable (which is usually the case for most applications) and not the location of the files being loaded (which is the case for Surelog).

Another restriction along the same lines is that Surelog expects all source files to be under a single root (since all paths are resolved based on the current working directory). This restricts copy-on-write (i.e. most of the design is to be treated read-only, possibly on a read-only network drive and copy the files being modified locally) like features for tools. A feature like this would require multiple distinct roots.

Bringing this to the table to brainstorm ideas on what the compromise solution would look like where these restrictions can be relaxed and multiple libraries can co-exist happily.

One potential solution -
* Accept a repeatable argument to select the current directory whose value can be used to resolve all following arguments on the command line).
* Wrap all location references in an internal class implementation (to track both relative and absolute paths) and resolve the paths once at the time of parsing command line arguments.

Appreciate any other suggestions that might help!!

alaindargelas, 2022-01-10:
No matter the solution proposed, it shall be fully in line with the SystemVerilog and Verilog standards, so users don't start creating projects that only Surelog can read and other tools can't. That is the purpose of a Standard.
Surelog uses the Verilog command line Standard where file locations are strictly relative to a single root or absolute paths.
There is no such a thing to my knowledge of multiple roots for relative paths.
Maybe the only thing we should add would be a **single** --work <path> option that points to the working directory of choice.

alaindargelas, 2022-01-10:
One has to look at the **Library** path specification from SystemVerilog in the Standard to make sure the solution is also compliant to the Library concepts.

hs-apotell, 2022-01-10:
> Maybe the only thing we should add would be a single --work option that points to the working directory of choice.

My proposed solution is a step further. Instead of a single `work` option, add multiple and for each such path track the corresponding `work` value.

> so users don't start creating projects that only Surelog can read and other tools can't. That is the purpose of a Standard.

I do not share this understanding of **Standard Compliance**. By your definition, Surelog is already fails because it supports `unsigned int`.

Standard dictates _minimum_ requirements and does not dictate how any specific requirement is implemented. It doesn't restrict enhancing the feature as long as the minimum requirements are met. The solution I proposed or the requirement I laid out is no way violating the dictated minimum requirements to meet the standard compliance definition. Solution still treats the current working directory as the default if none is provided (or `cd` value if provided) and all relative paths are treated relative to this directory. That meets the minimum requirements of the standard, in my opinion.

> There is no such a thing to my knowledge of multiple roots for relative paths.

This is the enhancement I am proposing, of course, without violating the standard. The enhancement is that the user can provide paths which are relative to some other previously provided base. AFAIK the standard doesn't dictate that a command that's compatible with Surelog has to be compatible with some other similar library/tool as well. If that were the case, every compiler would be non-compliant because they implement features that are not yet part of the standard and may not compile/link with some other compiler.

All that said, the crux of the problem is that the library takes **ownership** of the current working directory and modifies it in ways that's not always compatible with the executable it is linked as part of. Calls to `chdir` should be strictly prohibited. In some ways, this is a follow up of problems that we dealt with in handling filepaths where calls to collect files (and other similar `FileUtils` API) under a directory would result in a number of calls to `chdir` to walk into each subdirectory to scan for files. Instead the library needs to track the _base_ alongside the input relative path. For the default case of single root (which the standard requires), the _base_ will always be the chosen working directory but in the case of multiple  roots, there can be correspondingly multiple _base_ values and each path will know its relevant _base_.

alaindargelas, 2022-01-10:
I think you are proposing something which seems to be an alternative to this:

```
IEEE Std 1800-2017 IEEE Standard for SystemVerilog—Unified Hardware Design, Specification, and Verification Language
Page: 902
33. Configuring the contents of a design
33.1 General
This clause describes the following:
— Configuration libraries
— Configuration syntax
— Using libraries and configurations
```

Be able to pick files from diverse project roots. That seems what pre-compiled libraries were developed for in SV (Also to match a VHDL feature). Save parse/compilation time, and reuse.

Can you read the whole section 33 of the Standard and see if that would fit your requirements, I'd rather we spend time implement something in the Standard than something tool specific.

Note that independently having a -work to point each independent library compilation to an alternative path without doing a chdir is a good idea.

hs-apotell, 2022-01-14:
Okay! I read thru' the section you mentioned and it primarily talks about libraries, library map files and configuration files. It doesn't concern with source files at all. The specific issue I am having to deal with is with source files themselves. I would like to support multiple work directories (or base or root directory) within the same command so when resolving include paths and alike, they aren't rooted to merely one base.

In short, as of the current implementation, include/library/alike paths are relative to the current working directory (or absolute) and I would like to parameterize this so every such path can either be absolute or relative to the preceding work parameter that is not necessarily shared across all such paths.

> Note that independently having a -work to point each independent library compilation to an alternative path without doing a chdir is a good idea.

This will be part of the goal but it's tricky given they way how Surelog tracks paths as symbols in `SymbolTable`. That can only be one path which will have to be absolute if done this way - not ideal because this approach ends up with non-portable cache with absolute paths embedded in binaries. Plus all paths will have to be resolved very early in the process which may not always be feasible. Use of absolute paths in general would create more issues than solve any, if at all.

The other solution was to not use `SymbolTable` for paths at all and create a custom class to wrap _prefix_ and _suffix_ paths (which when combined gives the absolute). This approach will work well but ends up creating copies of the `fs::path` object since every use of file path (using `SymbolId`) will have to be swapped for this wrapper class. Doable but a rather big change.

Yet another approach was to create the above mentioned wrapper class with `SymbolTable` being the owner of such objects. What's handed out by the `SymbolTable` during registration is, say, a _PathId_ and that is what is shared in place of the current `SymbolId`. This approach seems most promising. Welcome your thoughts and input on this.

I am still trying to figure out what would be a good solution to deal with this scenario, I would welcome any suggestions you might have.

alaindargelas, 2022-01-18:
Example:
surelog  -work /home1 top.v mid.v -work /home2 bottom.v

You could create the following Symbols when parsing the command line (Literally using extra text @WORK, @FILE): 
```
ID SYMBOL
1  @WORK:/home1@FILE:top.v 
2  @WORK:/home1@FILE:mid.v
3  @WORK:/home2@FILE:bottom.v
```
Whenever in the code you need to read the file or output the file path in a message, you'd do a simple regexp to process the file name. You would not have to change any datastructures...

hs-apotell, 2022-01-20:
@alaindargelas Your proposed solution have a few problems.

* BaseClass::VpiFile would a problem since that function will now be returning an encoded path. This is public API and I would prefer not to expose this out as part of public API. This is internal to how UHDM/Surelog handles path and the external clients don't have to deal with the intricacies.

* It does not allow assigning a symbol pulled from the `SymbolTable` directly to an `fs::path` instance. i.e.
```const fs::path filepath = m_symbolTable->getSymbol(symbolId);``` wouldn't work because the symbol is now an encoded one. It has to first be decoded before use.

So, I started making some progress on introducing a new class `Path` that wraps both the prefix and the suffix path i.e. the work directory and the filepath, by your terminology. The instances of `Path` are managed by the `SymbolTable`. So far looking like a promising solution.

alaindargelas, 2022-01-20:
On the second point, that is what I described, a simple function would have to be inserted to decide. I think that solution was the least disruptive and definitely not changing anything for UHDM users.