chipsalliance/Surelog #2426: Wrong vpiTypespec of port of type struct array

RRozak, 2022-01-12:
When the following file is parsed:
```systemverilog
package flash_ctrl_pkg;
   typedef struct packed {
      logic phase;
   } data_region_attr_t;

   parameter data_region_attr_t HwDataAttr[1] = '{
      '{
         phase: 1'b1
      }
   };
endpackage : flash_ctrl_pkg

module flash_mp_data_region_sel import flash_ctrl_pkg::*; (
   input data_region_attr_t region_attrs_i [1],
   output logic x
);
   assign x = region_attrs_i[0].phase;
endmodule

module top(output logic o);
   import flash_ctrl_pkg::*; 
   flash_mp_data_region_sel u_hw_sel (
      .region_attrs_i(HwDataAttr),
      .x(o)
   );
endmodule
```
The `vpiTypespec` of `region_attrs_i` is `struct_typespec`, but should be the array of struct.
Part of uhdm:
```
    |vpiPort:
    \_port: (region_attrs_i), line:14:29, endln:14:43, parent:work@top.u_hw_sel
      |vpiName:region_attrs_i
      |vpiDirection:1
      |vpiHighConn:
      \_operation: , line:23:23, endln:23:33
        |vpiTypespec:
        \_struct_typespec: (flash_ctrl_pkg::data_region_attr_t), line:2:12, endln:2:18
          |vpiName:flash_ctrl_pkg::data_region_attr_t
          |vpiInstance:
          \_package: flash_ctrl_pkg (flash_ctrl_pkg::) tests/StructArrayParameterInitializedByPatternPassedAsPort/top.sv:1:1: , endln:11:28, parent:work@top
          |vpiPacked:1
          |vpiTypespecMember:
          \_typespec_member: (phase), line:3:13, endln:3:18, parent:flash_ctrl_pkg::data_region_attr_t
            |vpiName:phase
            |vpiTypespec:
            \_logic_typespec: , line:3:7, endln:3:12, parent:phase
              |vpiInstance:
              \_package: flash_ctrl_pkg (flash_ctrl_pkg::) tests/StructArrayParameterInitializedByPatternPassedAsPort/top.sv:1:1: , endln:11:28, parent:work@top
        |vpiOpType:75
        |vpiOperand:
        \_operation: , line:7:7, endln:9:8
          |vpiOpType:75
          |vpiOperand:
          \_tagged_pattern: 
            |vpiPattern:
            \_constant: , line:8:17, endln:8:21
              |vpiDecompile:1'b1
              |vpiSize:1
              |BIN:1
              |vpiConstType:3
            |vpiTypespec:
            \_string_typespec: (phase), line:8:10, endln:8:15
              |vpiName:phase
```

alaindargelas, 2022-01-13:
It is not:
```
    \_logic_net: (x), id:240
      |vpiName:x
      |vpiNetType:1
    |vpiArrayNet:                   <<<< It is an array_net with a sub type struct_net
    \_array_net: (work@top.u_hw_sel.region_attrs_i), id:192, parent:work@top.u_hw_sel, parID:224
```

alaindargelas, 2022-01-13:
The piece of the tree you are showing is not relevant to the type of the net.
It is showing the parameter assigned to the port. The parameter RHS operation typespec is of type struct_typespec, that could be argued it should be of type array_typespec

RRozak, 2022-01-13:
Ok, it is my mistake. I meant the typespec of the operation.
The port, as well as the parameter, has the type of the array.

alaindargelas, 2022-01-14:
Fixed by #2430

closed 2022-01-14 by alaindargelas