chipsalliance/Surelog #2428: Typedef statements sometimes result in variable creation

RRozak, 2022-01-13:
When the following file is parsed by Surelog:
```systemverilog
package spi_device_pkg;
   parameter int unsigned SramAw = 10;
   typedef logic [SramAw-1:0] sram_addr_t;
endpackage // spi_device_pkg

module top(output int o);
   import spi_device_pkg::*;

   assign o = $bits(sram_addr_t);
endmodule
```
The vpiNet of name `sram_addr_t` is created.
Here is the uhdm:
```
slpp_all/surelog.uhdm: Restored design Pre-Elab: 
design: (work@top)
|vpiName:work@top
|uhdmallPackages:
\_package: spi_device_pkg (spi_device_pkg::) spid_readsram.sv:1:1: , endln:4:11, parent:work@top
  |vpiName:spi_device_pkg
  |vpiFullName:spi_device_pkg::
  |vpiParameter:
  \_parameter: (spi_device_pkg::SramAw), line:2:27, endln:2:38, parent:spi_device_pkg::
    |UINT:10
    |vpiTypespec:
    \_int_typespec: , line:2:14, endln:2:17, parent:spi_device_pkg::SramAw
      |vpiInstance:
      \_package: spi_device_pkg (spi_device_pkg::) spid_readsram.sv:1:1: , endln:4:11, parent:work@top
    |vpiName:SramAw
    |vpiFullName:spi_device_pkg::SramAw
  |vpiParamAssign:
  \_param_assign: , line:2:27, endln:2:38, parent:spi_device_pkg::
    |vpiRhs:
    \_constant: , line:2:36, endln:2:38
      |vpiDecompile:10
      |vpiSize:32
      |UINT:10
      |vpiTypespec:
      \_int_typespec: , line:2:14, endln:2:17, parent:spi_device_pkg::SramAw
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (spi_device_pkg::SramAw), line:2:27, endln:2:38, parent:spi_device_pkg::
  |vpiTypedef:
  \_logic_typespec: (spi_device_pkg::sram_addr_t), line:3:12, endln:3:17
    |vpiName:spi_device_pkg::sram_addr_t
    |vpiInstance:
    \_package: spi_device_pkg (spi_device_pkg::) spid_readsram.sv:1:1: , endln:4:11, parent:work@top
    |vpiRange:
    \_range: , line:3:19, endln:3:29
      |vpiLeftRange:
      \_operation: , line:3:19, endln:3:25
        |vpiOpType:11
        |vpiOperand:
        \_constant: , line:3:19, endln:3:25
          |vpiDecompile:10
          |vpiSize:32
          |UINT:10
          |vpiTypespec:
          \_int_typespec: , line:2:14, endln:2:17
            |vpiInstance:
            \_package: spi_device_pkg (spi_device_pkg::) spid_readsram.sv:1:1: , endln:4:11, parent:work@top
          |vpiConstType:9
        |vpiOperand:
        \_constant: , line:3:26, endln:3:27
          |vpiDecompile:1
          |vpiSize:64
          |UINT:1
          |vpiConstType:9
      |vpiRightRange:
      \_constant: , line:3:28, endln:3:29
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
        |vpiConstType:9
  |vpiDefName:spi_device_pkg
|uhdmallModules:
...
|uhdmtopModules:
\_module: work@top (work@top) spid_readsram.sv:6:1: , endln:10:10
  |vpiName:work@top
  |vpiVariables:
  \_int_var: (work@top.o), line:6:23, endln:6:24, parent:work@top
    |vpiName:o
    |vpiFullName:work@top.o
    |vpiAutomatic:1
    |vpiVisibility:1
  |vpiParameter:
  \_parameter: (work@top.SramAw), line:2:27, endln:2:38, parent:work@top
    |UINT:10
    |vpiTypespec:
    \_int_typespec: , line:2:14, endln:2:17, parent:work@top.SramAw
      |vpiInstance:
      \_package: spi_device_pkg (spi_device_pkg::) spid_readsram.sv:1:1: , endln:4:11, parent:work@top
        |vpiName:spi_device_pkg
        |vpiFullName:spi_device_pkg::
        |vpiParameter:
        \_parameter: (spi_device_pkg::SramAw), line:2:27, endln:2:38, parent:spi_device_pkg::
          |UINT:10
          |vpiTypespec:
          \_int_typespec: , line:2:14, endln:2:17, parent:spi_device_pkg::SramAw
            |vpiInstance:
            \_package: spi_device_pkg (spi_device_pkg::) spid_readsram.sv:1:1: , endln:4:11, parent:work@top
          |vpiName:SramAw
          |vpiFullName:spi_device_pkg::SramAw
        |vpiParamAssign:
        \_param_assign: , line:2:27, endln:2:38, parent:spi_device_pkg::
          |vpiRhs:
          \_constant: , line:2:36, endln:2:38
            |vpiDecompile:10
            |vpiSize:32
            |UINT:10
            |vpiTypespec:
            \_int_typespec: , line:2:14, endln:2:17, parent:spi_device_pkg::SramAw
            |vpiConstType:9
          |vpiLhs:
          \_parameter: (spi_device_pkg::SramAw), line:2:27, endln:2:38, parent:spi_device_pkg::
        |vpiTypedef:
        \_logic_typespec: (spi_device_pkg::sram_addr_t), line:3:12, endln:3:17
          |vpiName:spi_device_pkg::sram_addr_t
          |vpiInstance:
          \_package: spi_device_pkg (spi_device_pkg::) spid_readsram.sv:1:1: , endln:4:11, parent:work@top
          |vpiRange:
          \_range: , line:3:19, endln:3:29
            |vpiLeftRange:
            \_operation: , line:3:19, endln:3:25
              |vpiOpType:11
              |vpiOperand:
              \_constant: , line:3:19, endln:3:25
                |vpiDecompile:10
                |vpiSize:32
                |UINT:10
                |vpiTypespec:
                \_int_typespec: , line:2:14, endln:2:17
                  |vpiInstance:
                  \_package: spi_device_pkg (spi_device_pkg::) spid_readsram.sv:1:1: , endln:4:11, parent:work@top
                |vpiConstType:9
              |vpiOperand:
              \_constant: , line:3:26, endln:3:27
                |vpiDecompile:1
                |vpiSize:64
                |UINT:1
                |vpiConstType:9
            |vpiRightRange:
            \_constant: , line:3:28, endln:3:29
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
              |vpiConstType:9
        |vpiDefName:spi_device_pkg
    |vpiName:SramAw
    |vpiFullName:work@top.SramAw
    |vpiImported:spi_device_pkg
  |vpiParamAssign:
  \_param_assign: , line:2:27, endln:2:38, parent:work@top
    |vpiRhs:
    \_constant: , line:2:36, endln:2:38
      |vpiDecompile:10
      |vpiSize:32
      |UINT:10
      |vpiTypespec:
      \_int_typespec: , line:2:14, endln:2:17, parent:work@top.SramAw
        |vpiInstance:
        \_package: spi_device_pkg (spi_device_pkg::) spid_readsram.sv:1:1: , endln:4:11, parent:work@top
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (work@top.SramAw), line:2:27, endln:2:38, parent:work@top
  |vpiTypedef:
  \_logic_typespec: (spi_device_pkg::sram_addr_t), line:3:12, endln:3:17
  |vpiTypedef:
  \_import: (spi_device_pkg), line:7:11, endln:7:28
  |vpiDefName:work@top
  |vpiNet:                                            <------------------- Here
  \_logic_net: (sram_addr_t)
    |vpiName:sram_addr_t
    |vpiNetType:1
  |vpiPort:
  \_port: (o), line:6:23, endln:6:24, parent:work@top
    |vpiName:o
    |vpiDirection:2
    |vpiLowConn:
    \_ref_obj: (work@top.o)
      |vpiName:o
      |vpiFullName:work@top.o
      |vpiActual:
      \_int_var: (work@top.o), line:6:23, endln:6:24, parent:work@top
```
But when I change that typedef to `typedef logic [SramAw:0] sram_addr_t;` (I removed `-1`), the variable is not created.
This variable causes name conflict with the type name. Could you remove the creation of these variables?

I have also found another possible problem, but it doesn't result in bugs in Verilator. I have noticed that the typedefs are duplicated. The typedef is in vpiPackage node (where it should be), but is is also in vpiModule.

alaindargelas, 2022-01-16:
Fixed by #2432

alaindargelas, 2022-01-16:
Fixed by #2432

alaindargelas, 2022-01-16:
The latter part is not an issue, it is necessary to import the typespecs and parameters imported by the import command into the module.

closed 2022-01-16 by alaindargelas