chipsalliance/Surelog #2433: Centos 7 build issues

dpetrisko, 2022-01-16:
```
[ 80%] Building CXX object CMakeFiles/surelog.dir/generated/src/parser/SV3_1aParser.cpp.o
lib/libuhdm.a(Serializer_restore.cpp.o): In function `std::filesystem::path::path<std::string, std::filesystem::path>(std::string const&, std::filesystem::path::format) [clone .constprop.7357]':
Serializer_restore.cpp:(.text+0xd8): undefined reference to `std::filesystem::path::_M_split_cmpts()'
lib/libuhdm.a(attribute.cpp.o): In function `UHDM::attribute::VpiFile() const':
attribute.cpp:(.text+0x495): undefined reference to `std::filesystem::path::_M_split_cmpts()'
lib/libuhdm.a(attribute.cpp.o): In function `UHDM::attribute::VpiDefFile() const':
attribute.cpp:(.text+0x585): undefined reference to `std::filesystem::path::_M_split_cmpts()'
lib/libuhdm.a(virtual_interface_var.cpp.o): In function `UHDM::virtual_interface_var::VpiFile() const':
virtual_interface_var.cpp:(.text+0x2f5): undefined reference to `std::filesystem::path::_M_split_cmpts()'
lib/libuhdm.a(let_decl.cpp.o): In function `UHDM::let_decl::VpiFile() const':
let_decl.cpp:(.text+0x2f5): undefined reference to `std::filesystem::path::_M_split_cmpts()'
lib/libuhdm.a(always.cpp.o):always.cpp:(.text+0x325): more undefined references to `std::filesystem::path::_M_split_cmpts()' follow
collect2: error: ld returned 1 exit status
gmake[3]: *** [third_party/UHDM/CMakeFiles/uhdm-dump.dir/build.make:107: third_party/UHDM/bin/uhdm-dump] Error 1
gmake[3]: Leaving directory '/mnt/users/ssd0/no_backup/petrisko/Surelog/build'
gmake[2]: *** [CMakeFiles/Makefile2:1256: third_party/UHDM/CMakeFiles/uhdm-dump.dir/all] Error 2
gmake[2]: *** Waiting for unfinished jobs....
```

CentOS Linux release 7.9.2009 (Core)
gcc (GCC) 8.3.1 20190311 (Red Hat 8.3.1-3)

I see that  ```target_link_libraries(surelog PRIVATE stdc++fs)``` is included in CMakeLists.txt, so I'm not sure why this causing problems

hzeller, 2022-01-17:
since this is in the UHDM sub-project, where it fails, maybe the `stdc++fs` needs to be added to that if not already.

hs-apotell, 2022-01-17:
UHDM is a library so there shouldn't be any need for library linkage dependencies. This could be a problem with the order in which the libraries are linked on the command line. Also,  price that the gcc version being used is 8.+. The minimum required is gcc9 is I recollect correctly.

dpetrisko, 2022-01-17:
I tried with gcc-9 (devtoolset-9) and got the same result

alaindargelas, 2022-01-18:
@dpetrisko, is this a relative new issue? Was it working prior? Or it never compiled?

dpetrisko, 2022-01-18:
Hi Alain, compilation worked as of 212277903666342f6952135ec7106b3594b04e69. Haven't traced down the exact breaking commit yet

alaindargelas, 2022-01-18:
I strongly suspect:
https://github.com/chipsalliance/Surelog/pull/2399/files
which introduced filepath in UHDM. @hzeller gave you the correct solution to the problem you observed:
I see that target_link_libraries(surelog PRIVATE stdc++fs) is included in CMakeLists.txt, so I'm not sure why this causing problems

"since this is in the UHDM sub-project, where it fails, maybe the stdc++fs needs to be added to that if not already."

You have to add the same stdc++fs to your own executable link command in your own CMakeLists.txt:
target_link_libraries(<your executable> PRIVATE stdc++fs)

closed 2022-01-18 by alaindargelas

dpetrisko, 2022-01-18:
Thanks everyone!

PR #2437 (2022-01-18): Update CMakeLists.txt
Fixes #2433 by adding stdc++fs dependency to uhdm library