chipsalliance/Surelog #2445: The genscope is not parsed

RRozak, 2022-01-19:
The problem is similar to #2350, but in this example the first submodule is in `for`.
Source:
```systemverilog
package prim_pad_wrapper_pkg;
   typedef enum  logic [2:0] {
      A = 3'h0,
      B = 3'h1
   } pad_type_e;
endpackage : prim_pad_wrapper_pkg

package pinmux_pkg;
   import prim_pad_wrapper_pkg::*;

   parameter int NDioPads = 1;

   typedef struct packed {
      pad_type_e [NDioPads-1:0] dio_pad_type;
   } target_cfg_t;

   parameter target_cfg_t DefaultTargetCfg = '{
      dio_pad_type: {NDioPads{B}}
   };
endpackage : pinmux_pkg

module prim_generic_pad_attr(output int a);
   import prim_pad_wrapper_pkg::*;
   parameter pad_type_e PadType = A;
   if (PadType == B) begin : gen_assign
      assign a = 1;
   end
endmodule : prim_generic_pad_attr

module prim_pad_attr(output int b);
   import prim_pad_wrapper_pkg::*;
   parameter pad_type_e PadType = A;

   if (1) begin : gen_generic
      prim_generic_pad_attr #(
         .PadType(PadType)
      ) u_impl_generic(.a(b));
   end
endmodule

module top(output int o);
   import pinmux_pkg::*;
   parameter target_cfg_t TargetCfg = DefaultTargetCfg;

  for (genvar k = 0; k < NDioPads; k++) begin : gen_dio_attr
     prim_pad_attr #(
        .PadType(TargetCfg.dio_pad_type[k])
     ) u_prim_pad_attr(.b(o));
  end
endmodule
```
There is no `gen_assign` scope in the uhdm:
```
|uhdmtopModules:
\_module: work@top (work@top) bit_select_gen_for.sv:41:1: , endln:50:10
  |vpiName:work@top
  |vpiVariables:
  \_int_var: (work@top.o), line:41:23, endln:41:24, parent:work@top
    |vpiName:o
    |vpiFullName:work@top.o
    |vpiAutomatic:1
    |vpiVisibility:1
  |vpiParameter:
  \_parameter: (work@top.DefaultTargetCfg), line:17:27, endln:19:5, parent:work@top
    |vpiTypespec:
    \_struct_typespec: (pinmux_pkg::target_cfg_t), line:13:12, endln:13:18, parent:work@top.DefaultTargetCfg
      |vpiName:pinmux_pkg::target_cfg_t
[... 544 lines cut ...]
                \_parameter: (work@top.gen_dio_attr[0].u_prim_pad_attr.gen_generic.u_impl_generic.PadType), line:24:25, endln:24:36, parent:work@top.gen_dio_attr[0].u_prim_pad_attr.gen_generic.u_impl_generic
              |vpiTypedef:
              \_enum_typespec: (prim_pad_wrapper_pkg::pad_type_e), line:5:6, endln:5:16, parent:work@prim_pad_attr.PadType
              |vpiTypedef:
              \_import: (prim_pad_wrapper_pkg), line:23:11, endln:23:34
              |vpiDefName:work@prim_generic_pad_attr
              |vpiDefFile:bit_select_gen_for.sv
              |vpiDefLineNo:22
              |vpiPort:
              \_port: (a), line:22:41, endln:22:42, parent:work@top.gen_dio_attr[0].u_prim_pad_attr.gen_generic.u_impl_generic
                |vpiName:a
                |vpiDirection:2
                |vpiHighConn:
                \_ref_obj: (work@top.gen_dio_attr[0].u_prim_pad_attr.gen_generic.b), line:37:27, endln:37:28
                  |vpiName:b
                  |vpiFullName:work@top.gen_dio_attr[0].u_prim_pad_attr.gen_generic.b
                  |vpiActual:
                  \_int_var: (work@top.gen_dio_attr[0].u_prim_pad_attr.b), line:30:33, endln:30:34, parent:work@top.gen_dio_attr[0].u_prim_pad_attr
                |vpiLowConn:
                \_ref_obj: (work@top.gen_dio_attr[0].u_prim_pad_attr.gen_generic.u_impl_generic.a)
                  |vpiName:a
                  |vpiFullName:work@top.gen_dio_attr[0].u_prim_pad_attr.gen_generic.u_impl_generic.a
                  |vpiActual:
                  \_int_var: (work@top.gen_dio_attr[0].u_prim_pad_attr.gen_generic.u_impl_generic.a), line:22:41, endln:22:42, parent:work@top.gen_dio_attr[0].u_prim_pad_attr.gen_generic.u_impl_generic
```

alaindargelas, 2022-01-21:
The outpermost gen_scope prevents the proper evaluation/payload forwarding of TargetCfg.dio_pad_type[k]

alaindargelas, 2022-01-22:
Fixed by #2451

closed 2022-01-22 by alaindargelas