chipsalliance/Surelog #2449: INTERNAL ERROR: Cache is saturated - what to do ?

hzeller, 2022-01-22:
On parsing a particular large file (some concatenated set of verilog files), the Preprocessing concludes with

```
[INF:PP0122] Preprocessing source file "myfile.v".

INTERNAL ERROR: Cache is saturated
[INF:PA0201] Parsing source file "myfile.v".
```

Does this mean that the output is not usable or simply that Surelog fell back to some non-cached representation ?

alaindargelas, 2022-01-22:
Meanwhile I investigate, use the -nocache option.
I know the problem.
Can I get an idea of the file size in Mb?

hzeller, 2022-01-22:
Yes, I ended up using `-nocache` and that ran successfully. Without the cache, it continued to do something, then had the same cache saturation in the parsing phase and finished without outputting anything.

The input file was about 150MiB plain Verilog (and the output UHDM after successful 35 min parsing about 950MiB). So - probably not your everyday input (as I said, I just concatenated all files in a project), but a good stress test.

So maybe a first step would be that if the cache recognizes that it can't deal with all the data, things should continue, but just stop using the cache ?

alaindargelas, 2022-01-22:
I both increased the cache capacity drastically from 16M AST nodes to 268M nodes, and a graceful waning and auto-off if still exceeding. 
Fixed by #2457

closed 2022-01-22 by alaindargelas