chipsalliance/Surelog #2472: Wrong detection of implicite net

kamilrakoczy, 2022-01-25:
TEST:
```
`default_nettype none

module func_block_top(inp, out1, out2, out3);
        function automatic [31:0] func3;
                localparam A = 31;
                parameter B = 1 - 0;
                input [31:0] inp;
                func3[A:B] = inp[A:B];
        endfunction
endmodule
```
Surelog fails with message, but it should pass:
```
[ERR:EL0535] dut.sv:8:9: Illegal implicit net "A".

[ERR:EL0535] dut.sv:8:11: Illegal implicit net "B".

[ERR:EL0535] dut.sv:8:20: Illegal implicit net "A".

[ERR:EL0535] dut.sv:8:22: Illegal implicit net "B".
```

Thomasb81, 2022-01-26:
According to chapter 22.8, which state : "When the ``default_nettype is set to `none`, all nets shall be explicitly declared"

so I would expect following source code
```
                localparam A = 31;
                parameter B = 1 - 0;
```
to explicitly specify the kind of net...

Thomasb81, 2022-01-26:
My mistake, according to 6.20.2 a parameter does not have a net_type but a type.
So Surelog is incorrectly considering `default_nettype none` directive. The tool should not produce error

alaindargelas, 2022-01-29:
Fixed by #2498

closed 2022-01-29 by alaindargelas