chipsalliance/Surelog #2474: Surelog allows for missing details of port

kamilrakoczy, 2022-01-25:
TEST:
```
module tb;

wire [3:0] a, y;

test uut (.a(a), .y(y));

endmodule

module test(a, b, y);

input  [3:0] a;
output [3:0] y;

assign y = a;

endmodule
```
Surelog should fail on this code, as ``b`` port of module ``test`` is not defined.

Thomasb81, 2022-01-25:
I disagree : the tool must not failed.

Port direction is omit, so tool should implicitly set `inout` direction (see 23.2.3 from 1800.1)
Then it seems legal to me to do not connect all pin at instantiation. Usually simulator assign high-impedance to unconnected input.  Output are just driving nothing. Here you just have a port b part of the boundary but not use.
It is a common practice in some EDA tool like synthesis tool to duplicate port on purpose and let the original port unconnected in module...

From syntax point of view it is correct. Eventually we need to check that Surlog has correctly fill the blank for us inside uhdm db. (direction size)

alaindargelas, 2022-01-25:
All commercial tools do error out on this design and fail compilation. The error is a semantic error, the type of the port is undefined, it's not just a grammar error.

Thomasb81, 2022-01-26:
> module test(a, b, y); << Lack of definition for b is the problem, not the instantiation
Yes I agree with that. 

I must acknowledge commercial tool produce error so I must have miss details that lead to a condition where tool must rise issue.
Sorry for the noise.

alaindargelas, 2022-01-29:
Fixed by #2502

closed 2022-01-29 by alaindargelas