chipsalliance/Surelog #2540: Windows CI build is broken

alaindargelas, 2022-02-07:
Run cd build
  cd build
  mv regression surelog-windows-cl-regression
  find surelog-windows-cl-regression -name "*.log" | tar czvfp surelog-windows-cl-regression.tgz -T -
  shell: C:\Program Files\Git\bin\bash.EXE --noprofile --norc -e -o pipefail {0}
  env:
    pythonLocation: C:\hostedtoolcache\windows\Python\3.8.[1](https://github.com/chipsalliance/Surelog/runs/5088744615?check_suite_focus=true#step:12:1)0\x64
    JAVA_HOME_8.0.3[2](https://github.com/chipsalliance/Surelog/runs/5088744615?check_suite_focus=true#step:12:2)2_x64: C:\hostedtoolcache\windows\jre\8.0.[3](https://github.com/chipsalliance/Surelog/runs/5088744615?check_suite_focus=true#step:12:3)22\x6[4](https://github.com/chipsalliance/Surelog/runs/5088744615?check_suite_focus=true#step:12:4)
    JAVA_HOME: C:\hostedtoolcache\windows\jre\8.0.322\x[6](https://github.com/chipsalliance/Surelog/runs/5088744615?check_suite_focus=true#step:12:6)4
    JAVA_HOME_[8](https://github.com/chipsalliance/Surelog/runs/5088744615?check_suite_focus=true#step:12:8)_0_322_X64: C:\hostedtoolcache\windows\jre\8.0.322\x64
mv: cannot stat 'regression': No such file or directory
Error: Process completed with exit code 1.

hs-apotell, 2022-02-07:
That's not the issue. The problem is the build failed long before this step.

The problem is that the Windows Server for GHA got upgraded from 2019 to 2022. Windows Server 2019 has VS2019 and Windows Server 2022 has VS2022. The paths for required files (vcvars.bat, to be specific) changed between these versions (2019 was still x86 and 2022 is now x64).

The migration is still in progress i.e. sometimes we get a Windows Server 2019 machine and other times a Windows Server 2022 machine. So build works sometimes and fails other times.

UHDM has the same problem and I have submitted a PR to fix it. But haven't been able to confirm the fix because of the migration still being in progress. Will follow up for Surelog as well once the fix is confirmed for UHDM.

alaindargelas, 2022-02-13:
This is fixed

closed 2022-02-13 by alaindargelas