chipsalliance/Surelog #2541: Querying for files included

hs-apotell, 2022-02-07:
Given a source file, how would I query for what files were included and their corresponding positions in the file?

`Include` is not part of the UHDM hierarchy. Can we make another UHDM model to represent this information?

alaindargelas, 2022-02-08:
UHDM does not carry pre-processor information. Surelog pre-processor on the other hand has all that information. 
If you link your application to libsurelog.a, you can retrieve that information from the PreprocessFile class, it has relations of who includes a given file and what files are included from a given file. Macros are also stored in FileContent data structures.
From the Design object, you can get the list of files (Compilation Units), then from the list of Compilation units you can navigate the include information.
Some class members might have to be exposed by public assessor functions so you can make those queries.

hs-apotell, 2022-02-09:
Are you referring to using https://github.com/chipsalliance/Surelog/blob/d07533d8e119868b422605df7aaffd3908a4394f/src/Design/Design.h#L198 or https://github.com/chipsalliance/Surelog/blob/d07533d8e119868b422605df7aaffd3908a4394f/src/Design/Design.h#L200?

I see it has pointers to `FileContent` but how do I go from `FileContent` back to to `PreprocessFile`? Does the relation already exist?

Instead, why not use https://github.com/chipsalliance/Surelog/blob/d07533d8e119868b422605df7aaffd3908a4394f/src/SourceCompile/Compiler.h#L67 since `CompilerSourceFile` already has direct access to `PreprocessFile`. https://github.com/chipsalliance/Surelog/blob/d07533d8e119868b422605df7aaffd3908a4394f/src/SourceCompile/CompileSourceFile.h#L91

Noted - `Compiler` class itself is shadowed by `SURELOG::scompiler` and also compiler.h isn't installed either.

alaindargelas, 2022-02-10:
You proposed it originally, but let me expand on it: You could create a new set of object model in UHDM. Just don't call them vpiXXX, as they are not part of the VPI API, call them uhdmFileInfo, uhdmMacroInfo, uhdmInclude or something like that.
Then from Surelog, populate that model the same way we populate the VPI UHDM using the pointers you found above. 
Your client code would interface with the UHDM model instead of the guts API of Surelog.

From the uhdmdesign object, you could have a vector of uhdmfileinfo, which in turn has include uhdmfileinfo vector and uhdmmacro info in a recursive fashion.

hs-apotell, 2022-04-12:
@alaindargelas What's the update on this? Are the all pieces in place for exposing the necessary information from within the UHDM model? I was planning on adding a list of all imports to the `design` model.
```
struct ImportFileInfo {
  SymbolId sourceFileId;
  SymbolId targetFileId;
  int32_t startLine;
  int32_t startColumn;
  int32_t endLine;
  int32_t endColumn;
};
```
Let me know if you had some other approach in mind.

alaindargelas, 2022-04-12:
This is way too low priority for me at this point, you have to handle it on your own.

hs-apotell, 2022-04-12:
We can probably handle this with some guidance.

The last I remember was that the end-column information for the import filename was missing. This was the only piece I wasn't clear on how to proceed.

Is `design` still the right place to expose this information from?

alaindargelas, 2022-04-12:
Yes, create a full schema in UHDM, (append to the existing Verilog model):

import_file_info.yaml

obj_def: import_file_info
    .....

hs-apotell, 2022-04-13:
There is already a schema named 'import' which is used for representing package import. There's only a single 'constant' property. Can we use this schema to represent file import as well or would you rather define a new schema?

Also, any reason why it's named `import` and not `import_typespec`??

alaindargelas, 2022-04-13:
Please create a new schema.

Sorry, I used a wrong name, based on the struct above.

It should be:

include_file_info.yaml

obj_def: include_file_info
.....

typespec is for data type.

hs-apotell, 2022-04-30:
Fixed by #2892

closed 2022-04-30 by hs-apotell