chipsalliance/Surelog #2545: [seeking advice] How to get the assignments of a given variable/net until a certain depth?

mysoreanoop, 2022-02-07:
Assume I have a handle to a variable/net in a module.  I want to trace that to the continuous assignment (or procedural, but see 3) that updates it. What I'm planning to do is save (in a list) every continuous assignment's LHS and RHS. And search this list when needed to get the first-degree assignment.
1. Is this an efficient way? Would vpi_handle_by_name against a module instance handle help here?

When I need to get the 2nd-degree assignments (the assignments of each of the variables in the first assignment), I repeat the same thing.
2. Is there a monolithic function call that can do this till a depth k, or the variable is from a port definition or something else, whichever comes quickest? Would it be useful to implement it for a general case?

3. If, say, I also need to get the procedural assignment that assigns to a variable, or a conditional continuous assignment (with an always_comb block), how best would one go about retrieving the condition (or a list of all conditions when nested, that can be manually ANDed)? Would it be best to maybe generate a list of all assignment handles within every unique nesting of a condition expression?

Thanks a lot in advance!

alaindargelas, 2022-02-08:
All applications have their needs, this seems pretty limited for general purpose.
Do something like this:

Do one pass over every single assignments inside a module, build a std::multimap<lhs, rhs>

Create an empty stack  std::stack<std::pair<net, int>> transitiveCone;
push the std::pair<net, depth (0)> you start from
set<net> result;
while (!transitiveCone.empty()) {
    pop the net
    find all keys in the multimap
        if key_depth >= limit 
              push key into result container 
       else 
           push the key in the stack and increase their depth
    if net has no entry in the map, push it to the result
}

retrieve the result

Try a similar approach for problem 3. Or let's talk about it after you solve 1 and 2

alaindargelas, 2022-02-23:
@mysoreanoop, have you made any progress? Need help?

mysoreanoop, 2022-03-01:
Still working on it; will resume next week. Thanks for checking in.

closed 2023-08-26 by alaindargelas