chipsalliance/Surelog #2554: Surelog fails to parse wire inside generate and block stmt

kamilrakoczy, 2022-02-08:
TEST:
```
module gen_test9;
        wire [1:0] w = 2'b11;
        generate
                begin : A
                        wire [1:0] x;
                        begin : B
                                wire [1:0] y = 2'b00;
                        end
                        begin : C
                                wire [1:0] z = 2'b01;
                        end
                        assign x = B.y ^ 2'b11 ^ C.z;
                end
        endgenerate
endmodule
```
Surelog fails with message:
```
dut.sv:6:3: Syntax error: no viable alternative at input 'wire [1:0] x;\n\t\t\tbegin',
```
but it is correct code.

alaindargelas, 2022-02-12:
That problem got introduced when we fixed the exponential parse time with the if generate statements.
https://github.com/chipsalliance/Surelog/commit/f5915e2861be4e9535e34cb8948ad256faf297e6
Or when I did the grammar speed optimization afterwards.
@QuantamHD can you verify that your large test still passes.
Can you create a simplified test that I put in Surelog regression to avoid asking you to check everytime?

alaindargelas, 2022-02-12:
Fixed by #2583

closed 2022-02-12 by alaindargelas