chipsalliance/Surelog #2557: Surelog does not build with Python enabled.

suzizecat, 2022-02-08:
Using the following command:
```
 make ADDITIONAL_CMAKE_OPTIONS="-DSURELOG_WITH_PYTHON=1 -DCMAKE_CXX_FLAGS=-fpermissive" PREFIX=~/opt/surelog
```

on master, Surelog fails to compile, providing this error message:
```
[ 98%] Building CXX object CMakeFiles/surelog.dir/src/Cache/PythonAPICache.cpp.o
/home/jfaucher/local-builds/Surelog/src/Cache/PythonAPICache.cpp:54:10: error: no declaration matches ‘std::filesystem::__cxx11::path SURELOG::PythonAPICache::getCacheFileName_(std::filesystem::__cxx11::path) const’
   54 | fs::path PythonAPICache::getCacheFileName_(fs::path svFileName) const {
      |          ^~~~~~~~~~~~~~
In file included from /home/jfaucher/local-builds/Surelog/src/Cache/PythonAPICache.cpp:23:
/home/jfaucher/local-builds/Surelog/src/Cache/PythonAPICache.h:44:25: note: candidate is: ‘std::filesystem::__cxx11::path SURELOG::PythonAPICache::getCacheFileName_(const std::filesystem::__cxx11::path&) const’
   44 |   std::filesystem::path getCacheFileName_(
      |                         ^~~~~~~~~~~~~~~~~
/home/jfaucher/local-builds/Surelog/src/Cache/PythonAPICache.h:33:7: note: ‘class SURELOG::PythonAPICache’ defined here
   33 | class PythonAPICache : Cache {
      |       ^~~~~~~~~~~~~~
  ```
 
Not using `-fpermissive` will also fails with an invalid cast from `void*` to something else I can't remember.

This is done with 
 - Ubuntu 20.04
 - cmake 3.16.3
 - Python 3.8.10
 - gcc 9.3.0

Best regards,
Julien FAUCHER

alaindargelas, 2022-02-09:
The Python API is not maintained anymore, that said, at least it should be compiling, I'll check.
Can I suggest you to take a look in the meantime at the UHDM C/C++ VPI, which is the main API for Surelog that heavily used?

suzizecat, 2022-02-09:
Hi, 
Thanks for the heads up. I wanted to try building a language server around Surelog, the issue being that there isn't a good framework in C++ to do so, while there is a great one in Python (pygls). 
I'll have a more detailed look at it.

Does the UHDM interface provides symbols locations in original source files as the Python API seems to do ? 

On a side note, if python interface is not available, you might want to edit the READMEs about this.

alaindargelas, 2022-02-10:
Fixed with #2570. Python compiles again.
Both the Python and The UHDM interfaces have source code location.
The Python interface will have more language level information like pre-processor information, macros, include file locations.
The UHDM interface is the VPI API, elaborated Verilog Object model, with location of all the objects, but all the pre-processing information is gone and elaboration has made some transformations to the design. All UHDM object refer to source code constructs, but some source code constructs have been processed, inlined, expanded, so you don't find the a corresponding object in UHDM.

alaindargelas, 2022-02-10:
BTW, there was another inquiry about language server: https://github.com/chipsalliance/Surelog/issues/1712

closed 2022-02-10 by alaindargelas

hs-apotell, 2022-02-10:
@suzizecat Are you still interested in building language server using Surelog? I am actively working on one and have basic features working. We can collaborate, if you are interested. BTW my entire implementation is in C++, no python involved.

suzizecat, 2022-02-10:
@hs-apotell I am still interested but don't really know if I'll be able to afford the time to do so. I would be curious to have a look at your implementation though. 

I have a basic LS working with Verible as a backbone, but Verible lacks some features to get a really nice result.
If you are curious, this implementation (based upon pygls and with a frontend only for verible) is : https://github.com/suzizecat/diplomat-ls

It is made to work with TerosHDL which provides an  ordered list of files, required for verible.
It support symbol navigation (find declaration/implmenetation) for everything properly supported by Verible.

It, however, lacks proper incremental updates capabilities but should be able to support it without too much trouble.
The index is refreshed only on saving files.