chipsalliance/Surelog #2559: Anonymous structs might not be handled correctly in Surelog.

QuantamHD, 2022-02-08:
```systemverilog
module anon_module 

   (
    output logic                                    all_done 
    );

   typedef struct packed {
     logic                      a;
     logic                      b;
   } typedef_struct_thing;

   typedef_struct_thing [50:0]   data;  

   struct packed {
     logic a;
     logic [100:0]      b;
   } [200:0] anon_struct;
     

   always_comb begin
     data[0].a = 0; //ok
     anon_struct[0].a = 0; // Not ok
     all_done = anon_struct.a;
   end

endmodule : anon_module
```

Problem observed in https://github.com/SymbiFlow/yosys-symbiflow-plugins/issues/218. The issue seems to be that there's a difference between the way anonymous structs are handled vs typedefed ones. I believe adding a dynamically generated typedef name for anonymous structs would also fix this issue.

alaindargelas, 2022-02-11:
If I simplify the test as:
```
module anon_module ();

   typedef struct packed {
     logic                      a;
   } typedef_struct_thing;

   typedef_struct_thing /*[50:0]*/ data;  

   struct packed {
     logic a;
   } /*[200:0]*/ anon_struct;
   
  assign   data.a = 0; //ok
  assign anon_struct.a = 0; // Not ok
  
endmodule 
```

The following UHDM is correct for both types, the hier path as a actual for the corresponding var and member:
```
  |vpiContAssign:
  \_cont_assign: , id:23, line:23:12, endln:23:22, parent:work@anon_module, parID:22
    |vpiRhs:
    \_constant: , id:6, line:23:21, endln:23:22, parID:7
    |vpiLhs:
    \_hier_path: (data.a), id:24, line:23:12, endln:23:16, parID:23
      |vpiName:data.a
      |vpiActual:
      \_ref_obj: (data), id:25, parent:data.a, parID:24
        |vpiName:data
        |vpiActual:
        \_struct_net: (work@anon_module.data), id:13, line:13:38, endln:13:42, parent:work@anon_module, parID:22
      |vpiActual:
      \_ref_obj: (a), id:26, parent:data.a, parID:24
        |vpiName:a
        |vpiActual:
        \_typespec_member: (a), id:2, line:8:33, endln:8:34
  |vpiContAssign:
  \_cont_assign: , id:27, line:24:10, endln:24:27, parent:work@anon_module, parID:22
    |vpiRhs:
    \_constant: , id:11, line:24:26, endln:24:27, parID:12
    |vpiLhs:
    \_hier_path: (anon_struct.a), id:28, line:24:10, endln:24:21, parID:27
      |vpiName:anon_struct.a
      |vpiActual:
      \_ref_obj: (anon_struct), id:29, parent:anon_struct.a, parID:28
        |vpiName:anon_struct
        |vpiActual:
        \_struct_net: (work@anon_module.anon_struct), id:17, line:19:18, endln:19:29, parent:work@anon_module, parID:22
      |vpiActual:
      \_ref_obj: (a), id:30, parent:anon_struct.a, parID:28
        |vpiName:a
        |vpiActual:
        \_typespec_member: (a), id:16, line:17:12, endln:17:13
```

alaindargelas, 2022-02-11:
If I re-add the arrays and the bit selects:
```
module anon_module ();

   typedef struct packed {
     logic                      a;
   } typedef_struct_thing;

   typedef_struct_thing [50:0] data;  

   struct packed {
     logic a;
   } [200:0] anon_struct;
   
  assign   data[0].a = 0; //ok
  assign anon_struct[0].a = 0; // Not ok
  
endmodule 
```
Then both bindings don't happen (No vpiActual for bit_select objects), so the problem is not specific to one struct vs the anonymous struct. Not in UHDM at least, it is pretty consistent. 
Now the question would be do we need to extend the bit_select and part_select to have vpiActual members so we can attach the bound object (the variable).

```
  |vpiContAssign:
  \_cont_assign: , id:37, line:13:12, endln:13:25, parent:work@anon_module, parID:36
    |vpiRhs:
    \_constant: , id:7, line:13:24, endln:13:25, parID:8
    |vpiLhs:
    \_hier_path: (data[0].a), id:38, line:13:12, endln:13:16, parID:37
      |vpiName:data[0].a
      |vpiActual:
      \_bit_select: (data), id:39, parent:work@anon_module.data[0].a, parID:40
        |vpiName:data
        |vpiIndex:
        \_constant: , id:4, line:13:17, endln:13:18
      |vpiActual:
      \_ref_obj: (a), id:41, parent:data[0].a, parID:38
        |vpiName:a
  |vpiContAssign:
  \_cont_assign: , id:42, line:14:10, endln:14:30, parent:work@anon_module, parID:36
    |vpiRhs:
    \_constant: , id:13, line:14:29, endln:14:30, parID:14
    |vpiLhs:
    \_hier_path: (anon_struct[0].a), id:43, line:14:10, endln:14:21, parID:42
      |vpiName:anon_struct[0].a
      |vpiActual:
      \_bit_select: (work@anon_module.anon_struct), id:44, parent:work@anon_module.anon_struct[0].a, parID:45
        |vpiName:anon_struct
        |vpiFullName:work@anon_module.anon_struct
        |vpiIndex:
        \_constant: , id:10, line:14:22, endln:14:23
      |vpiActual:
      \_ref_obj: (a), id:46, parent:anon_struct[0].a, parID:43
        |vpiName:a
```

QuantamHD, 2022-02-11:
I also filed this bug in the plugin repo https://github.com/SymbiFlow/yosys-symbiflow-plugins/issues/218. @kbieganski do you have any thoughts on the correct approach?

kamilrakoczy, 2022-02-16:
We already have all necessary information to add support for this in yosys. I've added support for this in: https://github.com/SymbiFlow/yosys-symbiflow-plugins/pull/233. I think we can close this issue, as we don't need this nodes to have ``vpiActual`` (for yosys).

closed 2022-02-16 by alaindargelas