chipsalliance/Surelog #2576: cmake not checking that orderedmultidict python module exists.

mithro, 2022-02-11:
As the `orderedmultidict` python module is required to build Surelog, cmake should be checking that it is installed.

mithro, 2022-02-11:
I think the preferred way to do this might be described at https://github.com/dev-cafe/cmake-cookbook/tree/v1.0/chapter-03/recipe-03

mithro, 2022-02-11:
https://github.com/dev-cafe/cmake-cookbook/blob/e2cdfb3c79477014966d893e9712b27aa18048b3/chapter-03/recipe-03/cxx-example/CMakeLists.txt#L11-L41
```cmake
find_package(PythonInterp REQUIRED)
find_package(PythonLibs ${PYTHON_VERSION_MAJOR}.${PYTHON_VERSION_MINOR} EXACT REQUIRED)

# Find NumPy location
execute_process(
  COMMAND
    ${PYTHON_EXECUTABLE} "-c" "import re, numpy; print(re.compile('/__init__.py.*').sub('',numpy.__file__))"
  RESULT_VARIABLE _numpy_status
  OUTPUT_VARIABLE _numpy_location
  ERROR_QUIET
  OUTPUT_STRIP_TRAILING_WHITESPACE
  )
if(NOT _numpy_status)
  set(NumPy ${_numpy_location} CACHE STRING "Location of NumPy")
endif()

# Find NumPy version
execute_process(
  COMMAND
    ${PYTHON_EXECUTABLE} "-c" "import numpy; print(numpy.__version__)"
  OUTPUT_VARIABLE _numpy_version
  ERROR_QUIET
  OUTPUT_STRIP_TRAILING_WHITESPACE
  )

include(FindPackageHandleStandardArgs)
find_package_handle_standard_args(NumPy
  FOUND_VAR NumPy_FOUND
  REQUIRED_VARS NumPy
  VERSION_VAR _numpy_version
  )
```

hzeller, 2022-02-11:
I think the culprit is actually UHDM, that needs this (and Surelog only be depending on it). Filed
https://github.com/chipsalliance/UHDM/issues/640

alaindargelas, 2022-02-14:
Fixed by #2596 that merged UHDM change https://github.com/chipsalliance/UHDM/pull/645

closed 2022-02-14 by alaindargelas