chipsalliance/Surelog #2667: Post Elaboration, information is lost

hs-apotell, 2022-02-24:
Test Case: ArianeElab
RHS expression on line 19 thru' 28 are replaced with constants and the original expressions are lost.

Original source:
```
    localparam VLEN       = (XLEN == 32) ? 32 : 64;
    localparam PLEN       = (XLEN == 32) ? 32 : 56;

    localparam IS_XLEN64  = (XLEN == 32) ? 1'b0 : 1'b1;
    localparam ModeW      = (XLEN == 32) ? 1 : 4;
    localparam ASIDW      = (XLEN == 32) ? 9 : 16;
    localparam PPNW       = (XLEN == 32) ? 22 : 44;
    localparam vm_mode_t MODE_SV = (XLEN == 32) ? ModeSv32 : ModeSv39;
    localparam SV         = (MODE_SV == ModeSv32) ? 32 : 39;
    localparam VPN2       = (VLEN-31 < 8) ? VLEN-31 : 8;
```

As per the UHDM model:
```
    localparam VLEN              =  64;
    localparam PLEN              =  56;

    localparam IS_XLEN64         = 1;
    localparam ModeW             = 4;
    localparam ASIDW             = 16;
    localparam PPNW              = 44;
    localparam vm_mode_t MODE_SV = 8;
    localparam SV                = 39;
    localparam VPN2              = 8;
```

Original source includes the term `XLEN` and those expressions are required to build results for operations like "Find All References".

alaindargelas, 2022-02-24:
Constant propagation (reducing expressions) are a must-do to be able to perform elaboration. If expressions are kept symbolically, then no constant propagation can be done and no elaboration can be done.
If you need to do find all references, you can look at the non-elaborated UHDM for the modules: vpiAllModules instead of the vpiTopModules (elaborated view).
Note that unfortunately, packages are also fully elaborated, so all params/local params will be constant propagated in packages. There is no view for packages that contain the original expressions, maybe we can add that.

hs-apotell, 2022-02-24:
>  There is no view for packages that contain the original expressions, maybe we can add that.

For LS, original text in the source will be a requirement for number of features. This will need to be retained in the UHDM hierarchy or provide some other access to this information.

alaindargelas, 2022-02-27:
Fixed by #2677

closed 2022-02-27 by alaindargelas