chipsalliance/Surelog #2669: Location information for typespec_member/enum_typespec

hs-apotell, 2022-02-24:
For code snippet below:

```
logic [62:36] wpri4;  // writes preserved reads ignored
xlen_e        sxl;    // xlen_e is an enum
```

here's the UHDM output -

```
|vpiTypespecMember:
\_typespec_member: (wpri4), id:313, line:57:23, endln:57:28
  |vpiName:wpri4
  |vpiTypespec:
  \_logic_typespec: , id:312, line:57:9, endln:57:14
    |vpiInstance:
    \_package: riscv (riscv::), id:0 dut.sv:1:1: , endln:671:11, parent:work@top, parID:13585
    |vpiRange:
    \_range: , id:309, line:57:16, endln:57:21, parent:riscv::status_rv_t, parID:306
      |vpiLeftRange:
      \_constant: , id:310, line:57:16, endln:57:18, parID:309
        |vpiDecompile:62
        |vpiSize:64
        |UINT:62
        |vpiConstType:9
      |vpiRightRange:
      \_constant: , id:311, line:57:19, endln:57:21, parID:309
        |vpiDecompile:36
        |vpiSize:64
        |UINT:36
        |vpiConstType:9
       
|vpiTypespecMember:
\_typespec_member: (sxl), id:314, line:58:23, endln:58:26
  |vpiName:sxl
  |vpiTypespec:
  \_enum_typespec: (riscv::xlen_e), id:279, line:42:5, endln:46:14
    |vpiName:riscv::xlen_e
    |vpiInstance:
    \_package: riscv (riscv::), id:0 dut.sv:1:1: , endln:671:11, parent:work@top, parID:13585
    |vpiBaseTypespec:
    \_logic_typespec: , id:278, line:42:18, endln:42:23
      |vpiInstance:
      \_package: riscv (riscv::), id:0 dut.sv:1:1: , endln:671:11, parent:work@top, parID:13585
      |vpiRange:
      \_range: , id:275, line:42:25, endln:42:28
        |vpiLeftRange:
        \_constant: , id:276, line:42:25, endln:42:26, parID:275
          |vpiDecompile:1
          |vpiSize:64
          |UINT:1
          |vpiConstType:9
        |vpiRightRange:
        \_constant: , id:277, line:42:27, endln:42:28, parID:275
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
    |vpiEnumConst:
    \_enum_const: (XLEN_32), id:281, line:43:9, endln:43:25
      |vpiName:XLEN_32
      |UINT:1
      |vpiDecompile:2'b01
      |vpiSize:2
    |vpiEnumConst:
    \_enum_const: (XLEN_64), id:284, line:44:9, endln:44:25
      |vpiName:XLEN_64
      |UINT:2
      |vpiDecompile:2'b10
      |vpiSize:2
    |vpiEnumConst:
    \_enum_const: (XLEN_128), id:287, line:45:9, endln:45:25
      |vpiName:XLEN_128
      |UINT:3
      |vpiDecompile:2'b11
      |vpiSize:2
```

`logic_typespec` instances are unique for every use. However, `enum_typespec` uses are not. The former can hold location information relevant to the context of use but the latter cannot.

Would it be possible to convert the latter to use a `ref_obj` so that it can hold the context location information? i.e.

Instead of `typespec_member` -> `enum_typespec`, you would have `typespec_member` -> `ref_obj` -> `enum_typespec`.

alaindargelas, 2022-02-27:
Fixed by #2679.
typespec_member have new ref location fields:
#define  vpiRefFile          3018
#define  vpiRefLineNo        3019
#define  vpiRefColumnNo      3020
#define  vpiRefEndLineNo     3021
#define  vpiRefEndColumnNo   3022

The ref_typespec (ref_obj was not possible) solution would have broken too much existing code.

hs-apotell, 2022-03-05:
Noticed a similar issue with `logic_typespec` as well when declared as `typedef`.
Test: ArianeElab
https://github.com/chipsalliance/Surelog/blob/c85f65ef41ea7bff7a55ae8deddf63b27a95b7c6/tests/ArianeElab/dut.sv#L30

```
|vpiTypespecMember:
    \_typespec_member: (cause), line:937:30, endln:937:35, parent:ariane_pkg::exception_t
      |vpiName:cause
      |vpiTypespec:
      \_logic_typespec: (riscv::xlen_t), line:30:13, endln:30:18, parent:riscv::
```

Shouldn't this be a clone instead of using the existing instance? I need the location information of the typespec which would be wrong here.

alaindargelas, 2022-07-31:
@hs-apotell you made several fixes regarding location. This should be closed.

closed 2022-07-31 by alaindargelas