chipsalliance/Surelog #2674: Should typespec::vpiParent ever be non-null??

hs-apotell, 2022-02-25:
`typespec` (and subclasses) can generally be shared across various variable declarations. Considering that fact, should `typespec::vpiParent` ever be non-nullptr??

At the moment, there are number of cases where `typespec` parent is being set to something legal. Just wondering if I am missing something obvious. If not, I can fix these as I find them.

alaindargelas, 2022-02-27:
typespec::vpiParent should always contain something. If you find nullptr, we should put the correct parent.

hs-apotell, 2022-02-27:
Hmm this is contradicting my statement. For non-elaborated model, `typespec::vpiParent` would be either be package or module depending on where its declared. But for elaborated model, however, the same `typespc` could be used by two different modules or in different context within the same module. Wouldn't the `vpiParent` pointer be ambiguous for elaborated models?

I see that `logic_typespec` are unique in the hierarchy (possibly because of the `range` that they hold which could be different for each use case) but other subclasses of `typespec` aren't unique. They are shared across uses.

This issue leans back into the other issue #2669. If we want to have a legal pointer for tracking parent of `typespec`, there needs to be a `ref_obj` in the hierarchy.

alaindargelas, 2022-02-28:
The definition typespec has a parent pointing to its definition location {The package, the design (if outside any scope), the module}, for complex types like structs it point where it is defined. Basic typespecs {logic, bit, byte...} are inlined, they get uniquely defined where used. All typespecs gets a parent.
In the elaborated model, typespecs get uniquified (cloned) and point to their respective instance parent, that is because of parameters or defparams that can influence their content.

Issue https://github.com/chipsalliance/Surelog/issues/2669 has been solved with location values, as stated, a ref_obj is out of the question because of the class inheritance of the object model. A new ref_typespec would have to be introduced and would break backward compatibility with too much code (And not adding much to the location solution).

hs-apotell, 2022-02-28:
May be I am just missing something obvious here. Please help me out.

Here's the UHDM for AllBindings test. I have enabled the `vpi_ids` command line argument to help uniquely identify the nodes. This examples has one enum definition, one module definition, and two unique use case of the enum in the module.
> In the elaborated model, typespecs get uniquified (cloned)

As per your statement, I expected to find 3 instances of `type_scr1_mem_resp_e` `enum_typespec`
1. Definition in enum.v, line 1
2. Use in top.v, line 1
3. Use in top.v, line 3

But there is only one instance and that happens to be the definition (id: 4). Also, the parent pointer is null for the definition.

Also, `typespec`s seems to be crossing AllModules & TopModules boundary. It was my understanding that everything gets cloned during the elaboration process. Is that not the case anymore?

```
design: (work@top), id:18
|vpiElaborated:1
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top), id:19 dut.sv:1:1: , endln:5:10, parent:work@top, parID:18
  |vpiFullName:work@top
  |vpiDefName:work@top
  |vpiNet:
  \_logic_net: (work@top.dmem2exu_resp_i), id:21, line:1:57, endln:1:72, parent:work@top, parID:19
    |vpiName:dmem2exu_resp_i
    |vpiFullName:work@top.dmem2exu_resp_i
  |vpiNet:
  \_logic_net: (work@top.dmem2exu_resp_ii), id:23, line:3:37, endln:3:53, parent:work@top, parID:19
    |vpiName:dmem2exu_resp_ii
    |vpiFullName:work@top.dmem2exu_resp_ii
  |vpiPort:
  \_port: (dmem2exu_resp_i), id:20, line:1:57, endln:1:72, parent:work@top, parID:19
    |vpiName:dmem2exu_resp_i
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: , id:22
      |vpiActual:
      \_logic_net: (work@top.dmem2exu_resp_i), id:21, line:1:57, endln:1:72, parent:work@top, parID:19
|vpiTypedef:
\_enum_typespec: (type_scr1_mem_resp_e), id:4, line:1:1, endln:5:24
  |vpiName:type_scr1_mem_resp_e
  |vpiInstance:
  \_design: (work@top), id:18
  |vpiBaseTypespec:
  \_logic_typespec: , id:3, line:1:14, endln:1:19
    |vpiRange:
    \_range: , id:0, line:1:20, endln:1:23
      |vpiLeftRange:
      \_constant: , id:1, line:1:20, endln:1:21, parID:0
        |vpiDecompile:1
        |vpiSize:64
        |UINT:1
        |vpiConstType:9
      |vpiRightRange:
      \_constant: , id:2, line:1:22, endln:1:23, parID:0
        |vpiDecompile:0
        |vpiSize:64
        |UINT:0
        |vpiConstType:9
  |vpiEnumConst:
  \_enum_const: (SCR1_MEM_RESP_NOTRDY), id:6, line:2:5, endln:2:36
    |vpiName:SCR1_MEM_RESP_NOTRDY
    |UINT:0
    |vpiDecompile:2'b00
    |vpiSize:2
  |vpiEnumConst:
  \_enum_const: (SCR1_MEM_RESP_RDY_OK), id:9, line:3:5, endln:3:36
    |vpiName:SCR1_MEM_RESP_RDY_OK
    |UINT:1
    |vpiDecompile:2'b01
    |vpiSize:2
  |vpiEnumConst:
  \_enum_const: (SCR1_MEM_RESP_RDY_ER), id:12, line:4:5, endln:4:36
    |vpiName:SCR1_MEM_RESP_RDY_ER
    |UINT:2
    |vpiDecompile:2'b10
    |vpiSize:2
|uhdmtopModules:
\_module: work@top (work@top), id:24 dut.sv:1:1: , endln:5:10
  |vpiName:work@top
  |vpiVariables:
  \_enum_var: (work@top.dmem2exu_resp_i), id:15, line:1:57, endln:1:72, parent:work@top, parID:24
    |vpiTypespec:
    \_enum_typespec: (type_scr1_mem_resp_e), id:4, line:1:1, endln:5:24
    |vpiName:dmem2exu_resp_i
    |vpiFullName:work@top.dmem2exu_resp_i
    |vpiAutomatic:1
    |vpiVisibility:1
  |vpiVariables:
  \_enum_var: (work@top.dmem2exu_resp_ii), id:16, line:3:37, endln:3:53, parent:work@top, parID:24
    |vpiTypespec:
    \_enum_typespec: (type_scr1_mem_resp_e), id:4, line:1:1, endln:5:24
    |vpiName:dmem2exu_resp_ii
    |vpiFullName:work@top.dmem2exu_resp_ii
    |vpiAutomatic:1
    |vpiVisibility:1
  |vpiDefName:work@top
  |vpiPort:
  \_port: (dmem2exu_resp_i), id:25, line:1:57, endln:1:72, parent:work@top, parID:24
    |vpiName:dmem2exu_resp_i
    |vpiDirection:1
    |vpiLowConn:
    \_ref_obj: (work@top.dmem2exu_resp_i), id:26, parent:dmem2exu_resp_i, parID:25
      |vpiName:dmem2exu_resp_i
      |vpiFullName:work@top.dmem2exu_resp_i
      |vpiActual:
      \_enum_var: (work@top.dmem2exu_resp_i), id:15, line:1:57, endln:1:72, parent:work@top, parID:24
    |vpiTypedef:
    \_enum_typespec: (type_scr1_mem_resp_e), id:4, line:1:1, endln:5:24
    |vpiInstance:
    \_module: work@top (work@top), id:24 dut.sv:1:1: , endln:5:10
```

alaindargelas, 2022-03-04:
Fixed by #2700

I made the following change, typespecs **now** always have a parent:
If the typespec is declared at the $root level, the parent is the design object
If the typespec is declared in a package it is the vpiTopPackage
If the typespec is declared in a module or interface or class or program it is the vpiAllModule, vpiAllInterface, vpiAllClass, vpiAllProgram object
If a typespec is declared in the generate statement, it is the generate statement instance (Fully elaborated)

closed 2022-03-04 by alaindargelas