chipsalliance/Surelog #2684: Module array instances get unrolled during elaboration

hs-apotell, 2022-02-28:
Test: ArrayInst

Issue: Both arrays `q` and `flop_instances` get unrolled during elaboration losing the original context.

Expected: Unrolled instances of modules would retain some context of original source. Also, need to track the original Range and corresponding location information.

alaindargelas, 2022-02-28:
Surelog primary concerns is to create a model that is executable by Synthesis (Yosys) and Simulation (Verilator). Elaboration has to happen to achieve that goal. 
Preserving the original source code representation only happens when it does not get in the way of Elaboration. Loop and instance unrolling will happen during elaboration. 
You can retrieve (reconstruct) the ranges from the instance indexes. Location information is present in UHDM.

hs-apotell, 2022-03-03:
> Preserving the original source code representation only happens when it does not get in the way of Elaboration. Loop and instance unrolling will happen during elaboration.

Agreed. But why is the non-elaborated model storing elaborated data. Shouldn't the non-elaborated model hold original source? Shouldn't the substitution and unrolling happen as part of elaboration process?

This is just one such instance but elaboration seems to be seeping into the non-elaboration side. `ArianeElab` is a good example. Compare the [attached file](https://github.com/chipsalliance/Surelog/files/8181300/dut.sv.log) with `dut.v` from ArianeElab. Attached file is generated from using the location information stored in the UHDM model and you can see how much of the data doesn't actually show up in non-elaborated model. 

Without this information none of the tools like Language Server, Lint, and alike would be possible.

alaindargelas, 2022-03-05:
You have to use both non-elaborated and elaborated model and you should have 99% of what you are looking for.
Adding the missing information to the non-elaborated model is double work (engineering work) and double the memory in UHDM.
BTW, you are welcome to contribute the engineering work if it is required by your project, but if you navigate both non-elaborated and elaborated, I think you are going to find most of the things you are looking for, if not, simply add to the non-elaborated model the non-elaborated part closer to source code.

hs-apotell, 2022-06-29:
> if you navigate both non-elaborated and elaborated, I think you are going to find most of the things you are looking for

This would be more work (having to roll back what has been unrolled during elaboration process) than just adding the necessary information to the non-elaborated tree. 

> simply add to the non-elaborated model the non-elaborated part closer to source code

We will do the engineering work to get the necessary information in the non-elaborated tree.

Specific question - In the non-elaborated tree would the specific instance in this test `ArrayInst` be a `module_array`? There are no other examples among any of the regression tests for this specific model. AFAIK none of the `XXX_array` are used in any of the regression tests.

hs-apotell, 2022-07-11:
Could you please provide some clarification/guidance on way forward for design. I am working on adding the missing pieces in non-elaborated model #2684 (instantiation of modules, gates, arrays, etc). Want to make sure I follow the committed path and not the abandoned one.

UHDM models have specialization for `variables` based on type (`bit_var`, `byte_var`,`int_typespec`, etc) and each of those specialization has the corresponding specialization (`bit_typespec`, `byte_typespec`, `int_typespec`, etc) for `typespec`.  Isn't this duplicating the type information? Is the `typespec` the way forward?

alaindargelas, 2022-07-12:
I'm about to check in an interface_array example with interfaces being sub elements of the array.