chipsalliance/Surelog #2685: Missing calls to instance::vpiTop

hs-apotell, 2022-02-28:
Assuming this call is to check whether an `instance` is in elaborated or non-elaborated hierarchy.

If not, how to disambiguate a `package` pointer in `VpiListener::enterPackage` call?

alaindargelas, 2022-03-03:
You don't have to listen to all the iterators in the design object, you can select what you want to listen to:
```
Instead of:

listen_designs(designs, linter);

You can do this:
  
  UHDM::VisitedContainer visited;
  if (vpiHandle itr = vpi_iterate(uhdmtopPackages, designHandle)) {
    while (vpiHandle obj = vpi_scan(itr)) {
      listen_any(obj, linter, &visited);
      vpi_free_object(obj);
    }
    vpi_free_object(itr);
  }
  if (vpiHandle itr = vpi_iterate(uhdmtopModules, designHandle)) {
    while (vpiHandle obj = vpi_scan(itr)) {
      listen_any(obj, linter, &visited);
      vpi_free_object(obj);
    }
    vpi_free_object(itr);
  }
  
```

alaindargelas, 2022-03-03:
But anyhow, I'll tag the top modules and top elaborated packages with the vpiTop attribute

From any object, you can also navigate up its parent tree and check check if the attribute is set.

alaindargelas, 2022-03-03:
Fixed by #2697

closed 2022-03-03 by alaindargelas