chipsalliance/Surelog #2689: typespec_member/long_int_typespec unsigned

hs-apotell, 2022-03-01:
For the snippet below -
```
typedef struct {
  longint unsigned pc;
} commit_log_t;
```
What UHDM field holds the sign-ness of the member itself? I didn't find anything in the `typespec_member` not in `long_int_typespec`. Here's the relevant UHDM dump output.
```
|vpiTypespecMember:
\_typespec_member: (pc), id:3406, line:662:26, endln:662:28, parent:riscv::commit_log_t, parID:3402
  |vpiName:pc
  |vpiTypespec:
  \_long_int_typespec: , id:3405, line:662:9, endln:662:16
    |vpiInstance:
    \_package: riscv (riscv::), id:0 dut.sv:1:1: , endln:671:11, parent:work@top, parID:15559
```

alaindargelas, 2022-03-05:
We need to add signeness to the Verilog model in UHDM

alaindargelas, 2022-03-23:
#2781 fix this problem for parameters and variables. I still have to add support for struct members.

alaindargelas, 2022-03-24:
Fixed by #2784

closed 2022-03-24 by alaindargelas