chipsalliance/Surelog #2714: parameter wrongly parsed as bit_typespec instead of logic_typespec

hs-apotell, 2022-03-06:
Test: FSMBsp13

https://github.com/chipsalliance/Surelog/blob/adbf40797649f890fdd79e35b6e6ffe30fdb2c72/tests/FSMBsp13/fsm1.v#L7-L10

https://github.com/chipsalliance/Surelog/blob/adbf40797649f890fdd79e35b6e6ffe30fdb2c72/tests/FSMBsp13/FSMBsp13.log#L6729-L6748

alaindargelas, 2022-03-27:
Both are wrong, the correct answer is unsigned:

6.20.2 Value parameters
A parameter constant can have a type specification and a range specification. The type and range of
parameters shall be in accordance with the following rules:
— A parameter declaration with no type or range specification shall default to the type and range of the
final value assigned to the parameter, after any value overrides have been applied. If the expression
is real, the parameter is real. If the expression is integral, the parameter is a logic vector of the
same size with range [size-1:0].
— A parameter with a range specification, but with no type specification, shall have the range of the
parameter declaration and shall be unsigned. The sign and range shall not be affected by value
overrides.
— A parameter with a type specification, but with no range specification, shall be of the type specified.
A signed parameter shall default to the range of the final value assigned to the parameter, after any
value overrides have been applied.
— A parameter with a signed type specification and with a range specification shall be signed and shall
have the range of its declaration. The sign and range shall not be affected by value overrides.
— A parameter with no range specification and with either a signed type specification or no type
specification shall have an implied range with an lsb equal to 0 and an msb equal to one less than the
size of the final value assigned to the parameter.
— A parameter with no range specification, with either a signed type specification or no type
specification, and for which the final value assigned to it is unsized shall have an implied range with
an lsb equal to 0 and an msb equal to an implementation-dependent value of at least 31.

alaindargelas, 2022-03-29:
Fixed by #2798

closed 2022-03-29 by alaindargelas