chipsalliance/Surelog #2715: nets::typespec_

hs-apotell, 2022-03-06:
Test: FSMBsp13

Issue: `nets::typespec_` is null

https://github.com/chipsalliance/Surelog/blob/adbf40797649f890fdd79e35b6e6ffe30fdb2c72/tests/FSMBsp13/fsm1.v#L5
```
\_logic_net: (work@top.F1.Read), line:5:5, endln:5:9, parent:work@top.F1
  |vpiName:Read
  |vpiFullName:work@top.F1.Read
  |vpiNetType:48
```

https://github.com/chipsalliance/Surelog/blob/adbf40797649f890fdd79e35b6e6ffe30fdb2c72/tests/FSMBsp13/fsm1.v#L11
```
\_logic_net: (work@top.F1.CurState), line:11:17, endln:11:25, parent:work@top.F1
  |vpiName:CurState
  |vpiFullName:work@top.F1.CurState
  |vpiNetType:48
  |vpiRange:
  \_range: , line:11:12, endln:11:15
    |vpiLeftRange:
    \_constant: , line:11:12, endln:11:13
      |vpiDecompile:3
      |vpiSize:64
      |UINT:3
      |vpiConstType:9
    |vpiRightRange:
    \_constant: , line:11:14, endln:11:15
      |vpiDecompile:0
      |vpiSize:64
      |UINT:0
      |vpiConstType:9
```

I need the location information for the data type and for `parameter`, the `typespec` holds that information. Shouldn't `nets` similarly have a valid `typespec` too?

Also, I am using the location information to decide whether two `parameter`s share the same type definition i.e.

```
parameter [3:0] ST_Read = 0, ST_Write = 1;
```
vs.
```
parameter [3:0] ST_Read = 0;
parameter [3:0] ST_Write = 1;
```

alaindargelas, 2022-03-29:
There is no typespec for nets of wire type.
There is the logic_typespec for reg type in the other hand.

alaindargelas, 2022-04-03:
#2814 should fix most of the problems with this issue. There  might be discussions of location information but that should be in the different issue, if needed.

closed 2022-04-03 by alaindargelas