chipsalliance/Surelog #2727: Attributes introducing discrepancies in location information

hs-apotell, 2022-03-09:
Looking for suggestions on how to resolve an issue with attributes.

In presence of attributes, the start line/column currently points to the start of the first attribute. In absence of it, the same points to the start of the keyword package (or module, program, etc). I would like this to be consistent and point always to the thing that the attribute is associated with i.e. the latter one. I looked at the code and the node representing the corresponding keyword is neither in the `VObjectType` enumeration nor the `FileContent::m_objects` has the corresponding node for it. How do I get to this information?

Also, I will have to use this new NodeId in DesignComponent::m_nodeIds instead of the current one. Is this something that would cause other problems (note that the new NodeId would not include the associated attributes)?

alaindargelas, 2022-03-10:
Example please.

hs-apotell, 2022-03-10:
```
(* packageWithAttribute *)
package package_with_attribute;
endpackage

package package_with_no_attribute;
endpackage

(* topWithAttribute *)
module top_with_attribute;

  (* innerWithAttribute *)
  module inner_with_attribute;
  endmodule
  
endmodule

module top_with_no_attribute;

  module inner_with_no_attribute;
  endmodule
  
endmodule

(* classWithAttribute *)
class class_with_attribute;

  (* innerClassWithAttribute *)
  class inner_class_with_attribute;
  endclass;

endclass

class class_with_no_attribute;

  class inner_class_with_no_attribute;
  endclass;

endclass

(* interfaceWithAttribute *)
interface interface_with_attribute;
 
  (* innerInterfaceWithAttribute *)
  interface inner_interface_with_attribute;
  endinterface;

endinterface

interface interface_with_no_attribute;
 
  interface inner_interface_with_no_attribute;
  endinterface;

endinterface
```

```
n<> u<0> t<Null_rule> p<114> s<113> l<1:1> el<1:0>
n<packageWithAttribute> u<1> t<StringConst> p<2> l<1:4> el<1:24>
n<> u<2> t<Attr_name> p<3> c<1> l<1:4> el<1:24>
n<> u<3> t<Attr_spec> p<4> c<2> l<1:4> el<1:24>
n<> u<4> t<Attribute_instance> p<7> c<3> s<5> l<1:1> el<1:27>
n<package_with_attribute> u<5> t<StringConst> p<7> s<6> l<2:9> el<2:31>
n<> u<6> t<Endpackage> p<7> l<3:1> el<3:11>
n<> u<7> t<Package_declaration> p<8> c<4> l<1:1> el<3:11>
n<> u<8> t<Description> p<113> c<7> s<12> l<1:1> el<3:11>
n<package_with_no_attribute> u<9> t<StringConst> p<11> s<10> l<5:9> el<5:34>
n<> u<10> t<Endpackage> p<11> l<6:1> el<6:11>
n<> u<11> t<Package_declaration> p<12> c<9> l<5:1> el<6:11>
[... 191 lines cut ...]
  |vpiDefName:work@top_with_attribute::inner_with_attribute
|uhdmallModules:
\_module: work@top_with_no_attribute (work@top_with_no_attribute), id:22 test.sv:17:1: , endln:22:10, parent:work@top_with_attribute, parID:13
  |vpiFullName:work@top_with_no_attribute
  |vpiDefName:work@top_with_no_attribute
|uhdmallModules:
\_module: work@top_with_no_attribute::inner_with_no_attribute (work@top_with_no_attribute::inner_with_no_attribute), id:23 test.sv:19:3: , endln:20:12, parent:work@top_with_attribute, parID:13
  |vpiFullName:work@top_with_no_attribute::inner_with_no_attribute
  |vpiDefName:work@top_with_no_attribute::inner_with_no_attribute
|uhdmtopModules:
\_module: work@top_with_attribute (work@top_with_attribute), id:24 test.sv:8:1: , endln:15:10
  |vpiName:work@top_with_attribute
  |vpiAttribute:
  \_attribute: (topWithAttribute), id:31, line:8:4, endln:8:20, parent:work@top_with_attribute, parID:24
    |vpiName:topWithAttribute
  |vpiDefName:work@top_with_attribute
  |vpiTop:1
  |vpiTopModule:1
|uhdmtopModules:
\_module: work@top_with_no_attribute (work@top_with_no_attribute), id:25 test.sv:17:1: , endln:22:10
  |vpiName:work@top_with_no_attribute
  |vpiDefName:work@top_with_no_attribute
  |vpiTop:1
  |vpiTopModule:1
```

hs-apotell, 2022-03-10:
As an example, Package_declaration includes attributes when available. I am looking for a node which never includes the attributes and always starts at the beginning of the keyword 'package'.

```
n<> u<7> t<Package_declaration> p<8> c<4> l<1:1> el<3:11>
n<> u<11> t<Package_declaration> p<12> c<9> l<5:1> el<6:11>
```

alaindargelas, 2022-03-10:
Like:
https://github.com/chipsalliance/Surelog/blob/fb4c954781e49e59e59a5b2e14beddbb4ea74a3e/src/DesignCompile/CompileExpression.cpp#L3649
You have to add attribute compilation everywhere they can appear and you are interested in modeling in UHDM.
It could be in the CompilePackage.cpp, CompileModule.cpp files, you'd have to probably save a pointer to the attribute in the corresponding object the attribute is the closest to.

In the AST, you can either navigate down object->child or 
navigate up and down to find the last object prior to the current object :object->parent->child so you can attach the attribute to the object you are currently compiling.

hs-apotell, 2022-03-10:
The attributes itself (and their location information as well) are getting parsed correctly. The issue is the position of attributes, when they do exist, get included in the thing they are closest to. From the above example -

```
\_package: package_with_attribute (package_with_attribute::), id:14 test.sv:1:1: , endln:3:11, parent:work@top_with_attribute, parID:13
  |vpiName:package_with_attribute
  |vpiFullName:package_with_attribute::
  |vpiAttribute:
  \_attribute: (packageWithAttribute), id:4, line:1:4, endln:1:24
    |vpiName:packageWithAttribute
  |vpiDefName:package_with_attribute
|uhdmallPackages:
\_package: package_with_no_attribute (package_with_no_attribute::), id:15 test.sv:5:1: , endln:6:11, parent:work@top_with_attribute, parID:13
  |vpiName:package_with_no_attribute
  |vpiFullName:package_with_no_attribute::
  |vpiDefName:package_with_no_attribute
```

notice that the location of `package_with_attribute` got offset-ed to include the attribute. I would like to not to do that. In the second case with `package_with_no_attribute` the location started with at the beginning of the keyword `package`. I would like this to be consistent. With attributes (and any number of them), the position of the actual _thing_ itself becomes unpredictable.

```
(* First Attribute *) (* Second Attribute *) package MyPackage; endpackage

(* First Attribute *)
(* Second Attribute *) package MyPackage; endpackage

(* First Attribute *)
(* Second Attribute *)
package MyPackage; endpackage
```

Three different ways of doing the same thing, declaring a package and all are legal. However, the start position of the `package` definition would always be at the beginning of "(* First Attribute *)" making the position of the _real_ package start i.e. the position of the keyword `package` ambiguous. There is no _node_ to designate the start of the _thing_ in the AST.

alaindargelas, 2022-03-10:
The AST strictly follows the grammar:
package_declaration  
    : ( attribute_instance )* PACKAGE identifier SEMICOLUMN 
    ( timeunits_declaration )? ( ( attribute_instance )* package_item )* 
    ENDPACKAGE ( COLUMN identifier )?  
    ; 

void SV3_1aTreeShapeListener::enterPackage_declaration(
    SV3_1aParser::Package_declarationContext *ctx) {
  std::string ident = ctx->identifier(0)->getText();
  ident = std::regex_replace(ident, std::regex(EscapeSequence), "");
  addDesignElement(ctx, ident, DesignElement::Package, VObjectType::slPackage);
}

You cannot change the grammar to introduce new arbitrary level that exclude the attribute_instance, but you can choose which object you add as the design element from the ctx context.

hs-apotell, 2022-03-10:
Ahh.. that's what I was looking for. So, the current implementation of `SV3_1aTreeShapeListener::exitPackage_declaration` doesn't add a `VObject` for `VObjectType::slPackage` (and similarly for module, interface, and others). It does only for `slPackage_declaration`. Can I introduce one for `VObjectType::slPackage` in the `FileContent::m_objects` collection or would that introduce other problems?? That would solve the issue. I can looking for that specific token type and use the location information from it to populate `UHDM::package`.

alaindargelas, 2022-03-11:
You can but you most likely will have to change some code in Surelog to account for in diverse parts.

Thomasb81, 2022-03-30:
> Three different ways of doing the same thing, declaring a package and all are legal. However, the start position of the `package` definition would always be at the beginning of "(* First Attribute [...]

What about getting the the position of the `PACKAGE` keyword from the ast instead of the position of `package_declaration` rule?

> package_declaration
> : ( attribute_instance )* PACKAGE identifier SEMICOLUMN
> ( timeunits_declaration )? ( ( attribute_instance )* package_item )*
> ENDPACKAGE ( COLUMN identifier )?
> ;

hs-apotell, 2022-04-04:
Fixed by #2796

closed 2022-04-04 by hs-apotell