chipsalliance/Surelog #2741: Allow for custom python version when building

kiteloopdesign, 2022-03-12:
Build process was erroring out due to `orderedmultidict` missing, although I had it installed in my system. The problem is that `make` seems to use the latest python it can find on `/usr/bin/` and in my case this was python3.11, while the version I use and the one under I had installed the required python module is python3.10.

I got around this issue by `sudo mv /usr/bin/python3.11 /usr/bin/python3.11.back', but I guess there may be some switch that can be passed on to cmake to specify the python version to be used?

alaindargelas, 2022-03-14:
@hs-apotell, please check this one.

hs-apotell, 2022-03-14:
It's not make it's cmake that decides the python version. To use a specific version of python you can set a global environment variable.

```
export ADDITIONAL_CMAKE_OPTIONS=-DPython3_ROOT_DIR=<location of installed python>
```

Optionally, if you are using cmake directly, then you can pass the same to cmake as well, as such.

```
cmake -DPython3_ROOT_DIR=<location of installed python> ........
```

If you have multiple version of python installed, you should consider using `update-alternatives`. Here's some help
https://linuxconfig.org/how-to-change-from-default-to-alternative-python-version-on-debian-linux

closed 2022-03-14 by hs-apotell

kiteloopdesign, 2022-03-14:
Thanks for the feeback