chipsalliance/Surelog #2742: Can't create my own test executable, problem with linker

kiteloopdesign, 2022-03-13:
I have successfully installed Surelog. I can run the surelog executable anywhere, and I have also been to run the following tests described on README.md

```
bin/surelog -help
bin/surelog -writepp -parse ../tests/UnitElabBlock/top.v
bin/hellosureworld ../tests/UnitElabBlock/top.v -parse -mutestdout
bin/hellouhdm ../tests/OneFF/dut.v -d uhdm -d coveruhdm -elabuhdm
```
However, I am getting errors when following these instructions

[image]

```
cmake ../ -DINSTALL_DIR=/usr/local
                                                                                                                                                                                                   
-- The C compiler identification is GNU 11.2.1
-- The CXX compiler identification is GNU 11.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kitesutra/cad/surelog/tests/TestInstall/build

make
                                                                                                                                                                                                                                                 
[ 50%] Building CXX object CMakeFiles/test_hellosureworld.dir/home/kitesutra/cad/surelog/src/hellosureworld.cpp.o
[100%] Linking CXX executable test_hellosureworld
/usr/bin/ld: cannot find -lsurelog
/usr/bin/ld: cannot find -lantlr4-runtime
/usr/bin/ld: cannot find -lflatbuffers
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_hellosureworld.dir/build.make:97: test_hellosureworld] Error 1
make[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/test_hellosureworld.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

Any help on how can I further debug this appreciated. Thanks

kiteloopdesign, 2022-03-13:
I am running above build from `surelog/tests/TestInstall/build` folder. 

Also note that I have run `sudo make install` from `surelog` top folder and it seems to have worked properly.

Note in case is of any help as I have read there is some issues with this OS: I am on Fedora 35

kiteloopdesign, 2022-03-13:
Adding more info. My installation folder is `/usr/local/include/Surelog` 
Is that uppercase S expected?

kiteloopdesign, 2022-03-13:
Apologies for the spamming, but found something interesting.

I realized that the surelog libs got installed on `/usr/local/lib64` instead of `/usr/local/lib` (must be a Fedora thing). I created a softlink on the latter to point to the former and then make command went pass that error to show a bunch of different ones. They seem related now with Python (I compiled with SURELOG_WITH_PYTHON option).

```
make                                                                                                                                                                                                                                                        master 
[ 50%] Building CXX object CMakeFiles/test_hellosureworld.dir/home/kitesutra/cad/surelog/src/hellosureworld.cpp.o
[100%] Linking CXX executable test_hellosureworld
/usr/bin/ld: /usr/local/lib/surelog/libsurelog.a(PythonAPI.cpp.o): in function `SWIG_Python_ErrorType(int)':
PythonAPI.cpp:(.text+0x1b): undefined reference to `PyExc_IOError'
/usr/bin/ld: PythonAPI.cpp:(.text+0x23): undefined reference to `PyExc_MemoryError'
/usr/bin/ld: PythonAPI.cpp:(.text+0x33): undefined reference to `PyExc_AttributeError'
/usr/bin/ld: PythonAPI.cpp:(.text+0x43): undefined reference to `PyExc_SystemError'
/usr/bin/ld: PythonAPI.cpp:(.text+0x53): undefined reference to `PyExc_ValueError'
/usr/bin/ld: PythonAPI.cpp:(.text+0x63): undefined reference to `PyExc_SyntaxError'
/usr/bin/ld: PythonAPI.cpp:(.text+0x73): undefined reference to `PyExc_OverflowError'
/usr/bin/ld: PythonAPI.cpp:(.text+0x83): undefined reference to `PyExc_ZeroDivisionError'
/usr/bin/ld: PythonAPI.cpp:(.text+0x93): undefined reference to `PyExc_TypeError'
/usr/bin/ld: PythonAPI.cpp:(.text+0xa3): undefined reference to `PyExc_IndexError'
/usr/bin/ld: PythonAPI.cpp:(.text+0xb3): undefined reference to `PyExc_RuntimeError'
/usr/bin/ld: /usr/local/lib/surelog/libsurelog.a(PythonAPI.cpp.o): in function `SwigPyObject_disown':
PythonAPI.cpp:(.text+0xc3): undefined reference to `_Py_NoneStruct'
...

```

kiteloopdesign, 2022-03-13:
Last piece of info: I re-built Surelog with no python support to make sure that was not the cause of the errors on above post. I am getting those same errors in this case, so I am totally lost here...

Any pointer on how to further debug this is greatly appreciated

alaindargelas, 2022-03-14:
Can you run:
sudo make install
make test_install

For Info:
Makefile test_install target:
test_install:
	cmake -DCMAKE_BUILD_TYPE=Release -DINSTALL_DIR=$(PREFIX) -S tests/TestInstall -B tests/TestInstall/build
	cmake --build tests/TestInstall/build -j $(CPU_CORES)

kiteloopdesign, 2022-03-14:
Hi Alain,
looks like I am getting similar errors (python-related) to the ones described on above comment (https://github.com/chipsalliance/Surelog/issues/2742#issuecomment-1066128851). See attached logfile for result of `make test_install` (`sudo make install` works just fine)

Is this expected? I mean, this test is done with a build with no SURELOG_WITH_PYTHON option, so why is this test is looking for python stuff?

[test_install.log](https://github.com/chipsalliance/Surelog/files/8244435/test_install.log)

alaindargelas, 2022-03-14:
I think you have a "history" problem with your previous builds attempts and the current state is not what you think it is, please try:

make clean
make 
sudo make install
make test_install

kiteloopdesign, 2022-03-15:
Thanks Alain. To make sure, I not only `sudo make uninstall`, but I completely started from scratch, re-cloning Surelog, and I am yet issuing this error (which I was also seeing before) when doing `make install`

```
make test_install                                                                                                                                                                      
cmake -DCMAKE_BUILD_TYPE=Release -DINSTALL_DIR=/usr/local -S tests/TestInstall -B tests/TestInstall/build
-- The C compiler identification is GNU 11.2.1
-- The CXX compiler identification is GNU 11.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/kitesutra/cad/Surelog/tests/TestInstall/build
cmake --build tests/TestInstall/build -j 8
gmake[1]: Entering directory '/home/kitesutra/cad/Surelog/tests/TestInstall/build'
gmake[2]: Entering directory '/home/kitesutra/cad/Surelog/tests/TestInstall/build'
gmake[3]: Entering directory '/home/kitesutra/cad/Surelog/tests/TestInstall/build'
gmake[3]: Leaving directory '/home/kitesutra/cad/Surelog/tests/TestInstall/build'
gmake[3]: Entering directory '/home/kitesutra/cad/Surelog/tests/TestInstall/build'
[ 50%] Building CXX object CMakeFiles/test_hellosureworld.dir/home/kitesutra/cad/Surelog/src/hellosureworld.cpp.o
[100%] Linking CXX executable test_hellosureworld
/usr/bin/ld: cannot find -lsurelog
/usr/bin/ld: cannot find -lantlr4-runtime
/usr/bin/ld: cannot find -lflatbuffers
collect2: error: ld returned 1 exit status
gmake[3]: *** [CMakeFiles/test_hellosureworld.dir/build.make:97: test_hellosureworld] Error 1
gmake[3]: Leaving directory '/home/kitesutra/cad/Surelog/tests/TestInstall/build'
gmake[2]: *** [CMakeFiles/Makefile2:85: CMakeFiles/test_hellosureworld.dir/all] Error 2
gmake[2]: Leaving directory '/home/kitesutra/cad/Surelog/tests/TestInstall/build'
gmake[1]: *** [Makefile:91: all] Error 2
gmake[1]: Leaving directory '/home/kitesutra/cad/Surelog/tests/TestInstall/build'
make: *** [Makefile:115: test_install] Error 2
```
This is strange because during the `make` process, a `bin/hellosureworld` (which I assume is the same/similar test?) executable is properly generated (for which I presume the linker is involved as well?), so I do not see how this one can't find the linker. 

Also note that I can properly run

```
bin/surelog -help
bin/surelog -writepp -parse ../tests/UnitElabBlock/top.v
bin/hellosureworld ../tests/UnitElabBlock/top.v -parse -mutestdout
bin/hellouhdm ../tests/OneFF/dut.v -d uhdm -d coveruhdm -elabuhdm
```

So maybe this is some incorrect makefile setup on the top surelog folder? Please let me know which logfile with extra debug info I can generate/look into for further debugging this. I do not have the time right now to go blindly at it, but I can do if you point me into where to look at

As a side comment, note that `make uninstall` recipe doesn't get rid of `uhdm` files. It is just removing

```
rm -f -r /usr/local/bin/surelog
rm -f -r /usr/local/lib/surelog
rm -f -r /usr/local/include/surelog
```
Also note that, for Fedora, Surelog lib files are in `/usr/local/lib64`, and so remain in the system. I manually deleted them anyway.

kiteloopdesign, 2022-03-15:
Ok, good news. I found what the issue is

`surelog -> ../lib64/surelog` link needs to be created on `/usr/local/lib` so the linker can find the missing .a files

kiteloopdesign, 2022-03-15:
On these instructions

[image]

what CmakeList.txt file are you referring to? note this link is pointing to

https://github.com/chipsalliance/Surelog/blob/master/tests/TestInstall/CMakeLists.txt

but there is also other CMakeList.txt file on the top Surelog folder

What I am trying to say is, how can I now make sure that I can use the Surelog library to create my own executable? Would it be possible to get a minimal makefile and dummy cpp which `\`includes` the surelog library? I am just very confused by the amount of cmakefiles, so I would like to do this outside Surelog folder with my own makefile.

kiteloopdesign, 2022-03-15:
Just to confirm that building the test from `tests/TestInstall/build` also works now

[ 50%] Building CXX object CMakeFiles/test_hellosureworld.dir/home/kitesutra/cad/Surelog/src/hellosureworld.cpp.o
[100%] Linking CXX executable test_hellosureworld
[100%] Built target test_hellosureworld
VERSION: 1.25
BUILT  : Mar 15 2022
[100%] Built target RunInstallTest

Question on above post remains anyway, ie. if I want to create a separate project which makes use of Surelog, what minimal makefile (or cmake?) do I need and how to set it up? 

Thanks

alaindargelas, 2022-03-15:
Good to hear. All these issues are due to Fedora not being tested. Ubuntu, Windows and Macos are run the the CI.
You should use https://github.com/chipsalliance/Surelog/blob/master/tests/TestInstall/CMakeLists.txt as an example Cmake, it is self contained.
And one of the hello*.cpp as you main, you need to use the UHDM API to develop whatever application you plan to develop.

kiteloopdesign, 2022-03-17:
Thanks Alain. I've been able to get it working. Hopefully this issue serves to people having problems with Fedora

closed 2022-03-17 by kiteloopdesign