chipsalliance/Surelog #2743: Error when installing via Nix on Fedora

kiteloopdesign, 2022-03-13:
Since I can't seem to get Surelog working under Fedora 35 (see #2742), I tried to install it via Nix. 

I successfully run `nix-shell` but then `make` is failing... 

```
[nix-shell:~/cad/Surelog]$ make
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_RULE_MESSAGES=on  -S . -B build
-- The C compiler identification is GNU 11.2.1
-- The CXX compiler identification is GNU 11.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - broken
CMake Error at /nix/store/paviiys71gp7bizzc57d1rdc9w4y79vg-cmake-3.19.7/share/cmake-3.19/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/kitesutra/cad/Surelog/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/gmake cmTC_caf54/fast && gmake[1]: Entering directory '/home/kitesutra/cad/Surelog/build/CMakeFiles/CMakeTmp'
    /usr/bin/gmake  -f CMakeFiles/cmTC_caf54.dir/build.make CMakeFiles/cmTC_caf54.dir/build
    gmake[2]: Entering directory '/home/kitesutra/cad/Surelog/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_caf54.dir/testCCompiler.c.o
    /usr/bin/cc    -o CMakeFiles/cmTC_caf54.dir/testCCompiler.c.o -c /home/kitesutra/cad/Surelog/build/CMakeFiles/CMakeTmp/testCCompiler.c
    Linking C executable cmTC_caf54
    /nix/store/paviiys71gp7bizzc57d1rdc9w4y79vg-cmake-3.19.7/bin/cmake -E cmake_link_script CMakeFiles/cmTC_caf54.dir/link.txt --verbose=1
    /usr/bin/cc CMakeFiles/cmTC_caf54.dir/testCCompiler.c.o -o cmTC_caf54 
    /nix/store/qvc6cz5d43jhhm4a3r48xkrc82xh1s95-binutils-2.35.1/bin/ld: /usr/libexec/gcc/x86_64-redhat-linux/11/liblto_plugin.so: error loading plugin: /nix/store/ikl21vjfq900ccbqg1xasp83kadw6q8y-glibc-2.32-46/lib/libc.so.6: version `GLIBC_2.33' not found (required by /usr/libexec/gcc/x86_64-redhat-linux/11/liblto_plugin.so)
    collect2: error: ld returned 1 exit status
    gmake[2]: *** [CMakeFiles/cmTC_caf54.dir/build.make:106: cmTC_caf54] Error 1
    gmake[2]: Leaving directory '/home/kitesutra/cad/Surelog/build/CMakeFiles/CMakeTmp'
    gmake[1]: *** [Makefile:140: cmTC_caf54/fast] Error 2
    gmake[1]: Leaving directory '/home/kitesutra/cad/Surelog/build/CMakeFiles/CMakeTmp'
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:21 (project)

-- Configuring incomplete, errors occurred!
See also "/home/kitesutra/cad/Surelog/build/CMakeFiles/CMakeOutput.log".
See also "/home/kitesutra/cad/Surelog/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:41: run-cmake-release] Error 1
```

alaindargelas, 2022-03-14:
Please check this PR:
https://github.com/chipsalliance/Surelog/pull/1606 and contact the contributor if any issues.

kiteloopdesign, 2022-03-14:
Thanks Alain. I actually had already seen that one already, I will follow it up on there.

sjkelly, 2022-03-14:
It looks like you are missing a C/C++ compiler. I guess this was missed in the nix shell environment.

alaindargelas, 2022-03-14:
@sjkelly, please submit a PR so others don't get tripped.

sjkelly, 2022-03-15:
@kiteloopdesign have you had luck? I note with the `shell.nix` on master the following:
```
steve:~/src/Surelog (master=)$ nix-shell --pure

[nix-shell:~/src/Surelog]$ which gcc
/nix/store/gg2rq3hrl3rf92nq6dnqhdyyxaa89aqf-gcc-wrapper-10.3.0/bin/gcc
```
So it does not appear that a patch is necessary since GCC is provided by Nix.

I am using Ubuntu, which makes me think on Fedora SELinux may be involved. You may also try running with `nix-shell --pure`

kiteloopdesign, 2022-03-15:
Thanks Alain and Steve.

> Fedora SELinux may be involved

Indeed I had a problem with this, seems to be happening to all people in Fedora. It meant I could not install Nix in all-user mode setup, but was able to do for single user mode. But, yeah, it might still be doing something funny here...

In any case, gcc seems to be being found

```
[nix-shell:~/cad/Surelog]$ which gcc
/nix/store/gg2rq3hrl3rf92nq6dnqhdyyxaa89aqf-gcc-wrapper-10.3.0/bin/gcc
```
Anyway, I think we can leave this here so far... I do not want to push on Nix too much as I do prefer to get Surelog working directly on Fedora. Otherwise I still have the options of Ubuntu running on VM or NixOS