chipsalliance/Surelog #2760: Elaboration reports internal error

hs-apotell, 2022-03-18:
Test: AmiqSimpleTestSuite
Run: uhdm-dump surelog.uhdm --elab

Log dump shows following repeated entries -
Internal Error: adding wrong object type (type_parameter) in a actual_group group!

Also, following legal assignment is reported with both lhs and rhs being null.
https://github.com/chipsalliance/Surelog/blob/7a9ef9e1bb9ac6088e98b475e267e9b959459eee/third_party/UVM/1800.2-2017-1.0/src/reg/uvm_vreg.svh#L456

alaindargelas, 2022-03-18:
UVM is not supported in UHDM yet and won't be for a while.

alaindargelas, 2022-03-18:
Same as https://github.com/chipsalliance/Surelog/issues/938

hs-apotell, 2022-03-18:
Can we make this explicitly clear in the logs. I didn't see any error messages except for the above one that explicitly states that UVM isn't supported yet.

alaindargelas, 2022-03-18:
Impossible, that equates to know that you are dealing with UVM or classes at least, which means you starting supporting at least something about it. Might as well do the work, if you have time please do.
This error message says you are attempting to do something wrong. It has no idea about the context.

hs-apotell, 2022-03-18:
This is what I had in mind when I requested - Wouldn't having a file like `uvm_pkg.sv` in the file list to parse a sufficient check? We aren't stopping anyone from doing what they need to do. But having a warning in the log reading something along the line "You seem to be using UVM which is not fully supported by Surelog." would inform the user. Or even allowing UVM paths only with an additional command like flag.

But again if the warning itself is not tied to the error being reported it is likely going to get overlooked anyways. Nevermind, forget it I mentioned. Not worth the effort.

closed 2022-04-10 by alaindargelas