chipsalliance/Surelog #2761: bit_select/part_select :: VpiParent

hs-apotell, 2022-03-18:
I am finding some contradictions in code about what `VpiParent` for `bit_select` and `part_select` be. Considering the following two statements -

```
1: logic [1:0] var;
2: var[1] = 1`b1;
```

* Should it be variable `var` as defined on line 1?
* Should it be `assignment` statement on line 2?
* Should it be that `ref_obj::vpiActual` points back to the corresponding `bit_select` or `part_select` (Seems pointless because this just creates a loop, but I found at least one such case).

Option two makes the most logical sense and is inline with the concept of _parent_ in rest of the UHDM model but then there is a missing piece to the puzzle. How do I get to the variable `var` defined on line 1 from a `bit_select` pointer. Seems like we are missing a property on both `bit_select` and `part_select`.

@alaindargelas Could you please comment on this. I am kind of stuck on this and is blocking my progress.

hs-apotell, 2022-03-19:
I think I figured it out. Please confirm -

[bit|part]_select -> VpiParent -> ref_obj -> VpiParent -> assignment
[bit|part]_select -> VpiParent -> ref_obj -> VpiActual -> var

alaindargelas, 2022-03-19:
That is correct, it is so much easier to reason on a real testcase (That compiles):
```
module top();

logic [2:0] var1;
assign var1[1:0] = 1'b10;

endmodule

Set breakpoint at: 
CompileHelper.cpp:2121  -- setParentNoOverride(rhs_exp, cassign);

You can then introspect in the debugger all the objects and see the relations you can't see with the global UHDM dump (else it would loop while dumping):

p UHDM::decompile(lhs_exp)
$1 = "decompile:\npart_select: , id:2, line:4:8, endln:4:17, parent:var1, parID:3\n|vpiParent:\n\\_ref_obj: var1 (var1), id:3, parID:5\n|vpiConstantSelect:1\n|vpiLeftRange:\n\\_constant: , id:0, line:4:13, endln:4:"...

p UHDM::decompile(lhs_exp->VpiParent())
$2 = "decompile:\nref_obj: var1 (var1), id:3, parID:5\n|vpiName:var1\n|vpiDefName:var1\n"

p UHDM::decompile(cassign)
$3 = "decompile:\ncont_assign: , id:5\n|vpiRhs:\n\\_constant: , id:4, line:4:20, endln:4:25\n  |vpiDecompile:1'b10\n  |vpiSize:1\n  |BIN:10\n  |vpiConstType:3\n|vpiLhs:\n\\_part_select: , id:2, line:4:8, endln:4:17, p"...

```

alaindargelas, 2022-03-22:
Fixed by #2777

closed 2022-03-22 by alaindargelas