chipsalliance/Surelog #2789: Anonymous enums are duplicated

rkapuscik, 2022-03-25:
For the following declaration:
```verilog
    enum {IDLE, INIT, SET_TIME, COUNTDOWN, DONE} state, state_next;
```
there are two enum typedefs in UHDM (note different vpi_ids):
```
  |vpiVariables:
  \_enum_var: (work@top.state), id:83, line:2:50, endln:2:55, parent:work@top, parID:100
    |vpiTypespec:
    \_enum_typespec: , id:77
      |vpiEnumConst:
      \_enum_const: (IDLE), id:78, line:2:11, endln:2:15, parID:77
      |vpiEnumConst:
      \_enum_const: (INIT), id:79, line:2:17, endln:2:21, parID:77
        |vpiName:INIT
        |INT:1
        |vpiDecompile:1
        |vpiSize:64
      |vpiEnumConst:
      \_enum_const: (SET_TIME), id:80, line:2:23, endln:2:31, parID:77
        |vpiName:SET_TIME
        |INT:2
        |vpiDecompile:2
        |vpiSize:64
      |vpiEnumConst:
      \_enum_const: (COUNTDOWN), id:81, line:2:33, endln:2:42, parID:77
        |vpiName:COUNTDOWN
        |INT:3
        |vpiDecompile:3
        |vpiSize:64
      |vpiEnumConst:
      \_enum_const: (DONE), id:82, line:2:44, endln:2:48, parID:77
        |vpiName:DONE
        |INT:4
        |vpiDecompile:4
        |vpiSize:64
    |vpiName:state
    |vpiFullName:work@top.state
    |vpiAutomatic:1
    |vpiVisibility:1
  |vpiVariables:
  \_enum_var: (work@top.state_next), id:90, line:2:57, endln:2:67, parent:work@top, parID:100
    |vpiTypespec:
    \_enum_typespec: , id:84
      |vpiEnumConst:
      \_enum_const: (IDLE), id:85, line:2:11, endln:2:15, parID:84
        |vpiName:IDLE
        |INT:0
        |vpiDecompile:0
        |vpiSize:64
      |vpiEnumConst:
      \_enum_const: (INIT), id:86, line:2:17, endln:2:21, parID:84
        |vpiName:INIT
        |INT:1
        |vpiDecompile:1
        |vpiSize:64
      |vpiEnumConst:
      \_enum_const: (SET_TIME), id:87, line:2:23, endln:2:31, parID:84
        |vpiName:SET_TIME
        |INT:2
        |vpiDecompile:2
        |vpiSize:64
      |vpiEnumConst:
      \_enum_const: (COUNTDOWN), id:88, line:2:33, endln:2:42, parID:84
        |vpiName:COUNTDOWN
        |INT:3
        |vpiDecompile:3
        |vpiSize:64
      |vpiEnumConst:
      \_enum_const: (DONE), id:89, line:2:44, endln:2:48, parID:84
        |vpiName:DONE
        |INT:4
        |vpiDecompile:4
        |vpiSize:64
    |vpiName:state_next
    |vpiFullName:work@top.state_next
    |vpiAutomatic:1
    |vpiVisibility:1
```
This causes conflicts as the identifiers are duplicated.

alaindargelas, 2022-03-25:
There is no other way to model that in the Verilog model.
These are not typedefs, so they need to be declared that way. You need to recognize that it is a "duplicate" and don't send the duplicated forward to Yosys/Verilator.

hs-apotell, 2022-03-26:
This problem is not just with unnamed enums. This is a problem for all `typespec`s since they are all clones. Enums and Structs are just more obvious.

alaindargelas, 2022-03-26:
The "compare" function should take care of that, it was supposed to be used for this purpose, comparing two UHDM objects

alaindargelas, 2022-03-31:
I found another solution, I'll cache the typespec at the instance level and reuse the same typespec when in cache.

alaindargelas, 2022-03-31:
Fixed by #2804

closed 2022-03-31 by alaindargelas