chipsalliance/Surelog #2799: Feature Request: Simpler C Like Compilation Model

QuantamHD, 2022-03-29:
At Google we have a build system (Bazel) that handles the incremental compilation and distributed execution.

It would be very advantageous if we had a compilation model that was more c like.

So giving sure a command like `surelog ... a.v` and getting a file `a.o` that we could link together in a final step.

alaindargelas, 2022-03-31:
Actually after giving it much thought, I think you can already achieve this result:

Given a list of files path1/file1.sv .... pathN/fileN.sv

1) Generate all the Post-processed files and parse the files in parallel
  a) surelog path1/file1.sv -writepp -fileunit                                                (Generate the post-processed text file and its cache)   
  b) surelog  -fileunit  -parseonly  slpp_unit/work/<hash_path1>/file1.sv   (Parse the post-processed text file and its cache)
...
  a) surelog pathN/fileN.sv -writepp -fileunit
  b) surelog  -fileunit  -parseonly  slpp_unit/work/<hash_pathN>/fileN.sv

2) Link everything (reloads all the files from cache and elaborates):
surelog  -fileunit  path1/file1.sv .... pathN/fileN.sv -parse (-elabuhdm)

QuantamHD, 2022-03-31:
Having an option to write the files next to the source file would be very ideal. so if you have a file a/b/c/d.sv after running both commands you would have a/b/c/d.sv.slpp and a/b/c/d.sv.slpa. If surelog could read those file first in some alternate mode that would make things really easy.

alaindargelas, 2022-04-01:
First option where let's say you have two files a/b/c1/d1.sv   a/b/c2/d2.sv from different directories and have their cache saves in respective directories then reassemble the cache from these directories looks very hard in the surelog flow.

For the 2nd option:
The link (elaboration step) is expecting all the cache files rooted in the same location. I'll verify that we can get this to work, if I can get that to work, is that acceptable?

a) surelog path1/file1.sv -writepp -fileunit -o  /a/b/c/d/cache_path
b) surelog -fileunit -parseonly slpp_unit/work/<hash_path1>/file1.sv  -o  /a/b/c/d/cache_path
...
a) surelog pathN/fileN.sv -writepp -fileunit  -o  /a/b/c/d/cache_path
b) surelog -fileunit -parseonly slpp_unit/work/<hash_pathN>/fileN.sv  -o  /a/b/c/d/cache_path
 

surelog -fileunit path1/file1.sv .... pathN/fileN.sv -parse   -o  /a/b/c/d/cache_path

alaindargelas, 2022-04-01:
The -cache <path> option does not really help, it moves where the cache is located, but not where the post-processed files are and also mangles the file names.

alaindargelas, 2022-04-01:
Another solution:

surelog file1.v   -fileunit -parse -noelab -nouhdm -nobuiltin -nocomp  (-o optional_path)
...
surelog fileN.v -fileunit  -parse -noelab -nouhdm -nobuiltin -nocomp  (-o optional_path)

surelog  file1.v ...  fileN.v -parse  (-o optional_path)

If that works, I can create an option like -sepcomp (Separate compile) that does the same as : -fileunit -parse -noelab -nouhdm -nobuiltin -nocomp

QuantamHD, 2022-04-01:
It sounds like my proposal would be hard to do. Maybe a surelog option that would merge N cache directories would be easier, that would also work for me?

alaindargelas, 2022-04-01:
@QuantamHD I sent you a meeting invite, I'm missing some information why the latest proposal would not work as-is.

alaindargelas, 2022-04-01:
Use the -nohash  #2811, it removes the hash and treats the cache as golden.
You should use the sequence:
surelog file1.v -fileunit -parse -noelab -nouhdm -nobuiltin -nocomp -nohash
...
surelog fileN.v -fileunit -parse -noelab -nouhdm -nobuiltin -nocomp -nohash

Copy over the caches in central location

surelog file1.v ... fileN.v -parse -nohash

alaindargelas, 2022-04-04:
Actually you need #2821

QuantamHD, 2022-04-12:
@alaindargelas In bazel you always need to create the files you declare as outputs. When the preprocessor makes an entire file blank it doesn't emit an slpa file in the cache which causes an error. Can we make a zero byte slpa file that surelog ignores in this case?

alaindargelas, 2022-04-12:
@QuantamHD, please add the actual command line you are using, and the input file was an empty file?

QuantamHD, 2022-04-12:
It looks like the parse cache isn't writing to the write folder.

QuantamHD, 2022-04-12:
```
svFileName = svFileName.parent_path().filename() / baseFileName;
```

Specifically this line in the ParseCache.cpp file. It's not correct under a -nohash flag

alaindargelas, 2022-04-12:
Can you either propose a fix or a concrete testcase to reproduce ? 
Exact command line, sample input file

alaindargelas, 2022-04-13:
That is not legal, the preprocessor fails, there is no condition for the ifdef
Error code 1 is FATAL/SYNTAX ERROR

[SNT:PP0106] tests/ScratchPad.sv:2:6: Syntax error: mismatched input '\n' expecting Spaces,
`ifdef
      ^-- tests/ScratchPad.sv:2:6:.

hzeller, 2022-04-13:
That file might have a syntax error, but it actually happens with any file:

```
touch foo.sv ; surelog -noelab -nouhdm -nobuiltin -nocomp -nohash -verbose -parse foo.sv ; echo "--- EXIT CODE: $?"
[INF:CM0023] Creating log file ./slpp_all/surelog.log.

[INF:PP0122] Preprocessing source file "foo.sv".

[INF:PA0201] Parsing source file "foo.sv".

[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 0
--- EXIT CODE: 1
```

slpp_all also looks good
```
find slpp_all/
slpp_all/
slpp_all/work
slpp_all/work/foo.sv
slpp_all/cache
slpp_all/cache/work
slpp_all/cache/work/foo.sv.slpp
slpp_all/surelog.log
slpp_all/file.lst
slpp_all/file_map.lst
```

So processing works fine, no errors are reported but the exit code is 1 for some reason.

alaindargelas, 2022-04-13:
-nohash does not work for a file on the current dir.... bug

QuantamHD, 2022-04-14:
slpa file generate is dumping all the files into the slpp_all/work directory they need to be sent to a folder representing their path.

a/b/c/d.v should result in these files

slpp_all/cache/work/a/b/c/d.v.slpp
slpp_all/work/a/b/c/d.v.slpa

Right now it results in:

slpp_all/cache/work/a/b/c/d.v.slpp
slpp_all/work/d.v.slpa

QuantamHD, 2022-04-15:
@alaindargelas This is working as intended. The only question I have is it possible to avoid reading the preprocessor cache? We know that we don't plan to change any define options once in the linking phase. This would help reduce the memory pressure on the process.

alaindargelas, 2022-04-16:
Yes, with #2863 the slpp only reloads the error/warning messages and not the bulk of the data.

closed 2022-04-16 by alaindargelas

alaindargelas, 2022-05-22:
@QuantamHD, with https://github.com/chipsalliance/Surelog/pull/2956
You should be able to simply call:

surelog -init

surelog -sepcomp -nohash .... file set 1 ....             \  In parallel
surelog -sepcomp -nohash ...... file set n .....          /

surelog -link -nohash    (-elabuhdm)

See README, SEPARATE COMPILATION FLOW OPTIONS
And example:
https://github.com/chipsalliance/Surelog/blob/master/tests/TestSepCompNoHash/runsurelog.sh

hzeller, 2022-05-23:
I think the README needs an example like that.