chipsalliance/Surelog #2802: regression.py: Python process that should wait for surelog to finish uses 100% CPU

hzeller, 2022-03-31:
Running the Python regression script, say with a few tests
```
python3 scripts/regression.py run --filters "BuildUVMPkg|Ariane"
```

... results not only in the surelog process it forks to use 100% CPU, but also almost 100% CPU for the python3 process that 
```
    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND 
3742259 hzeller   20   0  870480 840952   9988 R 100.0   1.3   0:07.65 surelog
3742289 hzeller   20   0  681904 663552  10868 R  99.7   1.0   0:07.54 surelog
3742226 hzeller   20   0   23364  15664   4948 R  99.3   0.0   0:07.59 python3
3742223 hzeller   20   0   23392  15848   4948 R  99.0   0.0   0:07.60 python3
```

Looking at an strace output of the python process, it looks like it is scanning through all processes in /proc, almost  as if it attempts to do some sort of `top` equivalent ? All it needs to do is `waitpid()` of sorts and don't use _any_ CPU time.
```
openat(AT_FDCWD, "/proc/1797/stat", O_RDONLY|O_CLOEXEC) = 19
newfstatat(19, "", {st_mode=S_IFREG|0444, st_size=0, ...}, AT_EMPTY_PATH) = 0
ioctl(19, TCGETS, 0x7fff7465e210)       = -1 ENOTTY (Inappropriate ioctl for device)
lseek(19, 0, SEEK_CUR)                  = 0
lseek(19, 0, SEEK_CUR)                  = 0
newfstatat(19, "", {st_mode=S_IFREG|0444, st_size=0, ...}, AT_EMPTY_PATH) = 0
read(19, "1797 (plasmashell) S 1 1796 1796"..., 8192) = 362
read(19, "", 7830)                      = 0
close(19)                               = 0
openat(AT_FDCWD, "/proc/1799/stat", O_RDONLY|O_CLOEXEC) = 19
newfstatat(19, "", {st_mode=S_IFREG|0444, st_size=0, ...}, AT_EMPTY_PATH) = 0
ioctl(19, TCGETS, 0x7fff7465e210)       = -1 ENOTTY (Inappropriate ioctl for device)
lseek(19, 0, SEEK_CUR)                  = 0
lseek(19, 0, SEEK_CUR)                  = 0
newfstatat(19, "", {st_mode=S_IFREG|0444, st_size=0, ...}, AT_EMPTY_PATH) = 0
read(19, "1799 (xembedsniproxy) S 1 1798 1"..., 8192) = 321
read(19, "", 7871)                      = 0
close(19)                               = 0
openat(AT_FDCWD, "/proc/1805/stat", O_RDONLY|O_CLOEXEC) = 19
newfstatat(19, "", {st_mode=S_IFREG|0444, st_size=0, ...}, AT_EMPTY_PATH) = 0
ioctl(19, TCGETS, 0x7fff7465e210)       = -1 ENOTTY (Inappropriate ioctl for device)
lseek(19, 0, SEEK_CUR)                  = 0
lseek(19, 0, SEEK_CUR)                  = 0
newfstatat(19, "", {st_mode=S_IFREG|0444, st_size=0, ...}, AT_EMPTY_PATH) = 0
read(19, "1805 (org_kde_powerde) S 1 1804 "..., 8192) = 190
read(19, "", 8002)                      = 0
close(19)                               = 0
openat(AT_FDCWD, "/proc/1811/stat", O_RDONLY|O_CLOEXEC) = 19
newfstatat(19, "", {st_mode=S_IFREG|0444, st_size=0, ...}, AT_EMPTY_PATH) = 0
ioctl(19, TCGETS, 0x7fff7465e210)       = -1 ENOTTY (Inappropriate ioctl for device)
lseek(19, 0, SEEK_CUR)                  = 0
lseek(19, 0, SEEK_CUR)                  = 0
newfstatat(19, "", {st_mode=S_IFREG|0444, st_size=0, ...}, AT_EMPTY_PATH) = 0
read(19, "1811 (kaccess) S 1 1810 1810 0 -"..., 8192) = 315
read(19, "", 7877)                      = 0
close(19)                               = 0

 ... and so on
```

So in the end, this will use twice as many CPU cores as requested.

hs-apotell, 2022-04-01:
It's not scanning all processes. It is scanning only for the child processes.

https://github.com/chipsalliance/Surelog/blob/6007db6f8ed48f48fccd6bdb9144cb200de52bf9/scripts/regression.py#L325

The `process` variable here is the running surelog process. 

Moreover, there's a yield after every iteration.
https://github.com/chipsalliance/Surelog/blob/6007db6f8ed48f48fccd6bdb9144cb200de52bf9/scripts/regression.py#L356

This is the _recommended_ way to use psutil. I am not familiar with the internals of the psutil itself. It could be that the library scans for all processes and filters based on the input process id. You can play around with the sleep time to reduce the processor overload. Careful, the larger the sleep, the farther you deviate from _true_ peak.

hzeller, 2022-04-01:
From what I understand, this should start a new sub-process and wait until it is finished. So we don't need to poll anything, just wait until the child process is finished.

The underlying implementation in Python should just do [`waidpid()`](https://linux.die.net/man/2/waitpid) and whatever the equivalent is on other operating systems.  Looks like there is [something like that](https://docs.python.org/3/library/os.html#os.waitpid) there.

hs-apotell, 2022-04-03:
There seems to be some misunderstanding here. The main process is polling to collect memory and CPU statistics of sub-processes. It's not polling for job to finish. The external processes (surelog and uhdm-dump) are being started as sub-processes but with a different API than what you are suggesting because I need access to the subprocess pid.

hzeller, 2022-04-03:
For each surelog process, there is a Python fork that runs at 100% CPU currently, but that doesn't need to be.
You can query the status and maximum CPU use without any active polling.

hzeller, 2022-04-04:
The command on unix would be `time`, which then outputs all the stats such as wall time, user time, system time, memory etc.

In a programmatic way, you can do that using `getrusage()`

hs-apotell, 2022-04-04:
Yes that works on Unix only. There is no such equivalent for other platforms.

hzeller, 2022-04-04:
There gotta be a way on other platforms to figure out the resource usage of a program after it finished without using 100% of another core to monitor the processes ?

hs-apotell, 2022-04-06:
On Windows, I don't know of any easy way to do it. There is a documented way to do it using WMI but AFAIK its process instrumentation.

As far as the CPU usage goes, current implementation is using a yield which can be relaxed to a .5 second sleep. That change should improve CPU usage.

hzeller, 2022-04-06:
Could you use `getrusage()` on Unix/Mac and have a fallback for Windows ? If we can't measure the time accurately (and on top it occupies half of the cores) the regression script is not as useful as the original one.

hs-apotell, 2022-04-12:
I will explore the WMI implementation for Windows as time permits.

As a stop-gap solution, I have changed the sleep timers to 0.5 seconds which should ease the burden on the processor.

closed 2022-08-24 by alaindargelas