chipsalliance/Surelog #2827: implicit vs. explicit assignment statement

hs-apotell, 2022-04-07:
Is there a field that dictates whether a continuous assignment is implicit or explicit?

```
wire [1:0] a;
assign a = x & y; // explicit assignment
wire [1:0] a = x & y; // implicit assignment
```

alaindargelas, 2022-04-10:
Yes property: vpiNetDeclAssign:1
```
|vpiElaborated:1
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top), id:27 tests/ScratchPad.sv:1:1: , endln:6:10, parent:work@top, parID:26
  |vpiFullName:work@top
  |vpiDefName:work@top
  |vpiNet:
  \_logic_net: (work@top.a), id:28, line:2:12, endln:2:13, parent:work@top, parID:27
    |vpiName:a
    |vpiFullName:work@top.a
    |vpiNetType:1
  |vpiNet:
  \_logic_net: (work@top.a), id:29, line:4:12, endln:4:13, parent:work@top, parID:27
    |vpiName:a
    |vpiFullName:work@top.a
    |vpiNetType:1
  |vpiContAssign:
  \_cont_assign: , id:4, line:3:8, endln:3:17, parent:work@top, parID:27
    |vpiRhs:
    \_operation: , id:2, line:3:12, endln:3:17, parID:4
      |vpiOpType:28
      |vpiOperand:
      \_ref_obj: (work@top.x), id:1, line:3:12, endln:3:13, parID:2
        |vpiName:x
        |vpiFullName:work@top.x
        |vpiActual:
        \_logic_net: (x), id:31
          |vpiName:x
          |vpiNetType:1
      |vpiOperand:
      \_ref_obj: (work@top.y), id:3, line:3:16, endln:3:17, parID:2
        |vpiName:y
        |vpiFullName:work@top.y
        |vpiActual:
        \_logic_net: (y), id:32
          |vpiName:y
          |vpiNetType:1
    |vpiLhs:
    \_ref_obj: (work@top.a), id:0, line:3:8, endln:3:9, parID:4
      |vpiName:a
      |vpiFullName:work@top.a
      |vpiActual:
      \_logic_net: (work@top.a), id:13, line:2:12, endln:2:13, parent:work@top, parID:30
        |vpiTypespec:
        \_logic_typespec: , id:9, line:2:1, endln:2:5
          |vpiRange:
          \_range: , id:5, line:2:7, endln:2:10
            |vpiLeftRange:
            \_constant: , id:6, line:2:7, endln:2:8, parID:5
              |vpiDecompile:1
              |vpiSize:64
              |UINT:1
              |vpiConstType:9
            |vpiRightRange:
            \_constant: , id:8, line:2:9, endln:2:10, parID:5
              |vpiDecompile:0
              |vpiSize:64
              |UINT:0
              |vpiConstType:9
        |vpiName:a
        |vpiFullName:work@top.a
        |vpiNetType:1
|uhdmtopModules:
\_module: work@top (work@top), id:30 tests/ScratchPad.sv:1:1: , endln:6:10
  |vpiName:work@top
  |vpiDefName:work@top
  |vpiTop:1
  |vpiNet:
  \_logic_net: (work@top.a), id:13, line:2:12, endln:2:13, parent:work@top, parID:30
  |vpiNet:
  \_logic_net: (work@top.a), id:24, line:4:12, endln:4:13, parent:work@top, parID:30
    |vpiTypespec:
    \_logic_typespec: , id:17, line:4:1, endln:4:5
      |vpiRange:
      \_range: , id:14, line:4:7, endln:4:10
        |vpiLeftRange:
        \_constant: , id:15, line:4:7, endln:4:8, parID:14
          |vpiDecompile:1
          |vpiSize:64
          |UINT:1
          |vpiConstType:9
        |vpiRightRange:
        \_constant: , id:16, line:4:9, endln:4:10, parID:14
          |vpiDecompile:0
          |vpiSize:64
          |UINT:0
          |vpiConstType:9
    |vpiName:a
    |vpiFullName:work@top.a
    |vpiNetType:1
  |vpiNet:
  \_logic_net: (x), id:31
  |vpiNet:
  \_logic_net: (y), id:32
  |vpiTopModule:1
  |vpiContAssign:
  \_cont_assign: , id:33, line:4:12, endln:4:13, parent:work@top, parID:30
    |vpiNetDeclAssign:1
    |vpiRhs:
    \_operation: , id:34, line:4:16, endln:4:21, parID:33
      |vpiOpType:28
      |vpiOperand:
      \_ref_obj: (work@top.x), id:35, line:4:16, endln:4:17, parID:34
        |vpiName:x
        |vpiFullName:work@top.x
        |vpiActual:
        \_logic_net: (x), id:31
      |vpiOperand:
      \_ref_obj: (work@top.y), id:36, line:4:20, endln:4:21, parID:34
        |vpiName:y
        |vpiFullName:work@top.y
        |vpiActual:
        \_logic_net: (y), id:32
    |vpiLhs:
    \_logic_net: (work@top.a), id:13, line:2:12, endln:2:13, parent:work@top, parID:30
  |vpiContAssign:
  \_cont_assign: , id:37, line:3:8, endln:3:17, parent:work@top, parID:30
    |vpiRhs:
    \_operation: , id:39, line:3:12, endln:3:17, parID:37
      |vpiOpType:28
      |vpiOperand:
      \_ref_obj: (work@top.x), id:40, line:3:12, endln:3:13, parID:39
        |vpiName:x
        |vpiFullName:work@top.x
        |vpiActual:
        \_logic_net: (x), id:31
      |vpiOperand:
      \_ref_obj: (work@top.y), id:41, line:3:16, endln:3:17, parID:39
        |vpiName:y
        |vpiFullName:work@top.y
        |vpiActual:
        \_logic_net: (y), id:32
    |vpiLhs:
    \_ref_obj: (work@top.a), id:38, line:3:8, endln:3:9, parID:37
      |vpiName:a
      |vpiFullName:work@top.a
      |vpiActual:
      \_logic_net: (work@top.a), id:13, line:2:12, endln:2:13, parent:work@top, parID:30
```

alaindargelas, 2022-04-10:
in a for loop, it is an assign_stmt, not a cont_assign

closed 2022-04-10 by alaindargelas