chipsalliance/Surelog #2831: Can the inlining/substitutions be delayed until elaboration?

hs-apotell, 2022-04-09:
For the following statements information is lost and is required for language server and linting. Is it possible to delay the inlining/substitution until elaboration so the information can be retained in non-elaborated model?
```
localparam ModeW = 4;
typedef struct packed {
  logic [ModeW-1:0] mode;
} satp_t;
```
Relevant UHDM output below. Notice that the range for the `logic_typespec` is `constant` and not an `operation` with `ModeW` as one of the operand.
```
  \_struct_typespec: (riscv::satp_t), id:478, line:82:13, endln:82:19, parent:riscv::, parID:0
    |vpiName:riscv::satp_t
    |vpiPacked:1
    |vpiTypespecMember:
    \_typespec_member: (mode), id:487, line:83:27, endln:83:31, parent:riscv::satp_t, parID:478
      |vpiName:mode
      |vpiTypespec:
      \_logic_typespec: , id:486, line:83:9, endln:83:26
        |vpiInstance:
        \_package: riscv (riscv::), id:0 dut.sv:1:1: , endln:671:11, parent:work@top, parID:15156
        |vpiRange:
        \_range: , id:479, line:83:16, endln:83:25, parent:riscv::satp_t, parID:478
          |vpiLeftRange:
          \_constant: , id:484, line:83:16, endln:83:21, parID:479
            |vpiDecompile:3
            |vpiSize:64
            |INT:3
            |vpiConstType:7
          |vpiRightRange:
          \_constant: , id:485, line:83:24, endln:83:25, parID:479
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
            |vpiConstType:9
      |vpiRefFile:dut.sv
      |vpiRefLineNo:83
      |vpiRefColumnNo:9
      |vpiRefEndLineNo:83
      |vpiRefEndColumnNo:26
```

alaindargelas, 2022-04-09:
It's right there:

```
module top();
localparam ModeW = 4;
typedef struct packed {
  logic [ModeW-1:0] mode;
} satp_t;
endmodule

====== UHDM =======
design: (work@top), id:18
|vpiElaborated:1
|vpiName:work@top
|uhdmallModules:
\_module: work@top (work@top), id:19 tests/ScratchPad.sv:1:1: , endln:6:10, parent:work@top, parID:18
  |vpiFullName:work@top
  |vpiParameter:
  \_parameter: (work@top.ModeW), id:0, line:2:12, endln:2:17, parent:work@top, parID:19
    |UINT:4
    |vpiTypespec:
    \_int_typespec: , id:3
    |vpiLocalParam:1
    |vpiName:ModeW
    |vpiFullName:work@top.ModeW
  |vpiParamAssign:
  \_param_assign: , id:1, line:2:12, endln:2:21, parent:work@top, parID:19
    |vpiRhs:
    \_constant: , id:2, line:2:20, endln:2:21
      |vpiDecompile:4
      |vpiSize:32
      |UINT:4
      |vpiTypespec:
      \_int_typespec: , id:3
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (work@top.ModeW), id:0, line:2:12, endln:2:17, parent:work@top, parID:19
  |vpiTypedef:
  \_struct_typespec: (satp_t), id:5, line:3:9, endln:3:15, parent:work@top, parID:19
    |vpiName:satp_t
    |vpiInstance:
    \_module: work@top (work@top), id:19 tests/ScratchPad.sv:1:1: , endln:6:10, parent:work@top, parID:18
    |vpiPacked:1
    |vpiTypespecMember:
    \_typespec_member: (mode), id:12, line:4:21, endln:4:25, parent:satp_t, parID:5
      |vpiName:mode
      |vpiTypespec:
      \_logic_typespec: , id:11, line:4:3, endln:4:8
        |vpiRange:
        \_range: , id:6, line:4:10, endln:4:19, parent:satp_t, parID:5
          |vpiLeftRange:
          \_operation: , id:8, line:4:10, endln:4:17, parID:6
            |vpiOpType:11
            |vpiOperand:
            \_ref_obj: (work@top.satp_t.ModeW), id:7, line:4:10, endln:4:15, parent:satp_t, parID:5    <<<<<<<<<<<<<<<<<<<<<<<<<<<<
              |vpiName:ModeW
              |vpiFullName:work@top.satp_t.ModeW
            |vpiOperand:
            \_constant: , id:9, line:4:16, endln:4:17, parID:8
              |vpiDecompile:1
              |vpiSize:64
              |UINT:1
              |vpiConstType:9
          |vpiRightRange:
          \_constant: , id:10, line:4:18, endln:4:19, parID:6
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
            |vpiConstType:9
      |vpiRefFile:tests/ScratchPad.sv
      |vpiRefLineNo:4
      |vpiRefColumnNo:3
      |vpiRefEndLineNo:4
      |vpiRefEndColumnNo:20
  |vpiDefName:work@top
|uhdmtopModules:
\_module: work@top (work@top), id:20 tests/ScratchPad.sv:1:1: , endln:6:10
  |vpiName:work@top
  |vpiParameter:
  \_parameter: (work@top.ModeW), id:21, line:2:12, endln:2:17, parent:work@top, parID:20
    |UINT:4
    |vpiTypespec:
    \_int_typespec: , id:22, parent:work@top.ModeW, parID:21
    |vpiLocalParam:1
    |vpiName:ModeW
    |vpiFullName:work@top.ModeW
  |vpiParamAssign:
  \_param_assign: , id:15, line:2:12, endln:2:21, parent:work@top, parID:20
    |vpiRhs:
    \_constant: , id:16, line:2:20, endln:2:21
      |vpiDecompile:4
      |vpiSize:32
      |UINT:4
      |vpiTypespec:
      \_int_typespec: , id:3
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (work@top.ModeW), id:21, line:2:12, endln:2:17, parent:work@top, parID:20
  |vpiTypedef:
  \_struct_typespec: (satp_t), id:5, line:3:9, endln:3:15, parent:work@top, parID:19
  |vpiDefName:work@top
  |vpiTop:1
  |vpiTopModule:1
===================
```

hs-apotell, 2022-04-10:
Something isn't right. I am working with ArianeElab test case and I don't see similar output. The example above was from that same test. Is there some additional command line parameter to prevent these substitutions?

alaindargelas, 2022-04-10:
Send the complete small testcase that reproduce the problem, not just a code snippet. The devil is in the detail, for example, package might have more constant pushing than modules.

hs-apotell, 2022-04-11:
```
package pkg;
  localparam ModeW = 4;
  localparam XLEN = 64;

  localparam logic [XLEN-1:0] INSTR_ADDR_MISALIGNED = 0;

  typedef struct packed {
    logic [ModeW-1:0] mode;
  } satp_t;
endpackage
```
For both line 5 and 8, the `operation` has been reduced to a `constant`
```
design: (unnamed), id:47
|vpiElaborated:1
|vpiName:unnamed
|uhdmallPackages:
\_package: pkg (pkg::), id:48 test.sv:1:1: , endln:10:11, parent:unnamed, parID:47
  |vpiName:pkg
  |vpiFullName:pkg::
  |vpiParamAssign:
  \_param_assign: , id:6, line:2:14, endln:2:23, parent:pkg::, parID:48
    |vpiRhs:
    \_constant: , id:5, line:2:22, endln:2:23
      |vpiDecompile:4
      |vpiSize:64
      |UINT:4
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (ModeW), id:7, line:2:14, endln:2:19
      |vpiLocalParam:1
      |vpiName:ModeW
  |vpiParamAssign:
  \_param_assign: , id:14, line:3:14, endln:3:23, parent:pkg::, parID:48
    |vpiRhs:
    \_constant: , id:13, line:3:21, endln:3:23
      |vpiDecompile:64
      |vpiSize:64
      |UINT:64
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (XLEN), id:15, line:3:14, endln:3:18
      |vpiLocalParam:1
      |vpiName:XLEN
  |vpiParamAssign:
  \_param_assign: , id:30, line:5:31, endln:5:56, parent:pkg::, parID:48
    |vpiRhs:
    \_constant: , id:29, line:5:55, endln:5:56
      |vpiDecompile:0
      |vpiSize:64
      |UINT:0
      |vpiConstType:9
    |vpiLhs:
    \_parameter: (INSTR_ADDR_MISALIGNED), id:31, line:5:31, endln:5:52
      |vpiTypespec:
      \_logic_typespec: , id:32, line:5:14, endln:5:19, parent:INSTR_ADDR_MISALIGNED, parID:31
        |vpiInstance:
        \_package: pkg (pkg::), id:48 test.sv:1:1: , endln:10:11, parent:unnamed, parID:47
        |vpiRange:
        \_range: , id:33, line:5:21, endln:5:29, parID:32
          |vpiLeftRange:
          \_constant: , id:34, line:5:21, endln:5:25, parID:33
            |vpiDecompile:63
            |vpiSize:64
            |INT:63
            |vpiConstType:7
          |vpiRightRange:
          \_constant: , id:35, line:5:28, endln:5:29, parID:33
            |vpiDecompile:0
[... 96 lines cut ...]
      \_logic_typespec: , id:45, line:8:5, endln:8:10
        |vpiInstance:
        \_package: pkg (pkg::), id:0 test.sv:1:1: , endln:10:11, parent:unnamed, parID:47
        |vpiRange:
        \_range: , id:38, line:8:12, endln:8:21, parent:pkg::satp_t, parID:37
          |vpiLeftRange:
          \_constant: , id:43, line:8:12, endln:8:17, parID:38
            |vpiDecompile:3
            |vpiSize:64
            |INT:3
            |vpiConstType:7
          |vpiRightRange:
          \_constant: , id:44, line:8:20, endln:8:21, parID:38
            |vpiDecompile:0
            |vpiSize:64
            |UINT:0
            |vpiConstType:9
      |vpiRefFile:test.sv
      |vpiRefLineNo:8
      |vpiRefColumnNo:5
      |vpiRefEndLineNo:8
      |vpiRefEndColumnNo:22
  |vpiDefName:pkg
  |vpiTop:1
```

alaindargelas, 2022-04-19:
I'm working on this, it is a large changelist and it increases runtime and uhdm size as it duplicates the packages

alaindargelas, 2022-04-20:
Fixed by #2875

closed 2022-04-20 by alaindargelas