chipsalliance/Surelog #2832: case + inside

hs-apotell, 2022-04-09:
Test case:
```
typedef enum logic [6:0] {
  EQ, NE, LTS, LTU, GES, GEU
} fu_op;

function automatic logic op_is_branch (input fu_op op);
  case (op) // inside
    EQ, NE: return 1'b1;
    default: return 1'b0;
  endcase
endfunction
```

Uhdm output without the `inside` keyword:
```
  \_case_stmt: , id:16, line:6:3, endln:9:10, parent:op_is_branch, parID:12
    |vpiCaseType:1
    |vpiCondition:
    \_ref_obj: (op_is_branch.op), id:15, line:6:9, endln:6:11, parent:op_is_branch, parID:12
      |vpiName:op
      |vpiFullName:op_is_branch.op
      |vpiActual:
      \_io_decl: (op), id:14, line:5:52, endln:5:54
    |vpiCaseItem:
    \_case_item: , id:17, line:7:5, endln:7:25, parID:16
      |vpiExpr:
      \_ref_obj: (op_is_branch.EQ), id:18, line:7:5, endln:7:7, parent:op_is_branch, parID:12
        |vpiName:EQ
        |vpiFullName:op_is_branch.EQ
      |vpiExpr:
      \_ref_obj: (op_is_branch.NE), id:19, line:7:9, endln:7:11, parent:op_is_branch, parID:12
        |vpiName:NE
        |vpiFullName:op_is_branch.NE
```

Uhdm output with the `inside` keyword:
```
  \_case_stmt: , id:16, line:6:3, endln:9:10, parent:op_is_branch, parID:12
    |vpiCaseType:1
    |vpiCondition:
    \_ref_obj: (op_is_branch.op), id:15, line:6:9, endln:6:11, parent:op_is_branch, parID:12
      |vpiName:op
      |vpiFullName:op_is_branch.op
      |vpiActual:
      \_io_decl: (op), id:14, line:5:52, endln:5:54        
    \_case_item: , id:17, line:7:5, endln:7:25, parID:16
      |vpiExpr:
      \_operation: , id:18, line:7:5, endln:7:7, parID:17
        |vpiOpType:37
        |vpiOperand:
        \_ref_obj: (EQ), id:19, line:7:5, endln:7:7
          |vpiName:EQ
      |vpiExpr:
      \_operation: , id:20, line:7:9, endln:7:11, parID:17
        |vpiOpType:37
        |vpiOperand:
        \_ref_obj: (NE), id:21, line:7:9, endln:7:11
          |vpiName:NE
```

Issues:
* No way to differentiate between presence or absence of the keyword `inside` since both the default and inside map to the same case type value of 1.
* In the case of `inside` use, shouldn't the `case_item` expression be a single `operation` with multiple operands rather than multiple expressions each with an `operation` (and one operand).

alaindargelas, 2022-04-10:
Fixed by #2835

closed 2022-04-10 by alaindargelas

hs-apotell, 2022-04-11:
> No way to differentiate between presence or absence of the keyword inside since both the default and inside map to the same case type value of 1.

How does your solution address the above issue? I didn't find any way to identify the presence of the keyword 'inside'.

alaindargelas, 2022-04-12:
[tests/CaseInside/CaseInside.log](https://github.com/chipsalliance/Surelog/pull/2835/files#diff-4fdb13ac1554de1b346b4e99c7610784f7ebfb19d7519bfa27b6345d12b53468)[](https://github.com/chipsalliance/Surelog/pull/2835/files#diff-4fdb13ac1554de1b346b4e99c7610784f7ebfb19d7519bfa27b6345d12b53468)
Viewed
@@ -888,20 +888,23 @@ design: (work@dm_csrs)
        |vpiCaseItem:
        \_case_item: , line:17:9, endln:21:12
          |vpiExpr:
          \_operation: , line:17:9, endln:17:34
            |vpiOpType:37
          \_operation: 
            |vpiOpType:95   <<< vpiInsideOp

alaindargelas, 2022-04-12:
I can't see the comments.
- Can't there be multiple case_items each with a different op-type?
 Yes, the operation with vpiInsideOpType will only be used for the ones with the inside keywork, other will have no operation, only expressions.
The inside does apply to the case item, please note the default is another case item,
check some of the examples of Ariane.
As far as I'm concerned this is all working the way it is supposed to.

https://github.com/chipsalliance/Surelog/blob/878ee9b836c446e7fe5f0588d1edf55ca27a10d2/tests/ArianeElab/dut.sv#L1144

```
 |vpiStmt:
      \_begin: (ariane_pkg::is_rs1_fpr), line:1143:25, endln:1155:12
        |vpiFullName:ariane_pkg::is_rs1_fpr
        |vpiStmt:
        \_case_stmt: , line:1144:13, endln:1154:20, parent:ariane_pkg::is_rs1_fpr
          |vpiCaseType:1
          |vpiQualifier:1
          |vpiCondition:
          \_ref_obj: (ariane_pkg::is_rs1_fpr::op), line:1144:26, endln:1144:28
            |vpiName:op
            |vpiFullName:ariane_pkg::is_rs1_fpr::op
            |vpiActual:
            \_io_decl: (op), line:1142:54, endln:1142:56, parent:ariane_pkg::is_rs1_fpr
          |vpiCaseItem:
          \_case_item: , line:1145:17, endln:1152:49
            |vpiExpr:
            \_operation: 
              |vpiOpType:95
              |vpiOperand:
              \_operation: , line:1145:17, endln:1145:30
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FMUL), line:1145:18, endln:1145:22
                  |vpiName:FMUL
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FMUL
                  |vpiActual:
                  \_enum_const: (FMUL), line:1112:44, endln:1112:48
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FNMADD), line:1145:23, endln:1145:29
                  |vpiName:FNMADD
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FNMADD
                  |vpiActual:
                  \_enum_const: (FNMADD), line:1112:95, endln:1112:101
              |vpiOperand:
              \_operation: , line:1146:17, endln:1146:25
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FCVT_F2I), line:1146:17, endln:1146:25
                  |vpiName:FCVT_F2I
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FCVT_F2I
                  |vpiActual:
                  \_enum_const: (FCVT_F2I), line:1114:32, endln:1114:40
              |vpiOperand:
              \_operation: , line:1147:17, endln:1147:25
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FCVT_F2F), line:1147:17, endln:1147:25
                  |vpiName:FCVT_F2F
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FCVT_F2F
                  |vpiActual:
                  \_enum_const: (FCVT_F2F), line:1114:52, endln:1114:60
              |vpiOperand:
              \_operation: , line:1148:17, endln:1148:22
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FSGNJ), line:1148:17, endln:1148:22
                  |vpiName:FSGNJ
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FSGNJ
                  |vpiActual:
                  \_enum_const: (FSGNJ), line:1114:62, endln:1114:67
              |vpiOperand:
              \_operation: , line:1149:17, endln:1149:24
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FMV_F2X), line:1149:17, endln:1149:24
                  |vpiName:FMV_F2X
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FMV_F2X
                  |vpiActual:
                  \_enum_const: (FMV_F2X), line:1114:69, endln:1114:76
              |vpiOperand:
              \_operation: , line:1150:17, endln:1150:21
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FCMP), line:1150:17, endln:1150:21
                  |vpiName:FCMP
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FCMP
                  |vpiActual:
                  \_enum_const: (FCMP), line:1116:32, endln:1116:36
              |vpiOperand:
              \_operation: , line:1151:17, endln:1151:23
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FCLASS), line:1151:17, endln:1151:23
                  |vpiName:FCLASS
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FCLASS
                  |vpiActual:
                  \_enum_const: (FCLASS), line:1118:32, endln:1118:38
              |vpiOperand:
              \_operation: , line:1152:17, endln:1152:34
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::VFMIN), line:1152:18, endln:1152:23
                  |vpiName:VFMIN
                  |vpiFullName:ariane_pkg::is_rs1_fpr::VFMIN
                  |vpiActual:
                  \_enum_const: (VFMIN), line:1120:32, endln:1120:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::VFCPKCD_D), line:1152:24, endln:1152:33
                  |vpiName:VFCPKCD_D
                  |vpiFullName:ariane_pkg::is_rs1_fpr::VFCPKCD_D
                  |vpiActual:
                  \_enum_const: (VFCPKCD_D), line:1120:141, endln:1120:150
            |vpiStmt:
            \_return_stmt: , line:1152:37, endln:1152:43
              |vpiCondition:
              \_constant: , line:1152:44, endln:1152:48
                |vpiDecompile:1'b1
                |vpiSize:1
                |BIN:1
                |vpiConstType:3
          |vpiCaseItem:
          \_case_item: , line:1153:17, endln:1153:49
            |vpiStmt:
            \_return_stmt: , line:1153:37, endln:1153:43
              |vpiCondition:
              \_constant: , line:1153:44, endln:1153:48
                |vpiDecompile:1'b0
                |vpiSize:1
                |BIN:0
                |vpiConstType:3
```

hs-apotell, 2022-04-12:
Check the comments on this page - https://github.com/chipsalliance/Surelog/pull/2835

In ArianElab -
https://github.com/chipsalliance/Surelog/blob/878ee9b836c446e7fe5f0588d1edf55ca27a10d2/tests/ArianeElab/dut.sv#L1144-L1154

```
|vpiStmt:
      \_begin: (ariane_pkg::is_rs1_fpr), line:1143:25, endln:1155:12
        |vpiFullName:ariane_pkg::is_rs1_fpr
        |vpiStmt:
        \_case_stmt: , line:1144:13, endln:1154:20, parent:ariane_pkg::is_rs1_fpr
          |vpiCaseType:1
          |vpiQualifier:1
          |vpiCondition:
          \_ref_obj: (ariane_pkg::is_rs1_fpr::op), line:1144:26, endln:1144:28
            |vpiName:op
            |vpiFullName:ariane_pkg::is_rs1_fpr::op
            |vpiActual:
            \_io_decl: (op), line:1142:54, endln:1142:56, parent:ariane_pkg::is_rs1_fpr
          |vpiCaseItem:
          \_case_item: , line:1145:17, endln:1152:49
            |vpiExpr:
            \_operation: 
              |vpiOpType:95
              |vpiOperand:
              \_operation: , line:1145:17, endln:1145:30
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FMUL), line:1145:18, endln:1145:22
                  |vpiName:FMUL
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FMUL
                  |vpiActual:
                  \_enum_const: (FMUL), line:1112:44, endln:1112:48
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FNMADD), line:1145:23, endln:1145:29
                  |vpiName:FNMADD
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FNMADD
                  |vpiActual:
                  \_enum_const: (FNMADD), line:1112:95, endln:1112:101
              |vpiOperand:
              \_operation: , line:1146:17, endln:1146:25
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FCVT_F2I), line:1146:17, endln:1146:25
                  |vpiName:FCVT_F2I
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FCVT_F2I
                  |vpiActual:
                  \_enum_const: (FCVT_F2I), line:1114:32, endln:1114:40
              |vpiOperand:
              \_operation: , line:1147:17, endln:1147:25
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FCVT_F2F), line:1147:17, endln:1147:25
                  |vpiName:FCVT_F2F
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FCVT_F2F
                  |vpiActual:
                  \_enum_const: (FCVT_F2F), line:1114:52, endln:1114:60
              |vpiOperand:
              \_operation: , line:1148:17, endln:1148:22
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FSGNJ), line:1148:17, endln:1148:22
                  |vpiName:FSGNJ
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FSGNJ
                  |vpiActual:
                  \_enum_const: (FSGNJ), line:1114:62, endln:1114:67
              |vpiOperand:
              \_operation: , line:1149:17, endln:1149:24
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FMV_F2X), line:1149:17, endln:1149:24
                  |vpiName:FMV_F2X
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FMV_F2X
                  |vpiActual:
                  \_enum_const: (FMV_F2X), line:1114:69, endln:1114:76
              |vpiOperand:
              \_operation: , line:1150:17, endln:1150:21
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FCMP), line:1150:17, endln:1150:21
                  |vpiName:FCMP
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FCMP
                  |vpiActual:
                  \_enum_const: (FCMP), line:1116:32, endln:1116:36
              |vpiOperand:
              \_operation: , line:1151:17, endln:1151:23
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::FCLASS), line:1151:17, endln:1151:23
                  |vpiName:FCLASS
                  |vpiFullName:ariane_pkg::is_rs1_fpr::FCLASS
                  |vpiActual:
                  \_enum_const: (FCLASS), line:1118:32, endln:1118:38
              |vpiOperand:
              \_operation: , line:1152:17, endln:1152:34
                |vpiOpType:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::VFMIN), line:1152:18, endln:1152:23
                  |vpiName:VFMIN
                  |vpiFullName:ariane_pkg::is_rs1_fpr::VFMIN
                  |vpiActual:
                  \_enum_const: (VFMIN), line:1120:32, endln:1120:37
                |vpiOperand:
                \_ref_obj: (ariane_pkg::is_rs1_fpr::VFCPKCD_D), line:1152:24, endln:1152:33
                  |vpiName:VFCPKCD_D
                  |vpiFullName:ariane_pkg::is_rs1_fpr::VFCPKCD_D
                  |vpiActual:
                  \_enum_const: (VFCPKCD_D), line:1120:141, endln:1120:150
            |vpiStmt:
            \_return_stmt: , line:1152:37, endln:1152:43
              |vpiCondition:
              \_constant: , line:1152:44, endln:1152:48
                |vpiDecompile:1'b1
                |vpiSize:1
                |BIN:1
                |vpiConstType:3
          |vpiCaseItem:
          \_case_item: , line:1153:17, endln:1153:49
            |vpiStmt:
            \_return_stmt: , line:1153:37, endln:1153:43
              |vpiCondition:
              \_constant: , line:1153:44, endln:1153:48
                |vpiDecompile:1'b0
                |vpiSize:1
                |BIN:0
                |vpiConstType:3
```

Expression `[FMUL:FNMADD]` & `[VFMIN:VFCPKCD_D]` are coming out as `vpiListOp`. Is that expected??

alaindargelas, 2022-10-11:
Correct, "ranges" in expressions are modeled as a list of 2 members, left
and right range