chipsalliance/Surelog #2856: Compile failing on windows msvc because of cmake binary output paths

0x5b25, 2022-04-14:
Compile failing  on windows msvc because of cmake binary paths, capnp.exe and surelog.exe can't be found causing custom commands failed.

CMake with msvc is putting binaries inside `$<CONFIG>` subfolders, causing the problem

Environment:
  * cmake version 3.22.0
  * visual studio 2019, msvc 16.11.7(x64)

Current workaround:
change line 115 in third_party/UHDM/CMakeLists.txt
```
- set(CAPNP_DIR ${CMAKE_CURRENT_BINARY_DIR}/third_party/capnproto/c++/src/capnp)
+ set(CAPNP_DIR ${CMAKE_CURRENT_BINARY_DIR}/third_party/capnproto/c++/src/capnp/$<CONFIG>)
```

and line 603, 617 in CMakeLists.txt
```
*line 603:
- ${CMAKE_RUNTIME_OUTPUT_DIRECTORY}/surelog -nobuiltin -createcache
+ ${CMAKE_RUNTIME_OUTPUT_DIRECTORY}/$<CONFIG>/surelog -nobuiltin -createcache

*line 617:
- ${CMAKE_RUNTIME_OUTPUT_DIRECTORY}/surelog -nobuiltin -createcache
+ ${CMAKE_RUNTIME_OUTPUT_DIRECTORY}/$<CONFIG>/surelog -nobuiltin -createcache
```

hs-apotell, 2022-04-16:
The builds on CI are passing and so are local builds as well. I am building it locally almost daily. Please check your configuration. 

If you are using VS to build, please share your CMakeSettings.json file.

alaindargelas, 2022-04-20:
@0x5b25, can we close this issue?

0x5b25, 2022-04-20:
Fine. I'll try a clean install of windows and cmake some other day i guess

closed 2022-04-20 by 0x5b25